espressif/esp-idf #12740: Fail to open NVS for read.

shlomo-edu, 2023-12-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I call the attached `read_device_name_from_flash()` function.
The `nvs_open()` returns: ESP_ERR_NVS_NOT_FOUND
Why?

`void read_device_name_from_flash()
{
    nvs_handle_t my_handle;
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) 
    {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);
    esp_err_t err = nvs_open("storage", NVS_READONLY, &my_handle);
    if (err != ESP_OK) 
    {
     ESP_LOGE("NVS", "Failed to open NVS handle: %s", esp_err_to_name(err));
     return;
    }   
    size_t required_size=30;
    err = nvs_get_str(my_handle, "device_name", my_device_name, &required_size);
    switch (err) {
    case ESP_OK:
        printf("NVS: my_device_name = %s\n", my_device_name);
        break;
    case ESP_ERR_NVS_NOT_FOUND:
        printf("NVS: my_device_name is not initialize yet!\n");
        break;
    default :
        printf("NVS: Error (%s) reading!\n", esp_err_to_name(err));
    }
    nvs_close(my_handle);
}
`

KonstantinKondrashov, 2023-12-07:
Hi @shlomo-edu!
The code that you attached is not readable, please format it.
Make sure that you use the partition table that has an nvs partition. [The NVS doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html?highlight=nvs#non-volatile-storage-library).
Also, try to run this example - https://github.com/espressif/esp-idf/tree/30870c8/examples/storage/nvs_rw_value.
Additionally, I suggest you erase the flash and flash all again.

TD-er, 2023-12-07:
Maybe you can also draw inspiration from the Arduino ESP32 implementation of the `Preferences` class: https://github.com/espressif/arduino-esp32/blob/master/libraries/Preferences/src/Preferences.cpp
This is actually an abstraction of the NVS layer in ESP-IDF.
Maybe this can help you find whether you're missing something.
I can confirm the Preferences class does work, so it is a working example of how to use NVS.

shlomo-edu, 2023-12-07:
Hi @KonstantinKondrashov 
Yes, the erase the flash fixed it.
Thanks.

closed 2023-12-07 by shlomo-edu