espressif/esp-idf #12742: Make the Scan Channel configurable for NimBLE Observer

FredoTorpedo, 2023-12-06:
### Is your feature request related to a problem?

We use ESP devices to collect beacon data and RSSI for locating the ESP. This location algorith is better if we now from which channel we receive. Therefore it would be nice to be able to configure the channel.

### Describe the solution you'd like.

In the file:

[IDF_PATH]\components\bt\host\nimble\nimble\nimble\host\include\host\ble_gap.h

It should be possible to set the scan channel by adding a `uint8_t scan_chan` parameter to the` struct ble_gap_upd_params ` and setting it before initializing scan mode.

The input could look like this:
enum{
    SCAN_CHANNEL_SET_ALL = 0,
    SCAN_CHANNEL_SET_37,
    SCAN_CHANNEL_SET_38,
    SCAN_CHANNEL_SET_39
};

With Bluedroid there has been a workaround which wasn't quite transparent in how it worked. Here:
[https://github.com/espressif/esp-idf/issues/4705#issuecomment-705937243](url)

But I assume theres some point where it is passed in the HCI (or lower) Layer.

It would be nice to make it available somehow.

Thank you.

### Describe alternatives you've considered.

Keeping Bluedroid, but the stack is way too big for logging Beacons.

jonsmirl, 2024-01-05:
I need the same thing and can't get this to work. How do you make it link? I added this:
~~~
typedef enum
{
    SCAN_ALL_CHANNEL = 0,
    ONLY_SCAN_CHANNEL_37,
    ONLY_SCAN_CHANNEL_38,
    ONLY_SCAN_CHANNEL_39,
} scan_mode_config_t;
extern "C" scan_mode_config_t scan_channel_setting(scan_mode_config_t scan_mode);
~~~
but it won't link
~~~
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ble_advert.cpp.obj):(.literal._Z9StartScanv+0xc): undefined reference to `scan_channel_setting'
/home/jonsmirl/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/12.2.0/../../../../xtensa-esp32s3-elf/bin/ld: esp-idf/main/libmain.a(ble_advert.cpp.obj): in function `_Z9StartScanv':
/home/jonsmirl/lowpan/main/ble_advert.cpp:250: undefined reference to `scan_channel_setting'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
~~~

And why isn't this function in the h files?

rahult-github, 2024-02-14:
Issue has been discussed here --> https://www.esp32.com/viewtopic.php?f=13&t=37638  ..  Will internally route to concerned team for further clarifications.

esp-lrh, 2024-03-27:
I left a comment in https://www.esp32.com/viewtopic.php?p=129856#p129856 , please note that.

rahult-github, 2024-09-24:
Closing issue. Please feel free to reopen in case of further requests.

closed 2024-09-24 by rahult-github