espressif/esp-idf #12743: nimble panic when using random address with idf-v5.2 branch

AxelLin, 2023-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-4169-gbf816394d4

### Espressif SoC revision.

ESP32C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3DevKit-M1

### Power Supply used.

USB

### What is the expected behavior?

My application using BLE random address was working fine if build with v5.1 branch, but hit panic when build with v5.2 branch.

### What is the actual behavior?

I hit below panic when switch to build with v5.2-dev-4169-gbf816394d4 for esp32c3.
(see the log below)

### Steps to reproduce.

I'm not allow to share my application, which is using nimble host with external bt controller.
But the key point is in ble_hs_pvcy_set_default_irk() the local_id could be NULL which causes this panic.

### Debug Logs.

```plain
I hit below panic when switch to build with v5.2-dev-4169-gbf816394d4 for esp32c3.

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Stack dump detected
Core  0 register dump:
MEPC    : 0x400587d2  RA      : 0x42027ffa  SP      : 0x3fcc18f0  GP      : 0x3fc94800
0x400587d2: memcpy in ROM

0x42027ffa: ble_hs_pvcy_set_default_irk at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:220

TP      : 0x3fc82f0c  T0      : 0x4005890e  T1      : 0x0000000f  T2      : 0xffffffff
0x4005890e: memset in ROM

S0/FP   : 0x00000000  S1      : 0x0000000d  A0      : 0x3fcc1901  A1      : 0x00000000
A2      : 0x00000006  A3      : 0x00000000  A4      : 0x3fcc1907  A5      : 0x3fcc1901
A6      : 0x3fcbe5c4  A7      : 0x00000014  S2      : 0x3fc9f000  S3      : 0x3fc9f000
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000
T3      : 0x3fc9f000  T4      : 0x00000001  T5      : 0x3fc9f000  T6      : 0x00000001
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x00000000
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000

Backtrace:

0x400587d2 in ?? ()
#0  0x400587d2 in ?? ()
#1  0x42027ffa in ble_hs_pvcy_set_default_irk () at /home/axel/esp/esp-idf/components/bt/host/nimble/nimble/nimble/host/src/ble_hs_pvcy.c:216
#2  0x00000000 in ?? ()
Backtrace stopped: frame did not save the PC
ELF file SHA256: c1655c9bc

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
```

### More Information.

I bisect to below bad commit:

commit e8170b6f5fd4185139b66a12eb5bb2e821dcebff (HEAD)
Author: Rahul Tank <rahul.tank@espressif.com>
Date:   Wed Sep 20 12:49:25 2023 +0530

     Added change to generate unique IRK for each chip

After further checking, I notice in ble_hs_pvcy_set_default_irk()
the local_id is NULL,
so below code causes above panic.

    /* Create key / value */
    memcpy (key_local_irk.addr.val , local_id, BLE_DEV_ADDR_LEN);

My setup is using BLE_OWN_ADDR_RANDOM,
1. The first problem is ble_hs_pvcy_set_default_irk() always calls BLE_ADDR_PUBLIC,
   which is not correct (because I'm using BLE_OWN_ADDR_RANDOM, thus local_id is still NULL.

2. The second problem is I setup the random address in my ble_hs_cfg.sync_cb callback.
   However, at the time ble_hs_pvcy_set_default_irk() is called, the sync_cb is not yet called.
   So calling ble_hs_id_addr with BLE_ADDR_RANDOM still returns error in ble_hs_pvcy_set_default_irk().

AxelLin, 2023-12-07:
> Is the external BT controller you are using not giving any response to read BD ADDR command ? That can be a possible reason for local_id's value to be NULL .
> 

It looks like nRF chipsets do not come with a public address registered and burned into the chip.
ble_hs_startup_read_bd_addr() returns 0, but the address is 00:00:00:00:00:00, which is the same as ble_hs_misc_null_addr.
As a result, ble_hs_id_addr returns BLE_HS_ENOADDR, local_id is NULL.

rahult-github, 2023-12-07:
Can you please help if attached patch works at your end. This should handle the case where controller address is all 0s and also generate IRK. 

[handle_all_zero_address.txt](https://github.com/espressif/esp-idf/files/13601324/handle_all_zero_address.txt)

AxelLin, 2023-12-07:
Hi @rahult-github 
Thanks for the quick fix, your patch works.

BTW, v5.1 branch also needs fix now, I hit the same issue after update esp-idf 5.1 branch.

AxelLin, 2023-12-23:
v5.1 fix: 04b6feb1addffc9c7e435a8d0b2839b9d6151431

closed 2023-12-23 by AxelLin