espressif/esp-idf #12746: Error: No mbedtls_ms_time available

Tim4497, 2023-12-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1-577-g6b1f40b9bf

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build correctly

### What is the actual behavior?

Build failed on C Object
[49/1166] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/platform_util.c.objFAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/platform_util.c.obj 

### Steps to reproduce.

Installing ESP-Matter on main, ESP-IDF on v5.1.1-577-g6b1f40b9bf

### Build or installation Logs.

```plain
[5/1285] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
esp_secure_cert,63,6,0xd000,8K,encrypted
nvs,data,nvs,0x10000,24K,
nvs_keys,data,nvs_keys,0x16000,4K,encrypted
otadata,data,ota,0x17000,8K,
phy_init,data,phy,0x19000,4K,
ota_0,app,ota_0,0x20000,1920K,
ota_1,app,ota_1,0x200000,1920K,
fctry,data,nvs,0x3e0000,24K,
fs_0,data,spiffs,0x3e6000,104K,
*******************************************************************************
[6/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesni.c.obj
[7/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesce.c.obj
[8/1285] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/x25519.c.obj
[9/1285] Generating args.gn
[10/1285] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/Hacl_Curve25519_joined.c.obj
[11/1285] Building C object esp-idf/mbedtls/mbedtls/3rdparty/everest/CMakeFiles/everest.dir/library/everest.c.obj
[12/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum_mod.c.obj
[13/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum_mod_raw.c.obj
[14/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/camellia.c.obj
[15/1285] Building C object esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj
[16/1285] Building C object esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m_driver_entrypoints.c.obj
[17/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chacha20.c.obj
[18/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/chachapoly.c.obj
[19/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1write.c.obj
[20/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/asn1parse.c.obj
[21/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/base64.c.obj
[22/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj
[... 24 lines cut ...]
[47/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/lmots.c.obj
[48/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/error.c.obj
[49/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/md.c.obj
[50/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/pem.c.obj
[51/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/platform_util.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/platform_util.c.obj 
/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/idf-tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.1.1-577-g6b1f40b9bf-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/build/config -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/mbedtls/port/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/mbedtls/mbedtls/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/mbedtls/mbedtls/library -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/build/esp-idf/mbedtls/mbedtls/library -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/newlib/platform_include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/freertos/esp_additions/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_hw_support/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/heap/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/log/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/soc/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/soc/esp32 -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/soc/esp32/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/hal/esp32/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/hal/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/hal/platform_port/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_rom/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_rom/include/esp32 -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_rom/esp32 -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_common/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_system/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_system/port/soc -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/esp_system/port/include/private -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/xtensa/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/xtensa/esp32/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/lwip/include -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/lwip/include/apps -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/t.winkler/Documents/REPLY/Schulthess_MATTER/laundry-washer/esp/ [... cut ...]
[60/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/pkwrite.c.obj
[61/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecp_curves.c.obj
[62/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/pkparse.c.obj
[63/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecp.c.obj
[64/1285] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/psa_crypto.c.obj
ninja: build stopped: subcommand failed.
```

Tim4497, 2023-12-07:
Only happening after switching IDF Version, not on new project.

Tim4497, 2023-12-11:
I have it enabled and still same error!

Harshal5, 2023-12-14:
Hello @Tim4497,

Could you share a small reproducible code and the sdkconfig that you have used?
I was unable to reproduce the above error with the following minimal code:

```
#include <stdio.h>
#include "mbedtls/platform_time.h"
#include "esp_log.h"

void app_main(void)
{
    mbedtls_ms_time_t current = mbedtls_ms_time();
    mbedtls_ms_time_t elapsed_ms;

    elapsed_ms = mbedtls_ms_time() - current;
    ESP_LOGE("", "%llu", elapsed_ms);
}
```

with the config `CONFIG_MBEDTLS_HAVE_TIME` being enabled or disabled.

mahavirj, 2024-01-04:
Closing due to lack of response, feel free to reopen if you are still facing the problem

closed 2024-01-04 by mahavirj