espressif/esp-idf #12752: Cannot select Host (NimBLE) when BT_CONTROLLER_DISABLED since v5.2+

AxelLin, 2023-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-dev-4169-gbf816394d4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

In v5.1 or earlier versions, It's ok to select select Host (NimBLE) when BT_CONTROLLER_DISABLED.

### What is the actual behavior?

Since v5.2, it's impossible to select Host (NimBLE) when BT_CONTROLLER_DISABLED.

See https://github.com/espressif/esp-idf/blob/master/components/bt/Kconfig#L24
config BT_NIMBLE_ENABLED should not depend on BT_CONTROLLER_ENABLED.
This dependency was added in v5.2+, no issue in v5.1 or earlier versions.

### Steps to reproduce.

Just use idf.py menuconfig
goto Bluetooth, select "Controller (disable)", then it's impossible to select NimBLE as host.

listout, 2023-12-09:
Seems like it's added as a feature in commit https://github.com/espressif/esp-idf/commit/fc174f581170cedfd4572d6ba13864850cf68fec

AxelLin, 2023-12-24:
> Seems like it's added as a feature in commit [fc174f5](https://github.com/espressif/esp-idf/commit/fc174f581170cedfd4572d6ba13864850cf68fec)

Above change is mainly for bluedroiad.
It should not touch the dependency of BT_NIMBLE_ENABLED.
Please fix this. @rahult-github

rahult-github, 2024-02-06:
Hi @AxelLin  , recently a commit was pushed into [master](https://github.com/espressif/esp-idf/commit/51efb2051655d4197f1bfaef8e68d53ef52b1e7c) to address this . Please help check if the fix works for you now.

AxelLin, 2024-02-13:
> Hi @AxelLin , recently a commit was pushed into [master](https://github.com/espressif/esp-idf/commit/51efb2051655d4197f1bfaef8e68d53ef52b1e7c) to address this . Please help check if the fix works for you now.

Hi @rahult-github 
I think the nimble_port.c needs below fix in esp_nimble_deinit(),
so the #if guard in esp_nimble_deinit() matches the one in esp_nimble_init().
```
diff --git a/porting/nimble/src/nimble_port.c b/porting/nimble/src/nimble_port.c
index 53b72c45..83b2c845 100644
--- a/porting/nimble/src/nimble_port.c
+++ b/porting/nimble/src/nimble_port.c
@@ -134,7 +134,7 @@ esp_err_t esp_nimble_init(void)
  */
 esp_err_t esp_nimble_deinit(void)
 {
-#if !SOC_ESP_NIMBLE_CONTROLLER
+#if !SOC_ESP_NIMBLE_CONTROLLER || !CONFIG_BT_CONTROLLER_ENABLED
 #if CONFIG_BT_CONTROLLER_ENABLED
     if(esp_nimble_hci_deinit() != ESP_OK) {
         ESP_LOGE(NIMBLE_PORT_LOG_TAG, "hci deinit failed\n");

```

RoshanESP, 2024-02-13:
Thank you @AxelLin for sharing the change. This change will be pushed soon.

closed 2024-02-20 by AxelLin