espressif/esp-idf #12753: esp-idf release/v5.1 rmt signal_range_max_ns not set more than 40ms

WOLEN6914183, 2023-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

use esp-idf release/v5.1

"I have a project that requires receiving infrared signals from a remote control. Some remote controls send signals that consist of an initial data segment followed by a 70ms high-level time and then the remaining data. Clearly, if I use the RMT (Remote Control) module to receive this infrared signal, the signal_range_max_ns parameter needs to be set to a value greater than or equal to 70ms. However, in practice, I found that when I set it to a value exceeding 33ms, the underlying system indicates that the set value is too large. I noticed that the unit for signal_range_max_ns in the component is in nanoseconds (ns), as a uint32_t, and it should theoretically support values up to around 4 seconds. Why does it show an error for values exceeding 33ms? Is there a solution, or is it limited to 33ms?"

Here is rmt componts about signal_range_max_ns set
 uint32_t filter_reg_value = ((uint64_t)group->resolution_hz * config->signal_range_min_ns) / 1000000000UL;
    uint32_t idle_reg_value = ((uint64_t)channel->resolution_hz * config->signal_range_max_ns) / 1000000000UL;
    printf("idle_reg_value%ld\n",idle_reg_value);
    ESP_RETURN_ON_FALSE_ISR(filter_reg_value <= RMT_LL_MAX_FILTER_VALUE, ESP_ERR_INVALID_ARG, TAG, "signal_range_min_ns too big");
    ESP_RETURN_ON_FALSE_ISR(idle_reg_value <= RMT_LL_MAX_IDLE_VALUE, ESP_ERR_INVALID_ARG, TAG, "signal_range_max_ns too big");

suda-morris, 2023-12-08:
@WOLEN6914183  as you can see from the formula `uint32_t idle_reg_value = ((uint64_t)channel->resolution_hz * config->signal_range_max_ns) / 1000000000UL;`

if you wants to achieve a bigger `signal_range_max_ns` while keeps the `idle_reg_value` within a acceptable range, you should decrease the `channel->resolution_hz`.

The channel resolution is set by `rmt_rx_channel_config_t::resolution_hz`.

WOLEN6914183, 2023-12-08:
ok, that’s it. I can indeed achieve greater delay by following your tips. Maybe I misunderstood the role of idle_reg_value. Now my problem is solved, thank you very much, thank you

dpnebert, 2024-10-21:
I know this is closed, but I had to set aside my RMT project for a few months and coming back to it, I have relearn what are the maximum values for each 32 bit register.  From the [API reference](https://docs.espressif.com/projects/esp-idf/en/v5.2.3/esp32s3/api-reference/peripherals/rmt.html#_CPPv420rmt_receive_config_t), it does not mention anything about being capped at 8-bits.  And not sure why the variable declaration is uint32_t instead of uint8_t.  I can see why @suda-morris and I thought we'd be able to put 0xFFFFFFFF for that value.