espressif/esp-idf #12754: ble_hid_device_demo Consumer Devices的多媒体按键问题

yzjie, 2023-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

准备制作一个USB HID到bluetooth ble hid的桥接转换，在ble_hid_device_demo上进行修改测试。
对于HID Usage Tables Version 1.12，15 Consumer Page (0x0C)这个表内容的处理，通过hid_consumer_build_report()函数进行了转换，我想知道这个转换是必须的么，如果我将hidReportMap相关内容修改如下

原内容：
```
    0x05, 0x0C,   // Usage Pg (Consumer Devices)
    0x09, 0x01,   // Usage (Consumer Control)
    0xA1, 0x01,   // Collection (Application)
    0x85, 0x03,   // Report Id (3)
    0x09, 0x02,   //   Usage (Numeric Key Pad)
    0xA1, 0x02,   //   Collection (Logical)
    0x05, 0x09,   //     Usage Pg (Button)
    0x19, 0x01,   //     Usage Min (Button 1)
    0x29, 0x0A,   //     Usage Max (Button 10)
    0x15, 0x01,   //     Logical Min (1)
    0x25, 0x0A,   //     Logical Max (10)
    0x75, 0x04,   //     Report Size (4)
    0x95, 0x01,   //     Report Count (1)
    0x81, 0x00,   //     Input (Data, Ary, Abs)
    0xC0,         //   End Collection
    0x05, 0x0C,   //   Usage Pg (Consumer Devices)
    0x09, 0x86,   //   Usage (Channel)
    0x15, 0xFF,   //   Logical Min (-1)
    0x25, 0x01,   //   Logical Max (1)
    0x75, 0x02,   //   Report Size (2)
    0x95, 0x01,   //   Report Count (1)
    0x81, 0x46,   //   Input (Data, Var, Rel, Null)
    0x09, 0xE9,   //   Usage (Volume Up)
    0x09, 0xEA,   //   Usage (Volume Down)
    0x15, 0x00,   //   Logical Min (0)
    0x75, 0x01,   //   Report Size (1)
    0x95, 0x02,   //   Report Count (2)
    0x81, 0x02,   //   Input (Data, Var, Abs)
    0x09, 0xE2,   //   Usage (Mute)
    0x09, 0x30,   //   Usage (Power)
    0x09, 0x83,   //   Usage (Recall Last)
    0x09, 0x81,   //   Usage (Assign Selection)
    0x09, 0xB0,   //   Usage (Play)
    0x09, 0xB1,   //   Usage (Pause)
    0x09, 0xB2,   //   Usage (Record)
    0x09, 0xB3,   //   Usage (Fast Forward)
    0x09, 0xB4,   //   Usage (Rewind)
    0x09, 0xB5,   //   Usage (Scan Next)
    0x09, 0xB6,   //   Usage (Scan Prev)
    0x09, 0xB7,   //   Usage (Stop)
    0x15, 0x01,   //   Logical Min (1)
    0x25, 0x0C,   //   Logical Max (12)
    0x75, 0x04,   //   Report Size (4)
    0x95, 0x01,   //   Report Count (1)
    0x81, 0x00,   //   Input (Data, Ary, Abs)
    0x09, 0x80,   //   Usage (Selection)
    0xA1, 0x02,   //   Collection (Logical)
    0x05, 0x09,   //     Usage Pg (Button)
    0x19, 0x01,   //     Usage Min (Button 1)
    0x29, 0x03,   //     Usage Max (Button 3)
    0x15, 0x01,   //     Logical Min (1)
    0x25, 0x03,   //     Logical Max (3)
    0x75, 0x02,   //     Report Size (2)
    0x81, 0x00,   //     Input (Data, Ary, Abs)
    0xC0,           //   End Collection
    0x81, 0x03,   //   Input (Const, Var, Abs)
    0xC0,            // End Collectionq
```
修改内容：
```
    0x05, 0x0C,             //  Usage Page (Consumer Devices)
    0x09, 0x01,             //  Usage (Consumer Control)
    0xA1, 0x01,             //  Collection (Application)
    0x85, 0x03,             //      Report ID
    0x15, 0x00,             //      Logical Minimum (0)
    0x26, 0x9C, 0x02,       //      Logical Maximum (668)
    0x19, 0x00,             //      Usage Minimum (Undefined)
    0x2A, 0x9C, 0x02,       //      Usage Maximum (AC Distribute Vertically)
    0x75, 0x10,             //      Report Size (16)
    0x95, 0x01,             //      Report Count (1)
    0x81, 0x00,             //      Input (Data,Ary,Abs)
    0xC0                    //  End Collection   
```
测试结果不对，蓝牙对于Consumer Devices如此处理的原因是什么？

weiyuhannnn, 2023-12-11:
@yzjie Hello, Thanks for your question. May I know which Espressif chip are you using? Thanks!

yzjie, 2023-12-15:
我这里把 Logical Minimum修改为2，hid_dev_send_report发送的数据减1，就正常了。
使用的过程有问题。

closed 2023-12-15 by yzjie