espressif/esp-idf #12757: when device power on, occure repeated restart

luacerbb, 2023-12-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf 4.4.4

### Espressif SoC revision.

ESP32-WROOM-32D

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

work proper.

### What is the actual behavior?

when power on, occure repeated restart.

### Steps to reproduce.

1、device power on
2、system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x02; calc_crc = 0x45 occure abort, and reboot repeted,can't restore right.  
3、The device recovers after being powered off and then power on. 

### Debug Logs.

```plain
I (29) boot: compile time 15:40:27
I (29) boot: chip revision: v1.1
I (30) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (31) boot.esp32: SPI Speed      : 40MHz
I (33) boot.esp32: SPI Mode       : QOUT
I (34) boot.esp32: SPI Flash Size : 4MB
I (35) boot: Enabling RNG early entropy source...
I (36) boot: Partition Table:
I (37) boot: ## Label            Usage          Type ST Offset   Length
I (39) boot:  0 nvs              WiFi data        01 02 0000b000 0000f000
I (41) boot:  1 otadata          OTA data         01 00 0001a000 00002000
I (43) boot:  2 ota_0            OTA app          00 10 00020000 001ce000
I (45) boot:  3 ota_1            OTA app          00 11 001f0000 001ce000
I (46) boot:  4 storage          unknown          40 06 003f4000 00008000
I (48) boot: End of partition table
I (49) boot_comm: chip revision: 1, min. application chip revision: 0
I (51) esp_image: segment 0: paddr=001f0020 vaddr=3f400020 size=310ech (200940) map
I (119) esp_image: segment 1: paddr=00221114 vaddr=3ffbdb60 size=05f00h ( 24320) load
I (127) esp_image: segment 2: paddr=0022701c vaddr=40080000 size=08ffch ( 36860) load
I (141) esp_image: segment 3: paddr=00230020 vaddr=400d0020 size=f0290h (983696) map
I (461) esp_image: segment 4: paddr=003202b8 vaddr=40088ffc size=145cch ( 83404) load
I (492) esp_image: segment 5: paddr=0033488c vaddr=00000000 size=0b6f4h ( 46836) 
I (522) boot: Loaded app from partition at offset 0x1f0000
I (522) boot: Disabling RNG early entropy source...
silicon revision 1,rst reason:4
nvs use:4.36 kb
 [inf] _log_upload_task(215): enter
E (3419) system_api: Base MAC address from BLK0 of EFUSE CRC error, efuse_crc = 0x02; calc_crc = 0x45

abort() was called at PC 0x400d7c5f on core 0

Backtrace: 0x40081c82:0x3ffd7630 0x40091371:0x3ffd7650 0x400979be:0x3ffd7670 0x400d7c5f:0x3ffd76e0 0x400d7ca1:0x3ffd7720 0x40176f30:0x3ffd7750 0x40175804:0x3ffd7770 0x40099331:0x3ffd7790

ELF file SHA256: a3790294b157b399
```

### More Information.

I want to know what causes this situation and how to solve it. thanks

KonstantinKondrashov, 2023-12-08:
Hi @luacerbb!
You can enable [CONFIG_ESP_MAC_IGNORE_MAC_CRC_ERROR ](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/mac_addr.c#L140C8-L144) option to skip this error.

luacerbb, 2023-12-11:
> Hi @luacerbb! You can enable [CONFIG_ESP_MAC_IGNORE_MAC_CRC_ERROR ](https://github.com/espressif/esp-idf/blob/release/v4.4/components/esp_hw_support/mac_addr.c#L140C8-L144) option to skip this error. It should not be changed, it should be constant, may be a hardware issue.

thanks, flash  spi mode set to QOUT mode,I don't know if that has anything to do with it.In addition to CRC errors, I am afraid there will be other problems, after all, we do not know the real cause

KonstantinKondrashov, 2023-12-11:
You can run the cmd `espefuse.py summary` to look at the MAC and CRC_MAC and compare them with the log.
The flash spi mode does not matter in this case.

closed 2023-12-18 by luacerbb