espressif/esp-idf #12760: idv version tagged release/v4.3 does no more compile since ~4th of December

iltis42, 2023-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v4.3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Docker build:   "docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.3 idf.py build" runs well,

e.g. here: 

https://github.com/iltis42/XCVario/actions/runs/7072852197

### What is the actual behavior?

Docker build:   "docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.3 idf.py build" **fails** with release/v4.3 since a few days, e.g. here:  

https://github.com/iltis42/XCVario/actions/runs/7134914257

Detailed error:

[499/1496] Building C object esp-idf/driver/CMakeFiles/__idf_driver.dir/twai.c.obj
FAILED: esp-idf/driver/CMakeFiles/__idf_driver.dir/twai.c.obj 
...
In file included from /opt/esp/idf/components/driver/twai.c:22:
/opt/esp/idf/components/driver/twai.c: In function 'twai_driver_install':
/opt/esp/idf/components/driver/twai.c:450:18: error: 'TWAI_TAG' undeclared (first use in this function); did you mean 'TWAI'?
         ESP_LOGE(TWAI_TAG, "init gpio for twai failed");

### Steps to reproduce.

docker run --rm -v $PWD:/project -w /project espressif/idf:release-v4.3 idf.py build

e.g. in a github workflow will fail.

### Build or installation Logs.

```plain
https://github.com/iltis42/XCVario/actions/runs/7072852197
```

### More Information.

Its confirmed by other users of my project

moshe-braner, 2023-12-09:
Yes this was my experience too, using the same .yml on several versions of the project.  Same error message.

AxelLin, 2023-12-23:
@suda-morris Just remind v4.3 still needs fix.

Alvin1Zhang, 2024-02-19:
Thanks for reporting, fix is available on release/4.3, https://github.com/espressif/esp-idf/commit/878fd239d6d2658261e452eb6bb94508d3fc0342, feel free to reopen.

closed 2024-02-19 by Alvin1Zhang