espressif/esp-idf #12762: sdkkonfig broken after update to IDF 5.1.2

Vincenzo1992, 2023-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

To have possibility of editing sdkconfig as in previous version/ before updating to the current.

### What is the actual behavior?

Even in HelloWorld example the sdkconfig can't be executed anymore (i worked with the same projects on previous version of idf for longer time) since the update any execution of sdkconfig results in the following in the "SDK configuration":
[image]

in the console:
[image]

Any time in the past i have met with issues of being unable to build or execute projects i would always in the end reinstall the whole thing but this time even downgrading back ESP IDE with esp idf 5.1.1 isn't helping anymore.

### Steps to reproduce.

1. Update ESP IDF version from 5.1.1 inside the Espressif IDE to version 5.1.2
2. Create new project `based on example delete build folder
3. Open sdkconfig

### Build or installation Logs.

```plain
IDF_PATH: C:/Espressif/frameworks/esp-idf-v5.1.2/
PROCESSOR_REVISION: 9a03
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
LIBCLANG_PATH: C:\Users\zapot\.rustup\toolchains\esp\xtensa-esp32-elf-clang\esp-clang\bin\libclang.dll
TEMP: C:\Users\zapot\AppData\Local\Temp
SystemRoot: C:\Windows
HOMEDRIVE: C:
USERPROFILE: C:\Users\zapot
TMP: C:\Users\zapot\AppData\Local\Temp
EFC_7660: 1
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 20
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.1_py3.11_env
ESP_ROM_ELF_DIR: C:\Espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 11
Java Runtime Version: 17.0.6+10-LTS
Eclipse Version: 4.29.0.v20230903-1000
Eclipse CDT Version: 11.3.0.202307110141
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v5.1.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.2

Executing action: confserver
Running ninja in directory C:\Users\zapot\workspace\hello_world\build
Executing "ninja confserver"...
[0/1] cmd.exe /C "cd /D C:\Users\zapot\workspace\hello_world\build && python C:/Espressif/frameworks/esp-idf-v5.1.2/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file C:/Users/zapot/workspace/hello_world/build/config.env && python -m kconfserver --env-file C:/Users/zapot/workspace/hello_world/build/config.env --kconfig C:/Espressif/frameworks/esp-idf-v5.1.2/Kconfig --sdkconfig-rename C:/Espressif/frameworks/esp-idf-v5.1.2/sdkconfig.rename --config C:/Users/zapot/workspace/hello_world/sdkconfig"

'python' is not recognized as an internal or external command,

operable program or batch file.

FAILED: CMakeFiles/confserver C:/Users/zapot/workspace/hello_world/build/CMakeFiles/confserver 

cmd.exe /C "cd /D C:\Users\zapot\workspace\hello_world\build && python C:/Espressif/frameworks/esp-idf-v5.1.2/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file C:/Users/zapot/workspace/hello_world/build/config.env && python -m kconfserver --env-file C:/Users/zapot/workspace/hello_world/build/config.env --kconfig C:/Espressif/frameworks/esp-idf-v5.1.2/Kconfig --sdkconfig-rename C:/Espressif/frameworks/esp-idf-v5.1.2/sdkconfig.rename --config C:/Users/zapot/workspace/hello_world/sdkconfig"

ninja: build stopped: subcommand failed.

ninja failed with exit code 1, output of the command is in the C:\Users\zapot\workspace\hello_world\build\log\idf_py_stderr_output_10468 and C:\Users\zapot\workspace\hello_world\build\log\idf_py_stdout_output_10468
```

ralphde, 2023-12-13:
When I have this issue I delete the build folder, restart Eclipse and build. After sdkconfig server app works usually. That is why I never reported this issue but I have seen it since I work with Eclipse IDF plugin (2 years)

Vincenzo1992, 2023-12-14:
> When I have this issue I delete the build folder, restart Eclipse and build. After sdkconfig server app works usually. That is why I never reported this issue but I have seen it since I work with Eclipse IDF plugin (2 years)

Yeah tried those things not what you are describing

Vincenzo1992, 2023-12-14:
> Hi @Vincenzo1992,
> 
> thank you for reporting this. We were able to find a problem with opening skdconfig via the "ESP-IDF: Load skdconfig" menu command, but opening sdkconfig with double-click works fine for me and for my colleague, who was also trying to reproduce this issue following the steps that you provided.
> 
> I have noticed in the log, that Python is not recognized during the execution of "ninja confserver", that's strange because we are passing the absolute path for Python from the IDF_PYTHON_ENV variable. From the product information log you provided, this variable is ok, so we are still looking for what could cause this issue. To help us investigate further, could you please provide the full product information log? The one that you provided seems to cut off, it misses some variables like PATH, etc. Thank you

I will paste more as soon as I get to the pc that has this issue it made me move to other machine that I haven't updated yet, in the end it looks like only sdkconfig tool is looking for python in the wrong place

ralphde, 2023-12-14:
Have you tried to reinstall the tools again?
What does the menu 'product information' dump out?

Indeed I do not recall very good what I did to solve the problem, but I
have it pretty regularly while migrating IDF

Best Regards

Mit freundlicher Empfehlung

Ralph Deffke

Vincenzo1992, 2023-12-14:
> Have you tried to reinstall the tools again? What does the menu 'product information' dump out? Indeed I do not recall very good what I did to solve the problem, but I have it pretty regularly while [...]

Yes yes been there done that many times before for few years

Vincenzo1992, 2023-12-14:
> Hi @Vincenzo1992,
>
> thank you for reporting this. We were able to find a problem with opening skdconfig via the "ESP-IDF: Load skdconfig" menu command, but opening sdkconfig with double-click works [...]

[image]

Vincenzo1992, 2023-12-25:
> Hi @Vincenzo1992,
>
> thank you for reporting this. We were able to find a problem with opening skdconfig via the "ESP-IDF: Load skdconfig" menu command, but opening sdkconfig with double-click works [...]

Hey, are there any news in here?, if you need we can even do a teamviewer if it helps you to find out what is missing.

sigmaaa, 2023-12-26:
Hi @Vincenzo1992,

It would be very helpful to have a call with you since this issue is hard to reproduce for us. Could you please contact us via ide.support@espressif.com and mention the time zone and preferable time for the meeting? After that, we will schedule the meeting via Microsoft Teams and send you an invite. Thank you for your help in resolving this