espressif/esp-idf #12763: ANY build for ESP32-s2  for board =  lolin_s2_mini fails with USBSerial not declared in scope of HardwareSerial.h

peteDDD, 2023-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

framework-espidf @ 3.50102.0 (5.1.2)

### Espressif SoC revision.

ESP32-S2  lolin_s2_mini

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2  lolin_s2_mini

### Power Supply used.

USB

### What is the expected behavior?

printf and ESP_LOG should output to serial port (usb programming connector)

### What is the actual behavior?

ANY code built with espidf @ 3.50102.0 (5.1.2)/arduinoespressif32 @ 3.0.0 specifying  lolin_s2_mini as the board and with Arduino Setup()/loop() mode will not build.   (see most minimal examples in Steps to reproduce below and in More Information, below.  
**Build fails in HardwareSerial.h:360:16:  with error: 'USBSerial' was not declared in this scope**

### Steps to reproduce.

platformio:
```
[env:lolin_s2_mini]
platform = espressif32
board = lolin_s2_mini
framework = espidf, arduino
```
Code:
```
#include <Arduino.h>
#include <stdio.h>
#include "sdkconfig.h"

void setup()
{
printf("Hello World\n");
}

void loop()
{}
```
sdkconfig includes:
`CONFIG_AUTOSTART_ARDUINO=y`

### Debug Logs.

```plain
In file included from C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/Arduino.h:196,
                 from C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp:17:
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp: In function 'void printBeforeSetupInfo()':     
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/HardwareSerial.h:360:16: error: 'USBSerial' was not declared in this scope; did you mean 'Serial'?
  360 | #define Serial USBSerial
      |                ^~~~~~~~~
C:/Users/peted/.platformio/packages/framework-arduinoespressif32-src-16f16f9b46f255eba682607449bb3231/cores/esp32/chip-debug-report.cpp:260:3: note: in expansion of macro 'Serial'
```

### More Information.

1: Using VS Code/Platformio (see paltformio.ini abov.  The esp-idf and arduinoespressif32 are the latest available as of 13:00 UTC  on 9 December 2023
2: Errors shown above occur when board = lolin-s2_mini.  However, does not occur is board = esp32-s3-devkitc-1, for example
3: Even the following code will produce the same errors when the board = lolin_s2_mini

```
#include <Arduino.h>

void setup()
{
}

void loop()
{}
```

atanisoft, 2023-12-09:
most likely this should be sent over to https://github.com/espressif/arduino-esp32 since it looks likely to be an issue with PlatformIO and arduino dependencies.

kutukvpavel, 2023-12-09:
I agree with @atanisoft, USBSerial as well as HardwareSerial.h are a part of Arduino Core. Unrelated to ESP-IDF.

peteDDD, 2023-12-09:
Thanks for that link.  I will indeed take it up with them.  (If this goes like the rest of that release, it may be a while...