espressif/esp-idf #12764: gdbgui not starting

urbu, 2023-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.3-dev-888-gcb66d6f6fa

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM-2

### Power Supply used.

USB

### What is the expected behavior?

gdbgui starts in browser

### What is the actual behavior?

gdbgui fails to start
```
Executing action: gdbgui
['gdbgui', '-g', 'xtensa-esp32s3-elf-gdb', '--gdb-args', '"-x=/home/urbu/projects/esp/esp-idf/examples/get-started/hello_world/build/gdbinit/gdbinit"']
gdbgui started as a background task 20528
Executing action: post_debug
usage: gdbgui [-h] [-g GDB_CMD] [-p PORT] [--host HOST] [-r]
              [--auth-file AUTH_FILE] [--user USER] [--password PASSWORD]
              [--key KEY] [--cert CERT] [--remap-sources REMAP_SOURCES]
              [--project PROJECT] [-v] [-n] [-b BROWSER] [--debug]
              [--args ...]
              [debug_program]
gdbgui: error: unrecognized arguments: --gdb-args
"gdbgui" exited with 2

```

### Steps to reproduce.

flash example code to esp
idf.py openocd
idf.py gdbgui (in new terminal tab, run from project folder)

### More Information.

I also tried the idf.py openocd gdbgui command.
There gdbgui doesn't crash but is run as a background task. Still no browser window is opend and if I check netstat. There are no ports other than openocd available on localhost. So It looks like gdbgui is not running
```
Executing action: openocd
Note: OpenOCD cfg not found (via env variable OPENOCD_COMMANDS nor as a --openocd-commands argument)
OpenOCD arguments default to: "-f board/esp32s3-builtin.cfg"
OpenOCD started as a background task 29803
Executing action: gdbgui
['gdbgui', '-g', 'xtensa-esp32s3-elf-gdb', '--gdb-args', '"-x=/home/urbu/projects/esp/esp-idf/examples/get-started/hello_world/build/gdbinit/gdbinit"']
gdbgui started as a background task 29845
Executing action: post_debug
Open On-Chip Debugger v0.12.0-esp32-20230921 (2023-09-21-13:41)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselecting 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (68:B6:B3:3E:0E:2C)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40378326, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu1] Target halted, PC=0x40378326, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)
```

dobairoland, 2023-12-11:
Hi @urbu. Can you please check the output of `gdbgui --version`. Only version 0.13.2.0 is supported through `idf.py gdbgui`. The right version should be available with all dependencies if you are in an ESP-IDF environment (created by `install.sh` and set-up by `. export.sh`).

If you want to use a different version of gdbgui then you can but you'll need to call it directly. I.E. you can still call `idf.py openocd` but you'll have to run `gdbgui` with all the arguments.

Please note that gdbgui's dependencies might cause that the debugging session is not opening in the browser. 

Here are our tested environment available in the ESP-IDF environments (unfortunately you need an older Python for this):
```
gdbgui==0.13.2.0; python_version < "3.11"
pygdbmi<=0.9.0.2; python_version < "3.11"
python-socketio<5; python_version < "3.11"
jinja2<3.1; python_version < "3.11"
itsdangerous<2.1; python_version < "3.11"
```

urbu, 2023-12-13:
Thank you very much for the help. That was it.
The main problem was that i was using python 3.11. So when I ran install.sh --enable gdbgui. Gdbgui was not installed because of the python version. But I did not notice the message in the console output. I taught I have to install gdbgui manually and when I did that I installed version 0.15.

For anybody else having similar problems. To fix everything I first uninstalled gdbgui. Than I installed python 3.10 next to python 3.11.
Now for esp-idf to use python 3.10 I had to remove the esp-idf folder (don't know how to undo the install.sh which creates a python environment). 
Then I got a fresh version of esp-idf from git and fixed the tools/detect_python.sh script:
from this: `for p_cmd in python3 python python3.8 python3.9 python3.10 python3.11 python3.12; do`
to this: `for p_cmd in python3.8 python3.9 python3.10 python3.11 python3.12; do`
That way the script doesn't find python 3.11 which is invoked with python3. But does find python3.11. 
After that run `install.sh --enable gdbgui `and `. ./export.sh` and everything worked like it should.

dobairoland, 2023-12-14:
You did choose a rather complicated solution but I'm happy that you resolved it.

dobairoland, 2024-01-04:
I've prepared a fix which will fix gdbgui support with Python 3.11+ on Linux and macOS.

closed 2024-01-08 by espressif-bot