espressif/esp-idf #12765: [ESP32-S3] ADC1 doesn't work under ULP-RISCV control in Light Sleep with XTAL_32K used as RTC clock

kutukvpavel, 2023-12-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3 rev 0.1 (ESP32-S3-WROOM-1 MON4R2, P1N8R8)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board, ESP32-S3-DevKitC-1 v1.6

### Power Supply used.

External 3.3V

### What is the expected behavior?

ADC1 should sample selected channel under ULP-RISCV control (`ulp_riscv_adc_read_channel`) regardless of sleep mode and RTC clock source.

### What is the actual behavior?

ADC1 works fine w/ULP-RISCV only in following "sleep mode + RTC clock source" combinations:
- Deep sleep + External XTAL_32K
- Deep sleep + Internal RC-oscillator 136kHz
- Light sleep + Internal RC-oscillator 136kHz

It fails when "Light Sleep + External XTAL_32K" combination is used. In this case, ADC1 samples input channel only once. Supposedly, it does so during the first `ulp_riscv_adc_read_channel` call since boot, see below, and any subsequent `ulp_riscv_adc_read_channel` calls return the exact same value, no matter what the actual voltage on the input pin is.

Confirmed for all devices listed above. 

### Steps to reproduce.

Easiest MWE:
1. Equip any ESP32-S3 board with an external 32.768kHz crystal with appropriate loading capacitors (XTAL_32K).
2. Build and flash ulp_riscv_adc example with RTC clock = XTAL_32K selected in menuconfig.
3. Sweep input voltage from 0 up to ~1.6-1.7V, observe expected operation (ULP wakes the SoC up)
4. Reset input voltage back to 0.
5. In ulp_riscv_adc_example_main.c change sleep mode to **Light** (replace esp_deep_sleep_start() with esp_light_sleep_start() ), build and flash.
6. Sweep input voltage from 0, through ~1.7V and all the way to VCC, observe no reaction (the ULP never wakes the SoC up, since the ADC "latched-up" the first result, which was close to 0).

Better example:
1. Equip any ESP32-S3 board with an external 32.768kHz crystal with appropriate loading capacitors (XTAL_32K).
2. Flash the following adapted ulp_riscv_adc example code (adapted to perform periodic ADC measurement output to debug monitor), with "RTC clock = Internal RC 136kHz" set in menuconfig:

**ulp_riscv_adc_example_main.c**
```
#include <stdio.h>
#include <inttypes.h>
#include "esp_sleep.h"
#include "ulp_riscv.h"
#include "ulp_adc.h"
#include "esp_adc/adc_oneshot.h"
#include "ulp_main.h"
#include "ulp/example_config.h"

extern const uint8_t ulp_main_bin_start[] asm("_binary_ulp_main_bin_start");
extern const uint8_t ulp_main_bin_end[]   asm("_binary_ulp_main_bin_end");

static adc_oneshot_unit_handle_t adc_handle;

static void init_ulp_program(void);

void app_main(void)
{
    ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON)); //So that ULP can use SAR ADC
    //ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RC_FAST, ESP_PD_OPTION_ON));
    init_ulp_program();
    ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());
    while (1)
    {
        ESP_ERROR_CHECK(ulp_riscv_run());
        esp_light_sleep_start();
        printf("ULP-RISC-V woke up the main CPU\n");
        printf("Value = %"PRIu32"\n", ulp_wakeup_result);
    }
}

static void init_ulp_program(void)
{
    ulp_adc_cfg_t cfg = {
        .adc_n    = EXAMPLE_ADC_UNIT,
        .channel  = EXAMPLE_ADC_CHANNEL,
        .width    = EXAMPLE_ADC_WIDTH,
        .atten    = EXAMPLE_ADC_ATTEN,
        .ulp_mode = ADC_ULP_MODE_RISCV,
    };

    ESP_ERROR_CHECK(ulp_adc_init(&cfg));

    esp_err_t err = ulp_riscv_load_binary(ulp_main_bin_start, (ulp_main_bin_end - ulp_main_bin_start));
    ESP_ERROR_CHECK(err);

    ulp_set_wakeup_period(0, 20000);
}
```

**main.c (ULP program)**
```
#include <stdint.h>
#include "ulp_riscv_utils.h"
#include "ulp_riscv_adc_ulp_core.h"

#include "example_config.h"

#define ULP_TIMER_PERIOD 20u //mS
#define WAKEUP_PERIOD 1000u //mS

volatile int32_t wakeup_result;

int main (void)
{
    static uint32_t cycles = 0;

    wakeup_result = ulp_riscv_adc_read_channel(EXAMPLE_ADC_UNIT, EXAMPLE_ADC_CHANNEL);

    if (++cycles > (WAKEUP_PERIOD / ULP_TIMER_PERIOD))
    {
        cycles = 0;
        ulp_riscv_wakeup_main_processor();
    }

    return 0;
}
```

**example_config.h**
Stays the same.

3. Observe normal operation (ADC readings correspond to input voltage, albeit with some delay).
4. In menuconfig set "RTC clock = External XTAL_32K", build and flash.
5. Observe the output listed in the next section.

### Debug Logs.

```plain
<Bootloader log omitted>
entry 0x403c9894
ULP-RISC-V woke up the main CPU
Value = 1867
ULP-RISC-V woke up the main CPU
Value = 1867
<Further output remains the same despite any changes of the input voltage>
```
The exact ADC code "latched" is kind of random, for 0.5V of input voltage I managed to score anything from 1300 to 2200 by resetting the SoC several times. The input is connected directly to an adjustable bench PSU (a low impedance source). Therefore, I am not sure whether this code is, in fact, a real measurement, or just some artifact.
```

### More Information.

To be honest, I'm having difficulties tracking down the cause of this behavior, mainly because TRM does not depict low-power clock paths clearly. For example, regarding ADC clock source, there is no mention of ADC1 RTC controller being able to use RTC_SLOW_CLK. I believe, I've seen somewhere through the manuals that it's clocked from 17.5M internal RC oscillator, so I tried to leave corresponding power domain ON during sleep (this line is commented-out in the code I listed above):

`ESP_ERROR_CHECK(esp_sleep_pd_config(ESP_PD_DOMAIN_RC_FAST, ESP_PD_OPTION_ON));`,

however, it made no difference. Looking inside IDF sources provided no insight as well, because enum `soc_periph_adc_rtc_clk_src_t` contains only single entry (`ADC_RTC_CLK_SRC_RC_FAST` - the 17.5M RC oscillator) for ESP32-S3.

I imagine that for the 32kHz clock to drive peripherals usually clocked by RC 17.5M clock there has to be a PLL, or am I wrong? How exactly are ULP-RISCV and ADC1 clocked during sleep modes with external XTAL_32K?

kutukvpavel, 2023-12-09:
Now, here's something interesting. If I set Debug Log Verbosity to "Debug" (used to be "Warning") in menuconfig the problem goes away.

kutukvpavel, 2023-12-10:
Well, I have no idea why, for example, placing `periph_ll_get_rst_en_mask(...)` in `pragma GCC optimize ("O0")` helps with this issue, or why optimization level influences program behavior in this case at all.
However, the root cause seems to be the same as in #12426. Moving discussion over there.

closed 2023-12-10 by kutukvpavel

KonstantinKondrashov, 2023-12-12:
Hi @kutukvpavel!
I want to clarify that you use the `External 32 kHz crystal`, not `External 32 kHz oscillator at XTAL_32K_P pin` option. And the 32 kHz crystal is connected to the right pins (XTAL_32K_P and XTAL_32K_N).
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/system_time.html#rtc-timer-clock-sources

You have not mentioned which example you used, probably based on this one [ulp_riscv adc](https://github.com/espressif/esp-idf/tree/master/examples/system/ulp/ulp_riscv/adc).

Do you observe the same issue on the master too? Could you try it?
I do not have a 32k Xtal near me for now to test it, I will ask somebody to try it.

You closed this issue, why? Is your guess about the relation to the optimization option not confirmed?

kutukvpavel, 2023-12-17:
@KonstantinKondrashov Sorry for the delay, a little busy ATM.

First of all, I indeed use an external crystal (resonator), not an oscillator (both in menuconfig and in hardware). It is connected as per Hardware Design Guidelines.

As for the example, I started with the one you correctly deduced, then modified it as describe above, for it to use Light Sleep instead of Deep Sleep (but keep as much initialization code as possible the same), because the issue only affects Light Sleep (for me).

I'm afraid I did not test master, but I'll try to do so ASAP.

I closed this one as a duplicate, since A) the solution proposed over at #12426 solved this issue for me and B) I tried to disable all optimizations in menuconfig for the example provided over at #12426 and it also helped. Therefore I concluded that my issue is just a different manifestation of the same bug under slightly different conditions. Which is expected for UB- and optmization-related issues.