espressif/esp-idf #12766: [ESP32-S3] ulp_adc_init(...) leaks ADC1 handle, preventing ADC1 reinitialization upon wakeup from Light Sleep

kutukvpavel, 2023-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3 rev 0.1 (ESP32-S3-WROOM-1 MON4R2, P1N8R8)

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board, ESP32-S3-DevKitC-1 v1.6

### Power Supply used.

External 3.3V

### What is the expected behavior?

Let's say we use `ulp_adc_init(...)` API to initialize ADC for use with ULP-RISCV during Light Sleep. It should be possible to reconfigure ADC1 to work with main cores after wakeup from Light Sleep.

### What is the actual behavior?

After `ulp_adc_init()` is executed, it's no longer possible to reclaim ADC1 by any means other than a reset. It happens due to ADC handle leakage inside `ulp_adc_init()`:
```
esp_err_t ulp_adc_init(const ulp_adc_cfg_t *cfg)
{
    esp_err_t ret = ESP_OK;

    ESP_GOTO_ON_FALSE(cfg, ESP_ERR_INVALID_ARG, err, TAG, "cfg == NULL");
    ESP_GOTO_ON_FALSE(cfg->adc_n == ADC_UNIT_1, ESP_ERR_INVALID_ARG, err, TAG, "Only ADC_UNIT_1 is supported for now");

    //-------------ADC1 Init---------------//
    adc_oneshot_unit_handle_t adc1_handle;   <--- THIS

    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = cfg->adc_n,
        .ulp_mode = cfg->ulp_mode,
    };

    if (init_config1.ulp_mode == ADC_ULP_MODE_DISABLE) {
        /* Default to RISCV for backward compatibility */
        ESP_LOGI(TAG, "No ulp mode specified in cfg struct, default to riscv");
        init_config1.ulp_mode = ADC_ULP_MODE_RISCV;
    }

    ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));

    //-------------ADC1 Config---------------//
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = cfg->width,
        .atten = cfg->atten,
    };
    ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, cfg->channel, &config));

    //Calibrate the ADC
#if SOC_ADC_CALIBRATION_V1_SUPPORTED
    adc_set_hw_calibration_code(cfg->adc_n, cfg->atten);
#endif

err:
    return ret;
}
```
The unit handle should be returned to caller, or a `ulp_adc_deinit()` method should be provided.

### Steps to reproduce.

1. Use `adc_ulp_init(...)` to configure ADC1 for use with ULP prior to entering Light Sleep.
2. Enter Light Sleep, wake up.
3. Try to use regular `adc_oneshot_new_unit(...)` and get "ADC1 already in use" in debug logs.

### More Information.

It seems in latest IDF releases ULP-related API was designed with only Deep Sleep in mind, and Light Sleep was left out.

One could try to use `adc_oneshot_new_unit(...)` in the first place, since configuration structure passed to this function contains `ulp_mode` memeber that can be set to ULP_RISCV. And I did try it, however, it didn't work as expected. Namely, attenuation/calibration got messed up, and ADC codes returned to ULP-RISCV w/6dB attenuation ranged from ~2000 (at 0V) to 4095 (at ~1.1V), if memory serves.

KonstantinKondrashov, 2023-12-11:
Hi @kutukvpavel!
Thank you for the report. We will discuss it with the team internally and I will back.

KonstantinKondrashov, 2023-12-12:
Hi @kutukvpavel! If you want to contribute then it would be great, you can create a PR to fix it (having `ulp_adc_deinit()` sounds good to us). Thanks.

ESP-Marius, 2024-08-02:
Should have been fixed with https://github.com/espressif/esp-idf/commit/8596d46774d65171b4540288918037c43e3debb4

closed 2024-08-02 by ESP-Marius