espressif/esp-idf #12768: When the ssl_mutual_auth routine is frequently connected: Dynamic Impl: alloc(5347 bytes) failed

LHYYSY, 2023-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

1Successfully connected to Amazon.
2Sometimes the connection works, and sometimes this error occurs

### What is the actual behavior?

[image]

### Steps to reproduce.

Every time you reconnect, there's a chance

### Debug Logs.

```plain
2 heap szie:24148
 2 heap szie:29260
I (19522) aliyunMqtt: Other event id:7
 2 heap szie:16240
E (20302) Dynamic Impl: alloc(5347 bytes) failed
E (20303) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7F00
I (20304) esp-tls-mbedtls: Certificate verified.
E (20309) esp-tls: Failed to open new connection
E (20333) transport_base: Failed to open a new connection
E (20335) mqtt_client: Error transport connect
I (20335) aliyunMqtt: MQTT_EVENT_ERROR
E (20339) aliyunMqtt: Last error reported from esp-tls: 0x801a
E (20346) aliyunMqtt: Last error reported from tls stack: 0x7f00
I (20371) aliyunMqtt: Last errno string (Success)
I (20371) aliyunMqtt: MQTT_EVENT_DISCONNECTED
```

### More Information.

TLS maximum incoming fragment length:5120.

Harshal5, 2023-12-13:
Hello @LHYYSY,

Could you also share the project sdkconfig that you are using and some minimal code that reproduces the issue you are facing?

Thanks!

closed 2023-12-18 by LHYYSY

MrRedKite, 2024-02-12:
Helle @SeanYang-7 ,
I get the same error, could you please share the solution to this problem?
Thanks

passy1977, 2024-12-11:
Me too, same problem