espressif/esp-idf #12770: I2C Interrupt Code Not Working

zeynepdicle, 2023-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I want to read data through I2C Interrupt in a bare-metal environment.

### What is the actual behavior?

I'm unable to enter interrupt and read data 

### Steps to reproduce.

```
#include <stdio.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <driver/i2c.h>
#include "esp_log.h"
#include "esp_intr_alloc.h"

#include <soc/i2c_reg.h>
#include <esp32/rom/gpio.h>
#include <soc/gpio_sig_map.h>
#include <hal/i2c_ll.h>
#include <hal/i2c_hal.h>
#include <hal/gpio_hal.h>

#define I2C_MASTER_NUM I2C_NUM_0
#define I2C_MASTER_SCL_IO 2
#define I2C_MASTER_SDA_IO 8
#define I2C_MASTER_FREQ_HZ 100000
#define DEVICE_ADDRESS 0x03

static const char *TAG = "I2C_TASK";

uint8_t receivedData;

static void i2c_master_init()
{
    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = I2C_MASTER_SDA_IO,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_io_num = I2C_MASTER_SCL_IO,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = I2C_MASTER_FREQ_HZ,
        .clk_flags = 0,
    };
    i2c_param_config(I2C_MASTER_NUM, &conf);
    i2c_driver_install(I2C_MASTER_NUM, conf.mode, 0, 0, ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_IRAM);
}

static void i2c_master_read_data()
{
    i2c_cmd_handle_t cmd = i2c_cmd_link_create();
    i2c_master_start(cmd);
    i2c_master_write_byte(cmd, (DEVICE_ADDRESS << 1) | I2C_MASTER_READ, true);
    i2c_master_read_byte(cmd, &receivedData, I2C_MASTER_ACK);
    i2c_master_stop(cmd);
    esp_err_t ret = i2c_master_cmd_begin(I2C_NUM_0, cmd, pdMS_TO_TICKS(500));
    i2c_cmd_link_delete(cmd);
    if (ret == ESP_OK)
    {
        ESP_LOGI(TAG, "I2C data read successfully");
        ESP_LOGI(TAG, "The receive data is : %d\n", receivedData);
    }
    else
    {
        ESP_LOGE(TAG, "I2C data read failed.");
    }
    vTaskDelay(pdMS_TO_TICKS(500));
}

static void IRAM_ATTR i2c_master_isr(void *arg)
{
    ESP_LOGI(TAG, "interrupt");
    i2c_master_read_data();
}

void app_main()
{
    i2c_master_init();
    esp_err_t ret = esp_intr_alloc(i2c_periph_signal[I2C_MASTER_NUM].irq, ESP_INTR_FLAG_SHARED | ESP_INTR_FLAG_IRAM, i2c_master_isr, NULL, NULL);
    if (ret == ESP_OK)
    {
        ESP_LOGI(TAG, "esp_intr_alloc successfully");
    }
    else
    {
        ESP_LOGE(TAG, "esp_intr_alloc failed");
    }
}

```

### Debug Logs.

```plain
I (139) boot: Disabling RNG early entropy source...
I (151) cpu_start: Pro cpu up.
I (160) cpu_start: Pro cpu start user code
I (160) cpu_start: cpu freq: 160000000 Hz
I (160) cpu_start: Application information:
I (163) cpu_start: Project name:     main
I (168) cpu_start: App version:      1
I (172) cpu_start: Compile time:     Dec  6 2023 11:48:45
I (178) cpu_start: ELF file SHA256:  23b3c8adcf70e294...
I (184) cpu_start: ESP-IDF:          v5.0.1-dirty
I (189) cpu_start: Min chip rev:     v0.3
I (194) cpu_start: Max chip rev:     v0.99
I (199) cpu_start: Chip rev:         v0.4
I (204) heap_init: Initializing. RAM available for dynamic allocation:
I (211) heap_init: At 3FC8E060 len 0004E6B0 (313 KiB): DRAM
I (217) heap_init: At 3FCDC710 len 00002950 (10 KiB): STACK/DRAM
I (224) heap_init: At 50000020 len 00001FE0 (7 KiB): RTCRAM
I (231) spi_flash: detected chip: generic
I (235) spi_flash: flash io: dio
W (239) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (252) cpu_start: Starting scheduler.
I (257) I2C_TASK: esp_intr_alloc successfully
```

KonstantinKondrashov, 2023-12-11:
Hi @zeynepdicle!
The i2c_driver_install() has already allocated the i2c isr handler. You do not need to allocate it twice. Please have a look the driver code - https://github.com/espressif/esp-idf/blob/master/components/driver/i2c/i2c.c#L402.

```c
esp_err_t i2c_driver_install(
...
    //hook isr handler
    ret = esp_intr_alloc(i2c_periph_signal[i2c_num].irq, intr_alloc_flags,
                           i2c_isr_handler_default, p_i2c_obj[i2c_num],
                           &p_i2c_obj[i2c_num]->intr_handle);
```

I2C example - https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/i2c_simple/main/i2c_simple_main.c

zeynepdicle, 2023-12-12:
> Hi @zeynepdicle! The i2c_driver_install() has already allocated the i2c isr handler. You do not need to allocate it twice. Please have a look the driver code - [...]

Does this imply that ESP-IDF utilizes the I2C function through Interrupt Service Routines (ISRs)? Consequently, it appears unnecessary to employ explicit interrupt functions.

KonstantinKondrashov, 2023-12-12:
Hi @zeynepdicle! Yes, I2C driver uses ISR. As I understand, you need to use a callback:

I2C master callbacks
When an I2C master bus triggers an interrupt, a specific event will be generated and notify the CPU. If you have some functions that need to be called when those events occurred, you can hook your functions to the ISR (Interrupt Service Routine) by calling [i2c_master_register_event_callbacks()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv435i2c_master_register_event_callbacks23i2c_master_dev_handle_tPK28i2c_master_event_callbacks_tPv). Since the registered callback functions are called in the interrupt context, user should ensure the callback function doesn't attempt to block (e.g. by making sure that only FreeRTOS APIs with ISR suffix are called from within the function). The callback functions are required to return a boolean value, to tell the ISR whether a high priority task is woke up by it.

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#register-event-callbacks

zeynepdicle, 2023-12-12:
> Hi @zeynepdicle! Yes, I2C driver uses ISR. As I understand, you need to use a callback:
>
> I2C master callbacks When an I2C master bus triggers an interrupt, a specific event will be generated and [...]

Yes, I want to do this, but the document you sent me does not match the latest version. Some of the functions mentioned here appear to be undefined in the build I'm using (v15.1.2).

closed 2024-01-29 by zeynepdicle