espressif/esp-idf #12771: Cannot jump to application after DFU download

StevenMacias, 2023-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-796-g30870c819f

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

After flashing a DFU image I expect to jump into the application by using the `-R` flag.

### What is the actual behavior?

I cannot execute my application after downloading a DFU image. Only through RESET button.

### Steps to reproduce.

1. Make sure the board is not affected by [AR2022-004](https://www.espressif.com/sites/default/files/advisory_downloads/AR2022-004%20Security%20Advisory%20for%20USB_OTG%20%26%20USB_Serial_JTAG%20Download%20Functions%20of%20ESP32-S3%20Series%20Products%20EN.pdf)
2. Burn USB_PHY_SEL `espefuse.py --port /dev/ttyACM0 burn_efuse USB_PHY_SEL 1`
3. Copy blink example to local folder
4. Create DFU image `idf.py dfu`
5. Put board in Bootloader mode (BOOT0 pin)
6. Check that the board is in DFU mode `idf.py dfu-list`
7. Flash DFU image with DFU utils `sudo dfu-util -v -d 303a:9 -R -D /home/steven/Workspace/local/espressif/blink/build/dfu.bin`
8. My application never runs, stuck in DFU mode.

### Debug Logs.

```plain
[blink] sudo dfu-util -v -d 303a:9 -R -D /home/steven/Workspace/local/espressif/blink/build/dfu.bin
[sudo] password for steven:
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.26 (11724)
DFU suffix version 100
Opening DFU capable USB device...
Device ID 303a:0009
Run-Time device DFU version 0110
DFU attributes: (0x0f) bitCanDnload bitCanUpload bitManifestationTolerant bitWillDetach
Detach timeout 700 ms
Claiming USB DFU (Run-Time) Interface...
Setting Alternate Interface zero...
Determining device status...
DFU state(0) = appIDLE, status(0) = No error condition is present
Device really in Run-Time Mode, send DFU detach request...
dfu-util: error detaching
Device will detach and reattach...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Interface #0 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download	[=========================] 100%       276480 bytes
Download done.
Sent a total of 276480 bytes
DFU state(2) = dfuIDLE, status(0) = No error condition is present
Done!
dfu-util: can't detach
Resetting USB to switch back to Run-Time mode
```
```

### More Information.

My custom board will not have RESET nor BOOT0 buttons. I would expect that the -R flag resets the board and since the BOOT0 pin is not being pulled down, my application should run.

igrr, 2023-12-12:
Unfortunately (or by design), all chips from ESP32 series are only reading the values of the strapping pins on external (power-on) reset or on RTC WDT reset. So it is not easily possible to "jump out of" the download mode purely by software.

dobairoland, 2023-12-12:
Esptool also just prints a warning in USB OTG mode that if the chip was put into download mode via IO0 then it cannot be reseted.

StevenMacias, 2023-12-13:
@igrr @dobairoland Thanks for your quick replies. It is a requirement for my project that the firmware update is done through DFU and that there is no user interaction. 

We have been considering developing DFU on the application layer and then also expose CDC-ACM through the USB. In this way we could reset the board through the RTS control line. 

Does this seems feasible, am I missing some incompatibility or an easier way to achieve that goal? Thank you!

StevenMacias, 2024-02-09:
We will be using [esp_tinyuf2](https://components.espressif.com/components/espressif/esp_tinyuf2) since the board can reboot itself after copying the .uf2 firmware.

closed 2024-02-09 by StevenMacias