espressif/esp-idf #12773: ESP32-S3 - I2S during Light Sleep

timsterc, 2023-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have an application where I am recording from stereo PDM microphones for a short duration (~10sec) then storing this on an SD card, along with values from some environmental sensors.

Everything is working nicely, but I'd like to improve the power performance for instances where the device is running on battery power.

Most of the time in the app is spent waiting on the blocking i2s_channel_read function to return when the data is available in the DMA buffer. I'd love to use light sleep if it's an option.

I've read through the datasheets, but for the life of me can't tell whether both the I2S and DMA peripherals will remain active during light sleep, and if so, how to properly configure them so that they aren't clock gated.

Any insight would be greatly appreciated!

Thanks!
-Tim

timsterc, 2024-02-08:
@L-KAYA  Any thoughts?

SanZamoyski, 2024-04-02:
Any update here?

L-KAYA, 2024-05-21:
@timsterc @SanZamoyski  Oh I'm so sorry for missing your question, and unfortunately, I2S can't work during the light sleep. 

Generally, only LP peripherals are supported to run during the sleep (i.e., support to clocked by RTC clock source), currently I2S doesn't have such a LP peripheral (LP_I2S) except ESP32-P4.

If you want to save power consumption, maybe you can try to set `FREERTOS_USE_TICKLESS_IDLE` and `PM_ENABLE` in the menu config. So that you system will go into light sleep during the `IDLE` task and I2S will only prevent the light sleep when APB bus is under using. Hope it can help.

Thanks!

Alvin1Zhang, 2025-01-13:
Thanks for reporting, feel free to reopen with more updates.

closed 2025-01-13 by Alvin1Zhang