espressif/esp-idf #12778: How to get all String Descriptors for the USB Sample ?

diplfranzhoepfinger, 2023-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib

roma-jam, 2023-12-12:
Hi @diplfranzhoepfinger, 

Strings descriptors as "Mobile AT Interface", "Mobile RNDIS Network Adapter", "Mobile Diag Interface" and others are the additional String Descriptors describing the Interfaces .
These String Descriptors are not handled in the `usb_host_lib` example.

In `usb_host` library only the primary String Descriptors are requested from the device (which describes iManufacturer, iProduct, iSerialNumber), that is why these descriptors are absent in the output of the example. 

Do you think that the feature to be able to get these descriptors via `usb_host` library will be useful?

diplfranzhoepfinger, 2023-12-12:
Hi @roma-jam 

> Do you think that the feature to be able to get these descriptors via `usb_host` library will be useful?

I would love it.

diplfranzhoepfinger, 2023-12-12:
i even would love to see: 

SimTech HS-USB AT Port 9011

SimTech HS-USB Modem 9011

: SimTech HS-USB NMEA 9011

diplfranzhoepfinger, 2024-01-31:
Any Timeframe when this will be ready ?

HDLA-BG, 2024-03-05:
We also have the same problem in our project.

DOevermann, 2024-03-05:
We also have the same problem in our project.

diplfranzhoepfinger, 2024-03-19:
@david-cermak would help in the esp-modem to get rid of these Descriptors: 

[https://github.com/espressif/idf-extra-components/blob/09e79a9b24177b2024d373822029e67ac2495701/usb/esp_modem_usb_dte/include/esp_modem_usb_config.h#L75](https://github.com/espressif/idf-extra-components/blob/09e79a9b24177b2024d373822029e67ac2495701/usb/esp_modem_usb_dte/include/esp_modem_usb_config.h#L75)

having here `, 4, 5)`seems to be very bad style. 

Thanks.