espressif/esp-idf #12779: Python path case sensitivity issue on Windows

dyarkovoy, 2023-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-888-gcb66d6f6fa 

### Espressif SoC revision.

all

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

all

### Power Supply used.

USB

### What is the expected behavior?

idf.py does not complain about wrong python path when building project

### What is the actual behavior?

When the path of python.exe recorded in project's build config is case-different on a case-insensitive filesystem (Windows), this error occurs:

`'c:\work\esp32\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe' is currently active in the environment while the project was configured with 'C:\work\esp32\.espressif\python_env\idf5.3_py3.11_env\Scripts\python.exe'. Run 'idf.py fullclean' to start again.`

the `C:` drive is identical to `c:` drive on Windows, but this is not taken into account

### Steps to reproduce.

1. Build any ESP-IDF project in VSCode with vscode-esp-idf-extension on Windows
2. Try to build the same project from command line using `idf.py build`
you should see the error as described above
...

### More Information.

I will submit PR correcting this behavior shortly

dobairoland, 2023-12-13:
@jakub-kocka PTAL

jakub-kocka, 2023-12-13:
@dyarkovoy, thanks for submitting also the [PR](https://github.com/espressif/esp-idf/pull/12780) which solves this.
I haven't been able to reproduce this issue but with the fix everything works as expected.

As you said, according to documentation and my tests the string path is changed only on Windows platform.

closed 2023-12-18 by espressif-bot