espressif/esp-idf #12782: MCPWM up/down mode fails when synchronized to 0 decrement

florentbr, 2023-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The timer is set 0xffff (65535) when the MCPWM module is synchronized to `0 decrement` in up-down mode:

// Timer 0:    3  49  76  100  125  150  174  199  224  249 
// sync to 0 decrement
// Timer 0:    0  65515  65488  65460  65432  65405  65377  65350  65322  65295

According to the reference manual , it should increment from 0 to 1 and not from 0 to 65535.
It works as expected if the synchronised value is set to `1 decrement` instead of `0 decrement`. 

### Version
I (31) boot: ESP-IDF 5.1.1 2nd stage bootloader
I (31) boot: compile time Dec 12 2023 07:37:23
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v3.1

### To reproduce

```c++

#include "soc/mcpwm_reg.h"
#include "soc/mcpwm_struct.h"
#include "esp_rom_gpio.h"
#include "esp_rom_sys.h"
#include "esp_timer.h"
#include "esp_check.h"
#include "hal/clk_gate_ll.h"
#include "hal/clk_tree_ll.h"
#include "hal/gpio_ll.h"

void test_bug_sync()
{
    const uint16_t sync_phase = 0;

    periph_ll_enable_clk_clear_rst(PERIPH_PWM0_MODULE);
    periph_ll_reset(PERIPH_PWM0_MODULE);  // clear mcpwm registers, reset timers

    // MCPWM_UPDATE_CFG_REG
    MCPWM0.update_cfg.val =
        (( 1                        ) << MCPWM_GLOBAL_UP_EN_S           ) |  // .global_up_en  1:enable
        (( 1                        ) << MCPWM_OP0_UP_EN_S              ) ;  // .op0_up_en  1:enable

    // MCPWM_CLK_CFG_REG
    MCPWM0.clk_cfg.val =
        (( (160 / 20) - 1           ) << MCPWM_CLK_PRESCALE_S           ) ;  // .clk_prescale for 20MHz

    // MCPWM_TIMER0_CFG0_REG
    MCPWM0.timer[0].timer_cfg0.val =
        (( 20 - 1                   ) << MCPWM_TIMER0_PRESCALE_S        ) |  // .timer_prescale for 1us tick
        (( MCPWM_TIMER0_PHASE / 2   ) << MCPWM_TIMER0_PERIOD_S          ) ;  // .timer_period  max

    // MCPWM_TIMER0_SYNC_REG
    MCPWM0.timer[0].timer_sync.val =
        (( 1                        ) << MCPWM_TIMER0_SYNCI_EN_S        ) |  // .timer_synci_en  1:enable
        (( sync_phase               ) << MCPWM_TIMER0_PHASE_S           ) |  // .timer_phase
        (( 1                        ) << MCPWM_TIMER0_PHASE_DIRECTION_S ) ;  // .timer_phase_direction  1:decrement

    // MCPWM_TIMER_SYNCI_CFG_REG
    MCPWM0.timer_synci_cfg.val =
        ( 4 << MCPWM_TIMER0_SYNCISEL_S        ) ;  // .timer0_syncisel  SYNC0 from GPIO matrix

    // MCPWM_TIMER0_CFG1_REG
    MCPWM0.timer[0].timer_cfg1.val =
        ( 2 << MCPWM_TIMER0_START_S ) |  // .timer_start  2:timer0 starts and runs on
        ( 3 << MCPWM_TIMER0_MOD_S   ) ;  // .timer_mod    3:up-down

    for (int n = 10; n--; ) {
        printf("%lu  ", (MCPWM0.timer[0].timer_status.val & 0xffff));
        esp_rom_delay_us(1);
    }

    // trigger a timer sync
    printf("\nsync to 0 decrement\n");
    MCPWM0.timer[0].timer_sync.val |= MCPWM_TIMER0_SYNC_SW;

    for (int n = 10; n--; ) {
        printf(" %lu ", (MCPWM0.timer[0].timer_status.val & 0xffff));
        esp_rom_delay_us(1);
    }

    printf("\n");
}

```

suda-morris, 2023-12-13:
@florentbr 

But if you change it to "0 increment", then it also works as expected, right?

I think the UP-DOWN mode is special, we can understand it in two phases. One is the increase phase, its range is `[0, peak_value)`, another is the decrease phase, its range is `[peak_value,1]`. So `0` stays in the "increase phase". If you set the sync phase to "0 decrease", then you will get this "undefined behaviour".

florentbr, 2023-12-13:
@suda-morris

> But if you change it to "0 increment", then it also works as expected, right?

Yes

> So 0 stays in the "increase phase". If you set the sync phase to "0 decrease", then you will get this "undefined behaviour".

That's the whole point of the issue, this undefined, unspecified behaviour which led me to waste a few hours of debugging. The parameters are within range according to the reference manual, so either it's a bug or there should be a mention of this specific edge case. I'm not trying to understand or dispute the behaviour. I just hopped this was documented/specified.

closed 2023-12-23 by espressif-bot