espressif/esp-idf #12786: IDF 5.1.4 SD card library fails writing files, while mounting and reading files works seamlessly

ralphde, 2023-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.4

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom design in production since 18 month 

### Power Supply used.

External 5V

### What is the expected behavior?

creating and writing files 

### What is the actual behavior?

mounts and reads files seamlessly 

### Steps to reproduce.

load SD card sample project set GPIO's according your board, compile and run. Message will be that the file could not be opened.

### Debug Logs.

```plain
[0;32mI (305) example: Using SPI peripheral[0m
                              
[0;32mI (305) example: Mounting filesystem[0m
                               
[0;32mI (305) gpio: GPIO[37]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (315) sdspi_transaction: cmd=52, R1 response: command not supported[0m

                                                                              
[0;32mI (355) sdspi_transaction: cmd=5, R1 response: command not supported[0m

[0;32mI (375) example: Filesystem mounted[0m
                                
Name: SD16G
                                                                   
Type: SDHC/SDXC
                                                               
Speed: 20.00 MHz (limit: 20.00 MHz)
                                           
Size: 30288MB
                                                                 
CSD: ver=2, sector_size=512, capacity=62029824 read_bl_len=9
                  
SSR: bus_width=1
                                                              
[0;32mI (375) example: Opening file /sdcard/hello.txt[0m
                    
[0;31mE (5885) example: Failed to open file for writing[0m
                  
[0;32mI (5885) main_task: Returned from app_main()[0m
```

### More Information.

Executing C:\Users\ralph\AppData\Local\Programs\Python\Python312\python.exe C:\IDF-5.1.2\esp-idf-v5.1.2\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20221002 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20221002 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.12.0-esp32-20230921 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.8 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.11 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230320 (recommended, installed)
OK

CDT Build environment variables
PATH: C:\Users\ralph\.espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin;C:\Users\ralph\.espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin;C:\Users\ralph\.espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin;C:\Users\ralph\.espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin;C:\Users\ralph\.espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin;C:\Users\ralph\.espressif\tools\esp-clang\15.0.0-23786128ae\esp-clang\bin;C:\Users\ralph\.espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin;C:\Users\ralph\.espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin;C:\Users\ralph\.espressif\tools\cmake\3.24.0\bin;C:\Users\ralph\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin;C:\Users\ralph\.espressif\tools\ninja\1.10.2\;C:\Users\ralph\.espressif\tools\idf-exe\1.0.3\;C:\Users\ralph\.espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64;C:\Users\ralph\.espressif\tools\dfu-util\0.11\dfu-util-0.11-win64;C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env\Scripts;C:\IDF-5.1.2\esp-idf-v5.1.2\tools;C:/eclipse-cpp-2023-09//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.9.v20231028-0858/jre/bin/server;C:/eclipse-cpp-2023-09//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.9.v20231028-0858/jre/bin;C:\Program Files\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Program Files (x86)\IVI Foundation\IVI\bin;C:\Program Files\IVI Foundation\IVI\bin;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files\AutoFirma\AutoFirma;C:\Program Files\Git\cmd;C:\Users\ralph\AppData\Local\Programs\Python\Python312\Scripts\;C:\Users\ralph\AppData\Local\Programs\Python\Python312\;C:\Users\ralph\AppData\Local\Microsoft\WindowsApps;;C:\eclipse-cpp-2023-09;
USERDOMAIN_ROAMINGPROFILE: DESKTOP-84B1J43
PROCESSOR_LEVEL: 25
SESSIONNAME: Console
ALLUSERSPROFILE: C:\ProgramData
PROCESSOR_ARCHITECTURE: AMD64
ESP_IDF_VERSION: 5.1
PSModulePath: C:\Program Files\WindowsPowerShell\Modules;C:\Windows\system32\WindowsPowerShell\v1.0\Modules
SystemDrive: C:
VXIPNPPATH: C:\Program Files (x86)\IVI Foundation\VISA\
OPENOCD_SCRIPTS: C:\Users\ralph\.espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\share\openocd\scripts
IDF_DEACTIVATE_FILE_PATH: C:\Users\ralph\AppData\Local\Temp\tmpt15e2xyeidf_7044
IVIROOTDIR64: C:\Program Files\IVI Foundation\IVI\
IDF_CCACHE_ENABLE: 1
USERNAME: ralph
ProgramFiles(x86): C:\Program Files (x86)
VXIPNPPATH64: C:\Program Files\IVI Foundation\VISA\
PATHEXT: .COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC
DriverData: C:\Windows\System32\Drivers\DriverData
OneDriveConsumer: C:\Users\ralph\OneDrive
NIEXTCCOMPILERSUPP: C:\Program Files (x86)\National Instruments\Shared\ExternalCompilerSupport\C\
ProgramData: C:\ProgramData
IDF_COMPONENT_MANAGER: 1
ProgramW6432: C:\Program Files
__PSLockDownPolicy: 0
HOMEPATH: \Users\ralph
PROCESSOR_IDENTIFIER: AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD
IVIROOTDIR32: C:\Program Files (x86)\IVI Foundation\IVI\
ProgramFiles: C:\Program Files
PUBLIC: C:\Users\Public
windir: C:\Windows
=::: ::\
ZES_ENABLE_SYSMAN: 1
LOCALAPPDATA: C:\Users\ralph\AppData\Local
USERDOMAIN: DESKTOP-84B1J43
LOGONSERVER: \\DESKTOP-84B1J43
OneDrive: C:\Users\ralph\OneDrive
APPDATA: C:\Users\ralph\AppData\Roaming
CommonProgramFiles: C:\Program Files\Common Files
OS: Windows_NT
COMPUTERNAME: DESKTOP-84B1J43
NIIVIPATH: C:\Program Files (x86)\IVI Foundation\IVI\
IDF_PATH: C:\IDF-5.1.2\esp-idf-v5.1.2
PROCESSOR_REVISION: 5000
CommonProgramW6432: C:\Program Files\Common Files
ComSpec: C:\Windows\system32\cmd.exe
IDF_MAINTAINER: 1
TEMP: C:\Users\ralph\AppData\Local\Temp
SystemRoot: C:\Windows
HOMEDRIVE: C:
USERPROFILE: C:\Users\ralph
TMP: C:\Users\ralph\AppData\Local\Temp
CommonProgramFiles(x86): C:\Program Files (x86)\Common Files
NUMBER_OF_PROCESSORS: 16
IDF_PYTHON_ENV_PATH: C:\Users\ralph\.espressif\python_env\idf5.1_py3.12_env
ESP_ROM_ELF_DIR: C:\Users\ralph\.espressif\tools\esp-rom-elfs\20230320\

Operating System: windows 10
Java Runtime Version: 17.0.9+9
Eclipse Version: 4.29.0.v20230903-1000
Eclipse CDT Version: 11.3.0.202307110141
IDF Eclipse Plugin Version: 2.11.1.202310270725
ESP-IDF v5.1.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.12.0

ralphde, 2023-12-13:
Well the problem is a bit deeper then recognized in the first moment. When using the SD-card sample project the SD card file system is corrupted by first mounting and requires formatting, in my case, with Windows. It seems that at the first run the card is mounted as mmsd card. Just a first correction of my issue inquiry. I will provide further log's regarding this issue.

ralphde, 2023-12-14:
So here are more details. Beginning with a fresh formatted SD card 32GB which works fine with ESP-IDF v5.0-dirty SD-card sample project, the following dump is given and after that the SD-card formatting is destroyed:

 [0;32mI (297) main_task: Calling app_main()[0m
                              
[0;32mI (297) example: Initializing SD card[0m
                              
[0;32mI (297) example: Using SPI peripheral[0m
                              
[0;32mI (297) example: Mounting filesystem[0m
                               
[0;32mI (307) gpio: GPIO[37]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;31mE (357) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
[0m
                                                                           
[0;31mE (357) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).[0m
             
[0;32mI (357) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;31mE (367) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE)
. Make sure SD card lines have pull-up resistors in place.[0m
                
[0;32mI (377) main_task: Returned from app_main()[0m
                         
Running this without formatting the SD when mount is not possible. But after this the SD mounts but can not write files. 

This is the log:

                                 
[0;32mI (297) main_task: Calling app_main()[0m
                              
[0;32mI (297) example: Initializing SD card[0m
                              
[0;32mI (297) example: Using SPI peripheral[0m
                              
[0;32mI (297) example: Mounting filesystem[0m
                               
[0;32mI (307) gpio: GPIO[37]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (317) sdspi_transaction: cmd=52, R1 response: command not supported[0m

                                                                              
[0;32mI (357) sdspi_transaction: cmd=5, R1 response: command not supported[0m

[0;32mI (377) example: Filesystem mounted[0m
                                
Name: SD16G
                                                                   
Type: SDHC/SDXC
                                                               
Speed: 20.00 MHz (limit: 20.00 MHz)
                                           
Size: luMB
                                                                    
CSD: ver=2, sector_size=512, capacity=62029824 read_bl_len=9
                  
SSR: bus_width=1
                                                              
[0;32mI (377) example: Opening file /sdcard/hello.txt[0m
                    
[0;31mE (23907) example: Failed to open file for writing[0m
                 
[0;32mI (23907) main_task: Returned from app_main()[0m
                      
                                                                    
As said every time a new format is required.

And here is how the same custom hardware behaves using  ESP-IDF v5.0-dirty

[0;32mI (268) example: Initializing SD card[0m
                              
[0;32mI (268) example: Using SPI peripheral[0m
                              
[0;32mI (268) example: Mounting filesystem[0m
                               
[0;32mI (268) gpio: GPIO[37]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (278) sdspi_transaction: cmd=52, R1 response: command not supported[0m

                                                                              
[0;32mI (318) sdspi_transaction: cmd=5, R1 response: command not supported[0m

[0;32mI (338) example: Filesystem mounted[0m
                                
Name: SD16G
                                                                   
Type: SDHC/SDXC
                                                               
Speed: 20 MHz
                                                                 
Size: luMB
                                                                    
CSD: ver=2, sector_size=512, capacity=62029824 read_bl_len=9
                  
SSR: bus_width=1
                                                              
[0;32mI (338) example: Opening file /sdcard/hello.txt[0m
                    
[0;32mI (368) example: File written[0m
                                      
[0;32mI (388) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt[0m

[0;32mI (388) example: Reading file /sdcard/foo.txt[0m
                      
[0;32mI (388) example: Read from file: 'Hello SD16G!'[0m
                    
[0;32mI (388) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (398) example: Card unmounted[0m
                                    
[0;32mI (408) gpio: GPIO[36]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (408) gpio: GPIO[34]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1|
 Pulldown: 0| Intr:0 [0m
                                                     
[0;32mI (418) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1|
 Pulldown: 0| Intr:0 [0m

GhotsFDS, 2024-03-31:
I have the same problem, how did you solve it
` (376) example: Initializing SD card
I (376) example: Using SPI peripheral
I (386) example: Bus data: MOSI - 23, MISO - 19, SCLK - 18
I (386) example: Mounting filesystem
I (396) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
Speed: 5.00 MHz (limit: 5.00 MHz)
Size: 29610MB
CSD: ver=2, sector_size=512, capacity=60641280 read_bl_len=9
SSR: bus_width=1
I (476) example: Opening file /sdcard/hello.txt
E (9936) example: Failed to open file for writing
I (9936) main_task: Returned from app_main()`

GhotsFDS, 2024-03-31:
you said every time a new format is required.but how to do

ralphde, 2024-03-31:
well my client did not want to finance the shit, so I rolled back to 4.4.6 so I never solved it. An Espressive guy reached out asking details but came back and said he can not reproduce the issue on his side. Obviously  they do not have the required hardware.

New format has to be done on Windows or Linux.

ralphde, 2024-03-31:
What IDF version are you using?

GhotsFDS, 2024-04-01:
> 您使用的是哪个 IDF 版本？

i using idf 5.1.2,but just like your situation ，When I use 5.0 version it's fine

GhotsFDS, 2024-04-05:
I (31) boot: ESP-IDF v5.2.1-dirty 2nd stage bootloader
I (31) boot: compile time Apr  5 2024 21:07:18
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v3.1
I (40) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0e7fch ( 59388) map
I (125) esp_image: segment 1: paddr=0001e824 vaddr=3ffb0000 size=017f4h (  6132) load
I (128) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=25dc4h (155076) map
I (184) esp_image: segment 3: paddr=00045dec vaddr=3ffb17f4 size=00f34h (  3892) load
I (186) esp_image: segment 4: paddr=00046d28 vaddr=40080000 size=0ede0h ( 60896) load
I (221) boot: Loaded app from partition at offset 0x10000
I (221) boot: Disabling RNG early entropy source...
I (233) cpu_start: Multicore app
I (241) cpu_start: Pro cpu start user code
I (241) cpu_start: cpu freq: 160000000 Hz
I (242) cpu_start: Application information:
I (245) cpu_start: Project name:     sd_card
I (249) cpu_start: App version:      v5.1.2-dirty
I (255) cpu_start: Compile time:     Apr  5 2024 21:06:55
I (261) cpu_start: ELF file SHA256:  0a90ba6dfdd0f907...
I (267) cpu_start: ESP-IDF:          v5.2.1-dirty
I (272) cpu_start: Min chip rev:     v0.0
I (277) cpu_start: Max chip rev:     v3.99
I (282) cpu_start: Chip rev:         v3.1
I (287) heap_init: Initializing. RAM available for dynamic allocation:
I (294) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (300) heap_init: At 3FFB3098 len 0002CF68 (179 KiB): DRAM
I (306) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (312) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (319) heap_init: At 4008EDE0 len 00011220 (68 KiB): IRAM
I (327) spi_flash: detected chip: generic
I (330) spi_flash: flash io: dio
W (334) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (347) main_task: Started on CPU0
I (357) main_task: Calling app_main()
I (357) example: Initializing SD card
I (357) example: Using SPI peripheral
I (357) example: Mounting filesystem
I (367) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (377) sdspi_transaction: cmd=52, R1 response: command not supported
I (417) sdspi_transaction: cmd=5, R1 response: command not supported
I (437) example: Filesystem mounted
Name: SZYL
Type: SDHC/SDXC
Speed: 5.00 MHz (limit: 5.00 MHz)
Size: 29610MB
CSD: ver=2, sector_size=512, capacity=60641280 read_bl_len=9
SSR: bus_width=1
I (447) example: Opening file /sdcard/hello.txt
E (9817) example: Failed to open file for writing
I (9817) main_task: Returned from app_main()

GhotsFDS, 2024-04-06:
I encountered something very strange，i run it with 5.0.4 today and than the issue happened.it was ok yesterday. I was very confused

GhotsFDS, 2024-04-06:
After I replaced all my devices with new ones, it still doesn't work on 5.0.4

GhotsFDS, 2024-04-06:
Is this the idf through the git got updated?

ralphde, 2024-04-06:
I made similar experience 5.0 worked, moved up to 5.1, did not, moving back did not work. I concluded that you can not have various IDF versions on the same machine. However the Espressif team denied this and made clear yes it will work, I then rolled back to 4.4.6 and it worked. But how did you change the various IDF versions? another theory I had was that the SD card driver worked in the first released 5.0 version, but later installed versions (5.0.x from Espressif server) have been 'contaminated' with the failing code.

GhotsFDS, 2024-04-06:
> I made similar experience 5.0 worked, moved up to 5.1, did not, moving back did not work. I concluded that you can not have various IDF versions on the same machine. However the Espressif team [...]

i remove all IDF in my pc ，and just install idf 5.0.4 but it do not work

ralphde, 2024-04-06:
did you delete the folder .espressif? and when did you install the first 5.0.4 version? long time ago?
Anyhow I have seen it working in the first released 5.0.x version as well and later it did not. same with me here. 4.4.6 works, if it is urgent you may cosnider to go with this version

GhotsFDS, 2024-04-06:
> Hi @GhotsFDS, could you please do two things to help us diagnose the issue?
> 
> * Attach the source code of the application. If you used an IDF example, please include any modifications you made. Also please attach your sdkconfig file.
> * Please increase the log level to debug or verbose in menuconfig, and attach the complete log you get when this issue occurs.
> 
> Thank you

i used an IDF example and i changed bus_config in “Kconfig.projbuild”
menu "SD SPI Example Configuration"

    config EXAMPLE_FORMAT_IF_MOUNT_FAILED
        bool "Format the card if mount failed"
        default n
        help
            If this config item is set, format_if_mount_failed will be set to true and the card will be formatted if
            the mount has failed.

    config EXAMPLE_PIN_MOSI
        int "MOSI GPIO number"   
        default 23 if IDF_TARGET_ESP32
        default 35 if IDF_TARGET_ESP32S2
        default 35 if IDF_TARGET_ESP32S3
        default 4  # C3 and others

    config EXAMPLE_PIN_MISO
        int "MISO GPIO number"
        default 19 if IDF_TARGET_ESP32
        default 37 if IDF_TARGET_ESP32S2
        default 37 if IDF_TARGET_ESP32S3
        default 6  # C3 and others

    config EXAMPLE_PIN_CLK
        int "CLK GPIO number"
        default 18 if IDF_TARGET_ESP32
        default 36 if IDF_TARGET_ESP32S2
        default 36 if IDF_TARGET_ESP32S3
        default 5  # C3 and others

    config EXAMPLE_PIN_CS
        int "CS GPIO number"
        default 5 if IDF_TARGET_ESP32
        default 34 if IDF_TARGET_ESP32S2
        default 34 if IDF_TARGET_ESP32S3
        default 1  # C3 and others

endmenu

GhotsFDS, 2024-04-06:
> did you delete the folderdelete the folder? and when did you install the first 5.0.4 version? long time ago? Anyhow I have seen it working in the first released 5.0.x version as well and later it did not. same with me here. 4.4.6 works, if it is urgent you may cosnider to go with this version

yes i delete the folder delete the folder，i install the first 5.0.4 version just one week ago.yesterday i install 5.2.1 and 5.0.4 still worked，but now it do not work

ralphde, 2024-04-06:
That is a miracle, indeed, did you have an older version before installed?

igrr, 2024-04-06:
@GhotsFDS your log still doesn't contain output at debug and verbose level. Please enable verbose logs in menuconfig (under component config, log output) and attach those.

GhotsFDS, 2024-04-06:
Let's go through the timeline，The first idf i install is 5.1.2 ,then I had the same problem as you,so at your prompt i install 5.0.4 ,and it can work.yesterday i  listen to your advice to try 5.2.1，it do not work ,but in this time 5.04 still work .but now it shout down,i didn't change anything related to the configuration during the process

GhotsFDS, 2024-04-06:
@igrr what should I do to make my log  contain output at debug and verbose level

GhotsFDS, 2024-04-06:
@igrr 
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6992
load:0x40078000,len:15512
load:0x40080400,len:3864
0x40080400: _init at ??:?

entry 0x40080650
I (27) boot: ESP-IDF v5.0.4-dirty 2nd stage bootloader
I (27) boot: compile time 19:11:07
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v3.1
I (35) boot.esp32: SPI Speed      : 40MHz
I (40) boot.esp32: SPI Mode       : DIO
I (44) boot.esp32: SPI Flash Size : 2MB
I (49) boot: Enabling RNG early entropy source...
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00100000
I (87) boot: End of partition table
I (91) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0f0c4h ( 61636) map
I (122) esp_image: segment 1: paddr=0001f0ec vaddr=3ffb0000 size=00f2ch (  3884) load
I (124) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=266a0h (157344) map
I (185) esp_image: segment 3: paddr=000466c8 vaddr=3ffb0f2c size=0126ch (  4716) load
I (187) esp_image: segment 4: paddr=0004793c vaddr=40080000 size=0db70h ( 56176) load
I (221) boot: Loaded app from partition at offset 0x10000
I (221) boot: Disabling RNG early entropy source...
I (232) cpu_start: Multicore app
I (232) cpu_start: Pro cpu up.
I (232) cpu_start: Starting app cpu, entry point is 0x400811c0
0x400811c0: call_start_cpu1 at D:/Espressif/frameworks/esp-idf-v5.0.4/components/esp_system/port/cpu_start.c:143

I (0) cpu_start: App cpu up.
D (243) clk: RTC_SLOW_CLK calibration value: 3154061
I (255) cpu_start: Pro cpu start user code
I (256) cpu_start: cpu freq: 160000000 Hz
I (256) cpu_start: Application information:
I (260) cpu_start: Project name:     sd_card
I (265) cpu_start: App version:      v5.0.4-dirty
I (271) cpu_start: Compile time:     Apr  6 2024 19:10:54
I (277) cpu_start: ELF file SHA256:  518e40d0e682913f...
I (283) cpu_start: ESP-IDF:          v5.0.4-dirty
I (288) cpu_start: Min chip rev:     v0.0
I (293) cpu_start: Max chip rev:     v3.99
I (298) cpu_start: Chip rev:         v3.1
D (303) memory_layout: Checking 9 reserved memory ranges:
D (308) memory_layout: Reserved memory range 0x3ff82000 - 0x3ff82000
D (314) memory_layout: Reserved memory range 0x3ffae000 - 0x3ffae6e0
D (321) memory_layout: Reserved memory range 0x3ffb0000 - 0x3ffb2b30
D (327) memory_layout: Reserved memory range 0x3ffe0000 - 0x3ffe0440
D (333) memory_layout: Reserved memory range 0x3ffe3f20 - 0x3ffe4350
D (340) memory_layout: Reserved memory range 0x40070000 - 0x40078000
D (346) memory_layout: Reserved memory range 0x40078000 - 0x40080000
0x40080000: _WindowOverflow4 at D:/Espressif/frameworks/esp-idf-v5.0.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (353) memory_layout: Reserved memory range 0x40080000 - 0x4008db70
0x40080000: _WindowOverflow4 at D:/Espressif/frameworks/esp-idf-v5.0.4/components/freertos/FreeRTOS-Kernel/portable/xtensa/xtensa_vectors.S:1743

D (359) memory_layout: Reserved memory range 0x50001fe8 - 0x50002000
D (366) memory_layout: Building list of available memory regions:
D (372) memory_layout: Available memory region 0x3ffae6e0 - 0x3ffb0000
D (378) memory_layout: Available memory region 0x3ffb2b30 - 0x3ffb8000
D (385) memory_layout: Available memory region 0x3ffb8000 - 0x3ffc0000
[... 142 lines cut ...]
D (1104) spi_master: device5 release bus
D (1104) spi_master: device5 locked the bus
D (1114) spi_master: device5 release bus
I (1114) example: Filesystem mounted
Name: SZYL
Type: SDHC/SDXC
Speed: 20 MHz
Size: 29610MB
CSD: ver=2, sector_size=512, capacity=60641280 read_bl_len=9
SSR: bus_width=1
I (1134) example: Opening file /sdcard/hello.txt
D (1134) spi_master: device5 locked the bus
D (1144) spi_master: device5 release bus
D (1144) spi_master: device5 locked the bus
D (1154) spi_master: device5 release bus
D (1154) spi_master: device5 locked the bus
D (1154) spi_master: device5 release bus
D (1164) spi_master: device5 locked the bus
D (1164) spi_master: device5 release bus
repeat
spi_master: device5 locked the bus
spi_master: device5 release bus
over and over

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14893325/sdkconfig.txt)

GhotsFDS, 2024-04-06:
@ralphde i have same issue in 4.4.6

GhotsFDS, 2024-04-06:
Maybe it's a hardware issue?

ralphde, 2024-04-06:
well you can not exclude anything, but you have full same issue, you can mount and read files but writing files fucks up the sd card and requires formatting?

ralphde, 2024-04-06:
I assume you have pull ups on all SPI pins, the MOSI and CLK close to the sd card the MISO close to the ESP32?

GhotsFDS, 2024-04-07:
> I assume you have pull ups on all SPI pins, the MOSI and CLK close to the sd card the MISO close to the ESP32?

What do you mean？ close to？In a physical sense？

ralphde, 2024-04-07:
sure, distance

GhotsFDS, 2024-04-07:
@igrr @ralphde hi guys.I feel like I've found the crux of the issue.
With the hardware-related questions from yesterday in mind, I focused on this particular issue today. I discovered that after connecting the SD card to the computer and manually formatting it, the program could run once on any version of the IDF. However, when I press the EN button to restart the ESP32, the program encounters an error:

Error Message:

E (416) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).

After reformatting the SD card, the program is able to run again. The perplexing part is that this was not an issue before. Previously, the program would run successfully and would not encounter errors after restarting by pressing the EN button. But today, after restarting the program, the SD card seems to have been damaged. When connected to the PC, it displays the message “This volume does not contain a recognized file system,” which is quite baffling.

The problem appears to have a level of uncertainty akin to quantum mechanics; I cannot predict what it will be like tomorrow. Currently, I suspect that the frequent operations performed on the SD card during testing may have led to some issues. Yet, this does not explain why the same situation occurs with a new SD card.

GhotsFDS, 2024-04-07:
there are several possibilities:

The program’s operations on the SD card have caused internal damage after multiple program starts.
Frequent testing and plugging/unplugging of the SD card have caused damage.
Both 1 and 2 are true.
In today’s tests, I found that versions 4.4, 5.0, 5.1, or 5.2 of the IDF can all run successfully. The key issue seems to be the impact on the SD card during the testing process. Given the current situation, the SD card’s program cannot be put into production as it appears to be very unstable. I am not sure if my conclusions and problems are universal or if there is a specific issue with my hardware. Perhaps you could continue with multiple tests.

ralphde, 2024-04-07:
well, keep in mind that this happened to me with a device in production, working fine in the hundreds of devices with 4.4.6 code. If there would be an issue with pull up resistors the SD would very likely always be able to be reformatted.

When I have been investigating this issue in IDF 5.1.x I ran into tickets talking about that the SPI driver MOSI signal was inverted. I guess that the SD card component relays on the same code.

And don't forget that we can mount and read existing files over and over again without issues. I also played with the CLK speed of the SD SPI, I assuming you have connected the SD card with simple SPI hence 4 wires? It would be interesting to connect with 8 wires, which is as well possible as you can see in the struct for the hardware definition.

It is a miracle why this happens with our hardware in production and it blocks us from migrating upwards.

@igrr I urge for further investigation.

igrr, 2024-04-08:
Based on

> D (64364) vfs_fat: vfs_fat_open: fresult=7

it is possible that the filesystem got corrupted somehow, and FATFS library considers the root directory to be full.

@ralphde if you get a chance to perform the same test (enabling verbose log level in menuconfig) on IDF v5.1.x and confirm if you get the same error (fresult=7), that would help.

Additionally, if you can get a raw disk image of the SD card after the error is reported and share it with us, we could check if the filesystem is indeed corrupted. (This option might work if the SD card isn't too large.)

ralphde, 2024-04-08:
I am working on a different project right know, this week no chance to perform this test
However, I confirm that on windows the same SD is considered to be full, that*s right. But all files are gone. Only a reformat brings the SD back to life

GhotsFDS, 2024-04-08:
> I am working on a different project right know, this week no chance to perform this test
> However, I confirm that on windows the same SD is considered to be full, that*s right. But all files are gone. Only a reformat brings the SD back to life

In fact, so am I.

GhotsFDS, 2024-04-08:
> a raw disk image of the SD card

How do I get a raw disk image of the SD card

GhotsFDS, 2024-04-08:
> Based on
>
>
> it is possible that the filesystem got corrupted somehow, and FATFS library considers the root directory to be full.
>
> @ralphde if you get a chance to perform the same test (enabling [...]

How do I get a raw disk image of the SD card

igrr, 2024-04-08:
> How do I get a raw disk image of the SD card

On Windows you can use a tool such as https://www.imgburn.com/ (I'm sure there are also plenty of others), on Linux and macOS you can use `dd` to copy from /dev/sdX or /dev/rdiskN to a file.

> However, I confirm that on windows the same SD is considered to be full

That would be consistent with the behavior on ESP side. It still needs to be determined what causes the card to be corrupted in such a way, though.

GhotsFDS, 2024-04-08:
> Additionally, if you can get a raw disk image of the SD card after the error is reported and share it with us, we could check if the filesystem is indeed corrupted. (This option might work if the SD card isn't too large.)

image of the SD card has 30gb，Maybe I need to buy a smaller SD card

GhotsFDS, 2024-04-09:
@igrr @ralphde guys, I have found the key to the issuse. Today, I compared the code of version 5.1 with version 5.0, and the code of 5.1 includes a formatting test.

**code in 5.0:**

`    // Read a line from file
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);

    // Strip newline
    char *pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    // All done, unmount partition and disable SPI peripheral
    esp_vfs_fat_sdcard_unmount(mount_point, card);
    ESP_LOGI(TAG, "Card unmounted");

    //deinitialize the bus after all devices are removed
    spi_bus_free(host.slot);`
    
**log in 5.0**

I (367) main_task: Calling app_main()
I (367) example: Initializing SD card
I (367) example: Using SDMMC peripheral
I (377) example: Mounting filesystem
I (407) example: Filesystem mounted
Name: SZYL
Type: SDHC/SDXC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 29610MB
CSD: ver=2, sector_size=512, capacity=60641280 read_bl_len=9
SSR: bus_width=1
I (417) example: Opening file /sdcard/hello.txt
I (437) example: File written
I (457) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
I (457) example: Reading file /sdcard/foo.txt
I (457) example: Read from file: 'Hello SZYL!'
I (467) example: Card unmounted
I (467) main_task: Returned from app_main()

**code in 5.1:**

`    // Rename original file
    ESP_LOGI(TAG, "Renaming file %s to %s", file_hello, file_foo);
    if (rename(file_hello, file_foo) != 0) {
        ESP_LOGE(TAG, "Rename failed");
        return;
    }

    ret = s_example_read_file(file_foo);
    if (ret != ESP_OK) {
        return;
    }

    // Format FATFS
    ret = esp_vfs_fat_sdcard_format(mount_point, card);
    if (ret != ESP_OK) {
        ESP_LOGE(TAG, "Failed to format FATFS (%s)", esp_err_to_name(ret));
        return;
    }

    if (stat(file_foo, &st) == 0) {
        ESP_LOGI(TAG, "file still exists");
        return;
    } else {
        ESP_LOGI(TAG, "file doesnt exist, format done");
    }

    const char *file_nihao = MOUNT_POINT"/nihao.txt";
    memset(data, 0, EXAMPLE_MAX_CHAR_SIZE);
    snprintf(data, EXAMPLE_MAX_CHAR_SIZE, "%s %s!\n", "Nihao", card->cid.name);
    ret = s_example_write_file(file_nihao, data);
    if (ret != ESP_OK) {
        return;
    }

    //Open file for reading
    ret = s_example_read_file(file_nihao);
    if (ret != ESP_OK) {
        return;
    }

    // All done, unmount partition and disable SPI peripheral
    esp_vfs_fat_sdcard_unmount(mount_point, card);
    ESP_LOGI(TAG, "Card unmounted");

    //deinitialize the bus after all devices are removed
    spi_bus_free(host.slot);`
    
  **log in 5.1**

I (368) main_task: Calling app_main()
I (368) example: Initializing SD card
I (368) example: Using SDMMC peripheral
I (378) example: Mounting filesystem
I (498) example: Filesystem mounted
Name: SZYL
Type: SDHC/SDXC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 29610MB
CSD: ver=2, sector_size=512, capacity=60641280 read_bl_len=9
SSR: bus_width=1
I (498) example: Opening file /sdcard/hello.txt
I (518) example: File written
I (538) example: Renaming file /sdcard/hello.txt to /sdcard/foo.txt
I (538) example: Reading file /sdcard/foo.txt
I (538) example: Read from file: 'Hello SZYL!'
I (548) vfs_fat_sdmmc: Formatting card, allocation unit size=16384

Because the SD card is being formatted here, if the SD card has a lot of space, then the program will stop running here for a long time.

I (14328) example: file doesnt exist, format done
I (14328) example: Opening file /sdcard/nihao.txt
I (14338) example: File written
I (14338) example: Reading file /sdcard/nihao.txt
I (14338) example: Read from file: 'Nihao SZYL!'
I (14348) example: Card unmounted
I (14348) main_task: Returned from app_main()

So, if we restart or interrupt the program while the SD card is being formatted then the SD card will get corrupted at that time.That's why we think that the program works only the first time, and it doesn't work when it starts againI used an example of 5.1 no matter which IDF version I used before and I restarted the program when it was formatted,so i have the problem that all IDF versions fail.(I didn't notice it because it took too long to format)
@igrr I suggest adding a hint to the remarks in this part of the code, such as (do not restart or interrupt the program during formatting, this will cause the SD card to be corrupted)

@ralphde about the issue “Failed to open file for writing”,This issue no longer appears when I replace the hardware of my SD card adapter module.I think you're probably buying the SD card adapter module in bulk, so all the tests have this issue.Maybe you can fix the damage of the SD card by formatting the SD card on your PC, and replace the SD card adapter, and then pay attention to the formatting problem during the test, which may solve your problem

ralphde, 2024-04-09:
nop sorry, this is a custom hardware working fine since since 4.4 using a custom firmware with no formatting at all

GhotsFDS, 2024-04-09:
> nop sorry, this is a custom hardware working fine since since 4.4 using a custom firmware with no formatting at all

Ok, these are two issue. Your issue actually appeared before I changed the equipment. I can run in versions 5.0 and 4.4 and I had an error in version 5.1.when I'm in version 5.1  the procedure writing failed at this time so the program was overed before step of formatting. When I just changed the equipment, there was no writing issue, and at the same time there was a formatting problem, which is why situation was completely different from the day before I changed the equipment.

GhotsFDS, 2024-04-09:
I suggest that espressif can annotate the formatted test code in these examples after 5.1.Because if people accidentally interrupt the program during formatting during the first test, the sd card will be damaged. In the next test, they may not be able to realize that the sd card has been damaged, causing confusion.

ralphde, 2024-04-09:
so then it must be a clk issue in >5.0 @igrr can you tell if there is a change in this direction?

GhotsFDS, 2024-04-09:
> Thanks for the suggestion, we did this recently: [e2da3a4](https://github.com/espressif/esp-idf/commit/e2da3a4259768dd4d115ccd74541c6409a932688)

So you guys actually found this problem two months ago, why didn't you modify the source code?

igrr, 2024-04-09:
We made this change for a different reason, we weren't aware that formatting also causes the issue you ran to. We will backport this change to v5.1.x release branch, it will be part of the next bugfix release.