espressif/esp-idf #12788: Pre-encrypted OTA with user-generated keys

eme-bennytang, 2023-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am trying to enable the pre-encrypted OTA to add more security to my project. (alongside with secureboot and flash encryption)

I am following below guideline to do it
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/ota.md#encrypted-ota

But I notice in this example, the image pre-encryption is actually encrypted with **PUBLIC** key, and we need to hardcode **PRIVATE** key in our source code to decrypt it.

Is there any reason with this?
Like in a normal signing process, a public key should be extracted from the private key, owned by the "signing party" (our customer), and shared to external developers (us), so that they can own the final signing process to ensure they can control the image release.

But if we are to do things around, we would be having access on private key, where we can actually extract the public key ourselves and thus encrypt and release the image without they participation.
That would defeat part of the purpose of the encryption isnt it? 
(I understand the encryption also serves as a protection against OTA image being intercepted on air, and reverse engineered it)

Further question:
Can we use an rsa 4096 key instead of rsa 3072 key in this process?

mahavirj, 2023-12-13:
@eme-bennytang 

> Is there any reason with this?

This is by design. [Pre-encrypted OTA](https://github.com/espressif/esp-idf/blob/692c1fcc52b9b936c73dead4ef0c2ea1fbdfb602/examples/system/ota/pre_encrypted_ota/README.md) scheme relies on using RSA for the encryption purpose (and not the signing) and hence the role for private key swaps here. You can consider this to be similar to the PGP encryption.

> But if we are to do things around, we would be having access on private key, where we can actually extract the public key ourselves and thus encrypt and release the image without they participation.

It is recommended to enable platform security features on device side and the private key be protected using secure storage (e.g., NVS encryption) or similar mechanism. Please refer to the [security](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/security/security.html) documentation guide.

>  Can we use an rsa 4096 key instead of rsa 3072 key in this process?

No. The pre-encrypted scheme only support RSA-3072 scheme. Please refer to this [component](https://github.com/espressif/idf-extra-components/blob/master/esp_encrypted_img/README.md) for more details.

Hope this helps!

eme-bennytang, 2023-12-14:
Hi @mahavirj ,

Thanks for the reply!
I did some digging on the encryption topic, and understand its originated from RSA encryption for the key swap (although I don't quite get the idea, but its how it is)

As you have mentioned, I am now looking for ways to protect the "image encryption private key" (which is just plain embedded in the image in the example).

I think I got 2 ways to do it in a more secure manner:

1. Enable NVS data encryption
[image]
This should be what I should be doing if I understand correctly
But this involves another key another protection...blah blah blah

2. Embed the key in the image but with some hardcode encryption/decryption
- In secure boot V2, a private key is assigned in the SDK config
- Public key is extracted from that private key and embed to bootloader  (I assume)
- When that bootloader 1st boot, burn that public key into efuse
- Private key is used when building the project, and to sign the application/bootloader built
That means the public key must have been embedded inside the bootloader.bin, but I can't find it
Thus there should be a way to deform it somehow and hide in the binaries?

mahavirj, 2023-12-14:
@eme-bennytang 

For (1), yes your understanding is correct. The pre-encrypted OTA specific private key can be protected using [secure storage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/security.html#secure-storage). NVS encrypted partition can be generated using the [NVS partition generation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_gen.html#generate-encrypted-nvs-partition) utility and then it can be flashed on the device.

Another option is to just keep the key embedded inside the application then enable [flash encryption](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/security.html#flash-encryption) scheme. This will ensure that the entire firmware partition (including the pre-encrypted ota key) is stored in encrypted manner on the external flash storage.

For secure boot v2 case:

- The digest (SHA256) of the public key is programmed in the EFuse block
- The public key and signature are appended to the bootloader (and application) in the format defined [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/secure-boot-v2.html#signature-block-format)

>  Thus there should be a way to deform it somehow and hide in the binaries?

Secure boot scheme alone may not help here. Please consider enabling flash encryption or NVS encryption as described above.

eme-bennytang, 2023-12-14:
Hi @mahavirj 

Thanks for the quick reply again!

If I am to understand correctly, the bootloader image DID NOT embed the public key ITSELF
The whole process of bootloader -> burn public key to efuse is as below:

1. Enabled secureboot and build the project
2. bootloader-unsigned.bin and bootloader.bin is built
3. The signature block in bootloader.bin is created using the SDK config designated key (private) to sign the bootloader-unsigned.bin (which is done automatically by the building process normally)
4. The signature block itself contains public key information
5. 1st power up with this bootloader, it will retrieve public key info from the signature block, and then burn the public key digest to efuse

In order words, the public key burnt into the efuse is controlled by the signature block appended in the bootloader.bin

mahavirj, 2023-12-14:
@eme-bennytang Yes, your understanding is correct. Bootloader first does the secure boot sanity verification for applicable partitions and then only burns the digest to efuse.

eme-bennytang, 2023-12-14:
@mahavirj 
Just to double confirm: 
In order words, the public key burnt into the efuse is controlled by the signature block appended in the bootloader.bin (which actually requires private key to create the signature block)

mahavirj, 2023-12-14:
@eme-bennytang 

Yes, right. FWIW, secure boot scheme can also be enabled externally (through host machine) and this guide [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/host-based-security-workflows.html#enable-secure-boot-v2-externally) should make the steps more clear for you.

Thanks.

closed 2023-12-14 by eme-bennytang