espressif/esp-idf #12789: Unexpected NimBLE GATT performance compared to Bluedroid

Scottapotamas, 2023-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I've been working on a series of latency benchmarks for different wireless radios/stacks, and measured some odd behaviours from NimBLE when compared to Bluedroid for GATT/SPP style transfers.

___

These measurements are for one-way transfer latency - no ack/response behaviour is implemented or measured.

[image]

[image]

[image]

Is this behaviour reasonable/expected for the NimBLE + ESP32 stack? Any suggestions?

___

Comparing server notify and client WriteNoResp is also inconsistent between Bluedroid and NimBLE:

[image]

[image]

___

# Reproduction Notes 

## Software

Code for `esp32-spp` (classic), `esp32-ble`, `esp32-nimble` [is on GitHub](https://github.com/Scottapotamas/embedded-wireless-latency-eval/tree/main/firmware). These vary between minor changes from Espressif examples to heavier modifications to achieve feature parity.

The biggest difference from IDF examples: I've removed UART bridge behaviour, test payloads are handled directly on device.

Where test payloads exceed MTU, I manually send them as smaller MTU sized packets where the benchmark task requires some kind of library level event to signal the next packet i.e. `BLE_GAP_EVENT_NOTIFY_TX`, similar to the approach used in Espressif throughput example.

I originally ran these tests with `IDF 5.1.1` but can reproduce them with latest `v5.3-dev-892-g692c1fcc52` which is ~3 days old.

`docker run -i --privileged --rm -v $PWD:/project -w /project -it espressif/idf:latest`

Other relevant changes with menuconfig:
- Change NimBLE internal logging to "Warning" to suppress extra terminal output (which affects results).
- Change compiler options to build for performance `-O2`

## Test Setup 

[image]

- Two ESP32-WROOM-32 devkit boards, positioned on table 1m apart.
- Signal generator provides a 3.3V 50 μs square pulse at configurable interval (250 ms for all tests) as stimulus signal. Connected to IO19, handled via GPIO ISR.
- When trigger is flagged, and BT connection is already established, send test payload
- Receiving board rx event occurs, payload data is sent to user's benchmark task via FreeRTOS queue.
- Benchmark task checks payload for length and CRC, if valid, drives IO18 high to signal valid transmission.
- Saleae Logic 8 captures trigger and complete signals at 100 MS/s (10 ns resolution).

I've measured trigger-to-output overhead at ~4.11 μs when tested in a loopback configuration.

All firmware variants support both server notify and client writes, so swapping the trigger/valid signal connections allows testing client-server direction as needed.

Scottapotamas, 2023-12-19:
After a bit more testing I worked out that manually specifying connection parameters is needed to allow NimBLE to match Bluedroid's defaults.

```c
    // Sets the client's BLE connection behaviours 
    // https://mynewt.apache.org/latest/network/ble_hs/ble_gap.html#c.ble_gap_update_params
    // ITVL uses 1.25 ms units
    // Timout is in 10ms units
    // CE LEN uses 0.625 ms units
    // BLE specifies minimum 7.5ms connection interval
    struct ble_gap_upd_params conn_parameters = { 0 };
    conn_parameters.itvl_min = 6;   // 7.5ms
    conn_parameters.itvl_max = 24;  // 30ms
    conn_parameters.latency = 0;
    conn_parameters.supervision_timeout = 20; 
    // https://github.com/apache/mynewt-nimble/issues/793#issuecomment-616022898
    conn_parameters.min_ce_len = 0x00;
    conn_parameters.max_ce_len = 0x00;

    ble_gap_update_params(peer->conn_handle, &conn_parameters);
```

This improves results substantially.

[image]

[image]

Remaining comments for Espressif:

1. While the lower bound latencies for both Notify and Writes is now in-line with BLE 7.5 ms interval minimums, NimBLE is still typically slower and has wider spread of latencies.
2. The 1KiB test (sent in smaller chunked packets due to MTU=200) is still far slower than Bluedroid.
3. I found the only mention of this in the [blecent_throughput](https://github.com/espressif/esp-idf/tree/8c9e29898feb8d2ff0c63f137c58d36df7fe8612/examples/bluetooth/nimble/throughput_app/blecent_throughput) example.

xyzzy42, 2023-12-29:
Have you verified the actual LL packet MTU is increased?

I've found that using NimBLE on ESP32-S3 (other ESP32s untested) that the LL MTU is not increased.  Calling ` ble_att_set_preferred_mtu()` and/or `ble_gattc_exchange_mtu()` will only change the ATT layer MTU.  The larger ATT packets will still be fragmented via L2CAP into 27 byte LL packets.  Which is of course quite disastrous for performance.

To get larger LL packets, it's necessary to send an HCI command to increase the controller's *connInitialMaxTxOctets* (I tested this) or the connection's *connMaxTxOctets* (probably, I haven't tested) value.

Other BLE stacks I've used haven't required this.  I think the flaw is in Espressif's controller implementation.  I think it's expected to increase *connMaxTxOctets* in response to receiving a `LL_LENGTH_REQ` PDU.  It doesn't do this.  The NimBLE controller (not used on ESP32) does this.  The BT core spec (Ver 5.4, Vol 6, Part B, §5.1.9 "Data Length Update procedure") seems to imply the controller should do this.

KaeLL, 2024-01-03:
@rahult-github thoughts?

Scottapotamas, 2024-01-04:
@xyzzy42 Thanks for chiming in, your hint led me down the right path.

Sniffing the transfers, I see the 200 byte MTU update packets during connection, but for the larger transfers i.e. 128B, I still saw fragmented 26B LL transfers as you described:

[image]

I was able to resolve this issue by calling  

```c
#define LL_PACKET_TIME (2120)
#define LL_PACKET_LENGTH (200)
// ...
ble_hs_hci_util_set_data_len( event->connect.conn_handle, LL_PACKET_LENGTH, LL_PACKET_TIME );
```

inside `BLE_GAP_EVENT_CONNECT`inside `BLE_GAP_EVENT_CONNECT` on successful connection.

This was needed on both the server and client boards. Wireshark trace shows a single packet for the 128B test (pictured below) and correctly used 6x single notification packets which matches the expected application-side 1024B chunking behaviour.

[image]

Here's a comparison between these different changes (HCI data also includes conn parameter changes)

[image]

With the HCI length call, the 1024B test is twice as quick (halved latency), with minimal improvements to the smaller one-packet sized tests. This brings it _roughly in line with Bluedroid_'s **default** performance.

Updated comments for Espressif:
- There is little/no documentation for much of NimBLE which makes finding these kinds of API calls hard.
- I'd probably consider the loss of MTU length across ESP-NimBLE VHCI to be a bug.
- Mentioning default behaviour and performance would save others from needing to troubleshoot poor default performance, I'd suggest a section similar to the [IDF Speed Optimization Docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/speed.html)?

xyzzy42, 2024-01-04:
You can also call `ble_gap_write_sugg_def_data_len()` before creating any connections to set the initial MTU length to be longer.

These two functions are totally missing in any documentation.  Even things like Espressif's "How do I increase the MTU?" FAQ does not mention them.

In other BLE stacks I've used, this isn't necessary on **both** peers in the connection, as it is with ESP32+NimBLE.  Only one, the GATT client, needs to send the length request.  Then the other peer will increase the MTU in response to that.

I'm not entirely sure if it should be the NimBLE host or the Espressif controller which should be doing this.  I think it's the controller.  But I'm pretty sure a direct HCI call via a barely known function in application code is not the correct way.

Scottapotamas, 2024-01-04:
Thanks for the suggestion.

Yeah I agree that calling against HCI functions from application space is **a bad idea™**.

The idiomatic NimBLE approach ([mynewt-nimble source](https://github.com/apache/mynewt-nimble/blob/master/nimble/host/src/ble_gap.c)):
- `ble_gap_set_data_len()` wraps the `ble_hs_hci_util_set_data_len()` function I hacked with above. I tested it and got the same behaviour.
- `ble_gap_write_sugg_def_data_len()` doesn't seem to impact my test code and/or ESP-WROOM-32 at all.

___

This was only a detail I found while testing NimBLE as a subset of other benchmarks, so I'm content with the 'fixed' results and can move on to other things. 

I'd like to see an official statement/explanation and some improvements for future users though.

xyzzy42, 2024-01-04:
> * `ble_gap_write_sugg_def_data_len()` doesn't seem to impact my test code and/or ESP-WROOM-32 at all.

Do you mean it does have a difference between using `ble_gap_set_data_len()` or that it doesn't cause the MTU to increase?

I tested this on ESP32-S3 and  `ble_gap_write_sugg_def_data_len()` did work to increase the MTU for new connections.  If I read the HCI spec correctly, it must be called before the connection is established.

Scottapotamas, 2024-01-05:
I tested `ble_gap_write_sugg_def_data_len()` in a few places (applied to both boards):

- fairly early in setup, prior to `nimble_port_run()`
- after setup, i.e. advertise/scan has started, prior to connection
- at time of connection

and I still saw LL fragmentation in Wireshark captures. Might be a subtlety I'm missing there.

trentp-igor, 2024-01-05:
I'm calling after `esp_nimble_hci_init()` and `nimble_port_run()` and before `ble_gap_adv_start()`.  Also I'm using ESP32-S3.

I wonder if this is a difference between the controller for the ESP32 vs the ESP32-S3?  I also wonder, if the ESP32 doesn't support this HCI command, if it returns an error code?  Since these commands never appear in any Espressif documentation, I doubt any difference in controller support between chips is documented either.

xyzzy42, 2024-01-05:
I'm calling after esp_nimble_hci_init() and nimble_port_run() and before ble_gap_adv_start(). Also I'm using ESP32-S3.

I wonder if this is a difference between the controller for the ESP32 vs the ESP32-S3? I also wonder, if the ESP32 doesn't support this HCI command, if it returns an error code? Since these commands never appear in any Espressif documentation, I doubt any difference in controller support between chips is documented either.

xyzzy42, 2024-06-18:
I found a new problem, which might explain some of the differences seen.  Setting `ble_gap_write_sugg_def_data_len()` after nimble_port_run, but before advertising starts worked to get a larger LL MTU sometimes, but not always.
1. Android phone, unbonded.  Works.
2. Android phone, bonded.  Works.
3. iPhone, unbonded. Works.
4. iPhone, **bonded**.  Fails!

I.e., it doesn't work when the client is an iPhone that is already bonded to the server (ESP32).

An examination of the packet capture shows than in the working cases, the LL_LENGTH_REQ packet is sent from the client to the server while the connection is still unencrypted.  In case 2, the connection is encrypted after this request and in case 1 & 3 the connection remains unencrypted.

But in case 4, the connection is encrypted first, and then the LL_LENGTH_REQ packet is sent.  It's the first packet sent after the encryption handshake finishes.  Something in the Espressif controller does not like this, and responds to the request with LL_REJECT_EXT_IND, LMP PDU Not Allowed.

I don't know the encryption is actually related to this problem or not.  This is happening inside the binary only controller code, so I can't debug it further.  But encryption is the only obvious difference between the accepted MTU requests and the rejected one.

I then tried using `ble_gap_set_data_len()` on the connection, after it's setup and the client (iPhone) initiated request has already failed.  This generates a LL_LENGTH_REQ from the server and the phone accepts it and increases the LL MTU.

chipweinberger, 2024-12-10:
Is 200 the optimal length? Why not say, 512?

```
#define LL_PACKET_TIME (2120)
#define LL_PACKET_LENGTH (200)
// ...
ble_hs_hci_util_set_data_len( event->connect.conn_handle, LL_PACKET_LENGTH, LL_PACKET_TIME );
```

chipweinberger, 2024-12-10:
I see this in the docs: https://mynewt.apache.org/latest/network/ble_hs/ble_gap.html

according to this, the maximum value is 0xFB, or 251. Not 200.

[image]

Scottapotamas, 2025-02-04:
@chipweinberger I didn't see this earlier sorry.

The 200 byte packet length I used in my code snip above was for consistency of packet size when comparing latency between Bluedroid and NimBLE (and others).

For real-world use, I believe the MTU should probably be sized to consider the underlying ATT and L2CAP header sizes and still pack nicely in order to reduce wasted frame time where possible. I've normally had to experimentally test on different devices due to subtleties in behaviour.

vandy, 2025-06-02:
The summary for future readers. 

My setup is ESP-IDF v5.4, ESP32h2, Android (nRFConnect), nRF52840 dongle + Wireshark. But I suppose the following is applied in general to other chipsets and esp-idf versions.

_Valid Range:_
`ATT_MTU` min size is `23`, max size is `527` (limited by the Nimble stack `BLE_ATT_MTU_MAX` in `ble_att.h`).
`LL packet` min size is `27`, max size is `251` (by BLE spec).

One could check these values on startup (in `ble_hs_cfg.sync_cb` function) with:
`ble_att_preferred_mtu()` which is controlled by `CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU` (`256` by default I believe);
`ble_gap_read_sugg_def_data_len` (returns `27` bytes and `328` usec for my chip).

It's also possible to change these values with `ble_att_set_preferred_mtu(value)` and `ble_gap_write_sugg_def_data_len(size, time)`.

_How it works:_
There is also another variable **payload_size**.

`ATT packet = opcode (1B) + [handle (2B)] + payload (up to ATT_MTU - 1[- 2] byte)`. E.g. when a gatt server sends a notification to the subscribed client: opcode is `0x1b`, handle is `0xXXXX` => therefore space for a payload is `ATT_MTU - 3`. For Read Response: opcode is `0x0b`, no handle => thefore space for a payload is `ATT_MTU - 1`.

The gatt server can only send `ATT_MTU - 3` bytes of data in a notification (even if the **payload_size** is greater than `ATT_MTU`). How many LL packets it takes depends on `LL packet` size. So the rule is simple - the server sends 1 ATT packet (if `ATT_MTU` is larger than `LL packet` it will take N LL packets).

At this stage there are some corner cases:
- Even if `ble_gap_read_sugg_def_data_len` returns `27` bytes at startup, as I understood BLE spec, the Controller is allowed to choose any value in the range. In my case the Controller (I'm programming a peripheral device) sends 251 bytes, 2120 usec to a Central in `LL_LENGTH_REQ` (for TX and RX, which is not the same as default value returned by the function).
- **The most important** thing is that if `ATT_MTU` is not exchanged, than it's supposed by the Nimble stack to be `23` bytes (which is min value).
Notification will provide the client with only 20 bytes of payload data, even if your `LL packet` is much larger and the payload is 128B or 500B. Actually it'll be 27 byte LL packet with 20 bytes of data.
My Android phone seems not to send `ATT_MTU` exchange, and I have to initiate the exchange from my Peripheral device (which is the server) with "the client command" `ble_gattc_exchange_mtu(connectionHandle, NULL, NULL)`. It's interesting fact that as the exchange happens Android sends a new `LL_LENGTH_REQ` with TX `min(LL packet max, ATT_MTU + 4), 2120 usec`. Before the `ATT_MTU` exchange it sent TX value `27 bytes, 328 usec`.
If `LL Packet` is not set with `ble_gap_write_sugg_def_data_len` controller uses (in my case) `251` bytes. After the `ATT_MTU` exchange happens my device could send data with chunks (LL packets) up to 251 bytes long (as long as the client supports such RX `LL packet`). I think that the Controller is free to choose the size and it looks at `ATT_MTU`. But if I set `LL packet` with the function, than the Controller uses this value for max LL packet size it sends. Therefore if one wants to ensure the size of LL packet to be consistent he needs to set the `LL Packet` explicitly.

`ATT_MTU` exchange is **crucial** thing to do if one wants to increase the size of transferred payload data.

After `ATT_MTU` is set to the increased value (both peers exchanged the information and `BLE_GAP_EVENT_MTU` event happened) `LL Packet` value begins to play role how many actual packets would be sent through the air. Increasing `LL Packet` (`BLE_GAP_EVENT_DATA_LEN_CHG` event) is the next important thing to do to increase the number of transferred data per packet.

When the client reads a characteristic value with Read Request and **payload_size** is greater than `ATT_MTU`, server sends 1 ATT packet with payload `ATT_MTU` - 1. The client could make a Long Attribute Read (with Blob Request and offset) to get the remaining data (each Blob Response will hold `ATT_MTU` bytes of data). This is the difference between Server Notification scenario and Client Read scenario.

The minimum viable code for BLE Peripheral (gatt server) will be:
```c
static int handleGapEvent(struct ble_gap_event *event, void *arg) {
  switch (event->type) {
    case BLE_GAP_EVENT_CONNECT:
      if (event->connect.status == 0) {
        ble_gattc_exchange_mtu(event->connect.conn_handle, NULL, NULL); // <- ensure that ATT_MTU exchange will happen
        // ble_gap_set_data_len(event->connect.conn_handle, 251, 2120); // <- could set LL Packet for specific connection
      }
    break;
  }

  return 0;
}

static void nimble_callback_on_sync() {
  ble_gap_write_sugg_def_data_len(251, 2120); // <- could set "default" LL Packet size for all future connections, ensure specific LL Packet size is used
  // ble_att_set_preferred_mtu(256); // <- could change default ATT_MTU size, could be whatever between 23 and 527 bytes.
  // ... set advertise params
  ble_gap_adv_start(BLE_OWN_ADDR_PUBLIC, NULL, 60000, &params, handleGapEvent, NULL);
}

static void nimbleTask() {
  nimble_port_run();
  nimble_port_freertos_deinit();
}

void ble_init() {
  nimble_port_init();
  ble_hs_cfg.sync_cb = on_sync;
  // ... other config settings
  ble_svc_gatt_init();
  ble_svc_gap_init();
  ble_gatts_count_cfg(ourServices);
  ble_gatts_add_svcs(ourServices);
  nimble_port_freertos_init(nimbleTask);
}
```
With such setup one could achieve the maximum possible data transferred in one packet (`251` bytes chunk).

rahult-github, 2026-09-21:
This is an old issue, and a lot of work has already gone into these points across various commits (host behaviour, examples, and
defaults). Rather than closing with a one-liner, here is a collated summary of what the thread found, what ESP-IDF does now, and how to get Bluedroid-like GATT performance on NimBLE.

The original NimBLE vs Bluedroid gap was not a lower GATT throughput ceiling on NimBLE. It came from defaults and from mixing up two different sizes:

  1. Connection interval / connection event length
        NimBLE did not use the same defaults as Bluedroid. Updating parameters (for example a 7.5 ms interval and a large enough max_ce_len) brings one-way latency in line with the BLE minimum interval.

  2. ATT MTU vs Link Layer PDU
        ble_att_set_preferred_mtu() and ble_gattc_exchange_mtu() only raise the ATT MTU. Without Data Length Extension (DLE), the controller still uses 27-byte LL PDUs, so large ATT PDUs are fragmented on air. That matches the 1 KiB result in this thread (ATT MTU 200, but still ~26-byte LL fragments).

 Once connection parameters were updated and DLE was applied on both peers (ble_hs_hci_util_set_data_len /  ble_gap_set_data_len), latency dropped sharply (the 1 KB case roughly halved) and results were in line with Bluedroid’s default behaviour.

 NimBLE has changed a lot since this was filed.

• On a successful connection the host now issues HCI Set Data Length with the maximum LL size (251 octets). You no longer need an obscure HCI call from application code for ESP NimBLE to use long LL PDUs.
  • ATT MTU is still not exchanged automatically (this is spec-correct). If nobody calls ble_gattc_exchange_mtu(), ATT MTU stays 23 and notify/write payloads stay around 20 bytes even when DLE is 251. Some phones never start the exchange; the ESP peripheral can start it with ble_gattc_exchange_mtu().
  • The Kconfig default CONFIG_BT_NIMBLE_ATT_PREFERRED_MTU is 256. This is configurable and can be increased ( to 512) via Kconfig or ble_att_set_preferred_mtu().

  The example in NimBLE that has more information for reference :  examples/bluetooth/nimble/throughput_app/

  GATT pair: gatt/blecent_throughput and gatt/bleprph_throughput. That example:

  • sets ATT MTU 512 and exchanges MTU
  • sets DLE 251 / 2120 µs
  • updates connection parameters (including full CE length)
  • pipelines notifications and enlarges MSYS/ACL pools

The parent README section Throughput Optimization lists the knobs. Shield-box GATT figures on current IDF (MTU 512, DLE 251, 7.5 ms) are about 750–794 kbps WRITE/NOTIFY on 1M PHY, and about 1.33–1.38 Mbps on 2M PHY which is matching to what bluedroid stack too shows. 

  Other points raised in this thread : 

  • 200 vs 251 vs 512: 251 (0xFB) is the BLE maximum LL PDU. 200 was only used to match the Bluedroid test size. ATT MTU 512 is the example choice so one ATT PDU fills a long LL PDU after headers (notify/write payload 509).

  • ble_gap_write_sugg_def_data_len(): this sets a controller default before connections. In this thread it was chip- and timing-dependent (worked on ESP32-S3 for some people, not on classic ESP32 as reported initially). The reliable path on current ESP NimBLE is the per-connection Set Data Length on connect (now also done by the host). The GATT throughput examples still call ble_hs_hci_util_set_data_len() on BLE_GAP_EVENT_CONNECT.

  • ATT MTU exchange is mandatory if you want more than 20 bytes of GATT payload, even if LL length is already 251. That matches the later summary in this issue.

  • iPhone already bonded, LL_LENGTH_REQ rejected with LL_REJECT_EXT_IND (LMP PDU Not Allowed): that is a controller interaction after encryption, not the original “NimBLE slower than Bluedroid by default” problem. The workaround reported here is to call     ble_gap_set_data_len() from the ESP after the link is up. If this still happens on a current IDF release, please open a new issue     with a sniffer log (chip, IDF version, iOS version, bonded vs unbonded).

Closing this issue. The host-side DLE-on-connect behaviour, the throughput example, and the 1M GATT numbers versus Bluedroid cover the original report. Please use examples/bluetooth/nimble/throughput_app/ as the reference if you are tuning GATT rate latency.

closed 2026-09-21 by rahult-github