espressif/esp-idf #12792: free() crashes in i2c_cmd_link_delete() when reading and writing on i2c

louisthetall, 2023-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-342-gbcf1645e44

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Free should not crash

### What is the actual behavior?

After writing and reading on the i2c for 10-15 seconds, a crash occurs.

### Steps to reproduce.

1. run program with i2c communication
2. wait for crash to occur

### Debug Logs.

```plain
assert failed: tlsf_free tlsf.c:1119 (!block_is_free(block) && "block already marked as free")

Backtrace: 0x400814c6:0x3ffed890 0x40092525:0x3ffed8b0 0x40096449:0x3ffed8d0 0x400953d2:0x3ffed9f0 0x4009527c:0x3ffeda10 0x40081e56:0x3ffeda30 0x40096479:0x3ffeda50 0x401893d6:0x3ffeda70 0x400dbd7b:0x3ffeda90 0x400fd00d:0x3ffedac0 0x400dea9a:0x3ffedaf0 0x400dcf52:0x3ffedb60 0x400dd6a1:0x3ffedbd0 0x4009299e:0x3ffedbf0
0x400814c6: panic_abort at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/esp_system/panic.c:472

0x40092525: esp_system_abort at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/esp_system/port/esp_system_chip.c:84

0x40096449: __assert_func at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/newlib/assert.c:81

0x400953d2: tlsf_free at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/heap/tlsf/tlsf.c:1119 (discriminator 1)

0x4009527c: multi_heap_free_impl at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/heap/multi_heap.c:231

0x40081e56: heap_caps_free at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/heap/heap_caps.c:388

0x40096479: free at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/newlib/heap.c:39

0x401893d6: i2c_cmd_link_delete at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/driver/i2c/i2c.c:1159

0x400dbd7b: i2cComms::write(unsigned char, unsigned char*, int, unsigned char*, int) at C:/Users/Louis/Documents/Troo/FarmTRX/troo-idf/components/comms/I2cCommsDrv.cpp:247

0x400fd00d: pcap04::startCdc() at C:/Users/Louis/Documents/Troo/FarmTRX/ms/ms-firmware/components/pcap04/pcap04.cpp:670

0x400dea9a: MsCapacitanceSensor::GetCapacitance(MSRawSensor&, char*) at C:/Users/Louis/Documents/Troo/FarmTRX/ms/ms-firmware/components/farmTrxMs/MsCapacitanceSensor.cpp:167

0x400dcf52: MoistureSensor::MainTask() at C:/Users/Louis/Documents/Troo/FarmTRX/ms/ms-firmware/components/farmTrxMs/FarmTrxMs.cpp:160

0x400dd6a1: MoistureSensor::startMainTask(void*) at C:/Users/Louis/Documents/Troo/FarmTRX/ms/ms-firmware/components/farmTrxMs/include/farmTrxMs.hpp:131

0x4009299e: vPortTaskWrapper at C:/.espressif/frameworks/esp-idf-v5.1/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:162

ELF file SHA256: ed933863b3680717

I (17590) esp_core_dump_flash: Save core dump to flash...
I (17597) esp_core_dump_flash: Erase flash 24576 bytes @ 0x10000
I (21) esp_core_dump_flash: Write end offset 0x5f04, check sum length 4
I (22) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
```

### More Information.

In my i2c dirver read(), I print out the address of `cmd` before calling `i2c_cmd_link_delete(cmd);`
```
cmd = i2c_cmd_link_create();
i2c_master_start(cmd);
i2c_master_write_byte(cmd, chip_addr << 1 | WRITE_BIT, ACK_CHECK_EN);
i2c_master_write(cmd, data_addr, len_addr, ACK_CHECK_EN);
i2c_master_start(cmd);
i2c_master_write_byte(cmd, chip_addr << 1 | READ_BIT, ACK_CHECK_EN);
if (len_data > 1) {
	i2c_master_read(cmd, data_rx, len_data - 1, I2C_MASTER_ACK);
}
i2c_master_read_byte(cmd, data_rx + len_data - 1, I2C_MASTER_NACK);
i2c_master_stop(cmd);
i2c_master_cmd_begin(mI2cNum, cmd, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT));
if (cmd) {
	printf("%s free %p, ", __func__, cmd);
	i2c_cmd_link_delete(cmd);
}
```
In i2c_cmd_link_delete(), I print the address of `i2c_cmd_handle_t cmd_handle` followed by `i2c_cmd_link_t *ptmp` as follows:

```
void i2c_cmd_link_delete(i2c_cmd_handle_t cmd_handle)
{
    i2c_cmd_desc_t *cmd = (i2c_cmd_desc_t *) cmd_handle;
    printf("free %p, ", cmd_handle);
    /* Memory should be freed only if allocated dynamically.
     * If the user gave the buffer for a static allocation, do
     * nothing. */
    if (cmd == NULL || i2c_cmd_link_is_static(cmd)) {
        return;
    }
    while (cmd->free) {
        i2c_cmd_link_t *ptmp = cmd->free;
        cmd->free = cmd->free->next;
	printf("free %p\n", ptmp);
        free(ptmp);
    }
    cmd->cur = NULL;
    cmd->free = NULL;
    cmd->head = NULL;
    free(cmd_handle);
    return;
}
```

Output:
```
read free 0x3ffb7bb8, free 0x3ffb7bb8, free 0x3ffb7c18
free 0x3ffb7c30
free 0x3ffb9d04
free 0x3ffb9d1c
free 0x3ffb9d34
free 0x3ffb9d4c
free 0x3ffb9d64
free 0x3ffb9d7c
read free 0x3ffb7bb8, free 0x3ffb7bb8, free 0x3ffb7c18
free 0x3ffb7c30
free 0x3ffb9d02

== Crash occurs ==
```

The address seems to always be multiples of 4, but when it changes to a multiple of 2, the crash occures.

mythbuster5, 2023-12-14:
This has not been maintained for a long time. Would you like try `i2c_cmd_link_create_static` and `i2c_cmd_link_delete_static`?

o-marshmallow, 2023-12-14:
@louisthetall IDF v5.1.2 presents a much more convenient function for your use case [i2c_master_transmit_receive](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2c.html#_CPPv427i2c_master_transmit_receive23i2c_master_dev_handle_tPK7uint8_t6size_tP7uint8_t6size_ti). In your case:
```
void i2cComms::write(unsigned char chip_addr, unsigned char* data_addr, int len_addr, unsigned char* data_rx, int len_data)
{
    // Replace the first parameter with your actual I2C dev
    i2c_master_transmit_receive(this->i2c_dev, data_addr, len_addr, data_rx, len_data, pdMS_TO_TICKS(I2C_MASTER_TIMEOUT));
}
```

Underneath, this function doesn't perform any dynamic allocation so it should be faster and not present any risk in terms of memory corruption

AxelLin, 2023-12-16:
> This has not been maintained for a long time. Would you like try `i2c_cmd_link_create_static` and `i2c_cmd_link_delete_static`?

@mythbuster5 @o-marshmallow 

I think this is not a good excuse of not taking a look at this issue.

The thing is i2c_cmd_link_create() / i2c_cmd_link_delete() could be used by
may external libraries or user's application code.
If this is indeed a bug, had better fix it rather than switching to other APIs.

MMI, 2024-11-17:
_Maybe_ me too -- I have a project that has been stable for years using esp-idf 4.x.  Now the 4.x support is over, I have been migrating the project to 5.3.  For the first time, I'm seeing assertions from `tlsf_free()`.  I am suspicious of dynamic memory allocation in my own FW so I avoid it but I do use i2c extensively and I've also seen traces implicating freeRTOS itself (in the task termination path).
🤷