espressif/esp-idf #12793: BLE mesh support for both provisionor and node in single code.

ankit-bansal-oxit, 2023-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32 | ESP32 C3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-devkit | esp32cd-devkitM-1

### Power Supply used.

USB

### What is the expected behavior?

Hello,

I have the project requirement that, in  which we have 4 devices, in which 3 are nodes, and one is provisoionor as well as ble gateway. The role of the devices depend on what we have written on the NFC-eeprom.  The device boot up, and check it role from the nfc-eeprom and start working accordingly. I written the codes for both node and provisionor seperately and all thing were wroking properly.  The problem start when I try to merge the code. The program crashes when I enable "Support for BLE Mesh Node" and "Support for BLE Mesh Provisioner".
We really need to configure our devices dynamically or runtime. 

### What is the actual behavior?

The device crashes with following mesasge 
I (466) phy_init: phy_version 1130,b4e4b80,Sep  5 2023,11:09:30

assert failed: bt_mesh_prov_init prov.c:1799 (prov_info->uuid)
Core  0 register dump:
MEPC    : 0x40380700  RA      : 0x40388a7e  SP      : 0x3fca7930  GP      : 0x3fc90600  
0x40380700: panic_abort at /home/ankit/sdk-soft/esp-idf/components/esp_system/panic.c:452

0x40388a7e: prvGetDisinheritPriorityAfterTimeout at /home/ankit/sdk-soft/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:2381

TP      : 0x3fc7680c  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000089  S1      : 0x00000001  A0      : 0x3fca796c  A1      : 0x3fc91b61  
A2      : 0x00000001  A3      : 0x00000029  A4      : 0x00000001  A5      : 0x3fc94000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000009  S3      : 0x3fca7a7e  
S4      : 0x3fc91b60  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
3fca7930: 0x00000000 0x0000002e 0x3c0e1204 0x4038e6d8 0x3fc91f38 0x3c0e1204 0x3fc92324 0x3c0e11c3
0x4038e6d8: malloc at /home/ankit/sdk-soft/esp-idf/components/newlib/heap.c:24

3fca7950: 0x3fc91f48 0x3fca7964 0x3fc91f4c 0x3c0e1188 0x3fc91b60 0x39393731 0x3fca9900 0x65737361
[... 152 lines cut ...]
3fca7b70: 0x3fca7b4c 0x00000000 0x00000013 0x3fca6d48 0x5f435442 0x4b534154 0x7bfb7c00 0x00925144
3fca7b90: 0x00000000 0x3fca7b40 0x00000013 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9aad4
3fca7bb0: 0x3fc9ab3c 0x3fc9aba4 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000
3fca7bd0: 0x420b7ebe 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7bf0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7c10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7c30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7c50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7c70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fca7c90: 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000054 0x00000000 0x3fca7ca4 0x00000000
3fca7cb0: 0x00000000 0x00000000 0x3fca7cbc 0xffffffff 0x3fca7cbc 0x3fca7cbc 0x00000000 0x3fca7cd0
3fca7cd0: 0xffffffff 0x3fca7cd0 0x3fca7cd0 0x00000001 0x00000001 0x00000000 0x8500ffff 0x00000000
3fca7cf0: 0xb33fffff 0x00000000 0x00000054 0x00000000 0x3fca7cfc 0x00000000 0x00000000 0x00000000
3fca7d10: 0x3fca7d14 0xffffffff 0x3fca7d14 0x3fca7d14 0x00000000 0x3fca7d28 0xffffffff 0x3fca7d28

ELF file SHA256: 2114bf16fae8bb51
Warning: checksum mismatch between flashed and built applications. Checksum of built application is 57b84372a1af69a9338b88c6e31383e6cecefa8f01456ea8d49a7f28863b969d

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048b82
0x40048b82: ets_secure_boot_verify_bootloader_with_keys in ROM
```

ankit-bansal-oxit, 2023-12-15:
Ok the problem is solved. I need to provide the prov->uuid. this would be the uuid assign as a node. I provide and its working as intended.

closed 2023-12-15 by ankit-bansal-oxit

ghoshaljayant, 2023-12-26:
@ankit-bansal-oxit  i am still facing the same issue, can you please help me how you solved the problem ? As per your comment i have tried setting esp_ble_mesh_prov_t provision prov_uuid           = dev_uuid, but it is still crashing.

forx157, 2023-12-26:
@ghoshaljayant 
Hi, Could you please provide the log and the sdkconfig file?

ankit-thealchemist, 2023-12-26:
> @ankit-bansal-oxit i am still facing the same issue, can you please help me how you solved the problem ? As per your comment i have tried setting esp_ble_mesh_prov_t provision prov_uuid = dev_uuid, but it is still crashing.

In the provisionar file, use dev_uuid from the node example and put it into the, prov_uuid,
esp_ble_mesh_prov_t provision prov_uuid = dev_uuid

ghoshaljayant, 2023-12-26:
@ankit-thealchemist did as you mentioned, still not working , tried idf.py fullclean build ,

@forx157  attaching sdkconfig and logs , kindly have a look and help me.

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/13769487/sdkconfig.txt)
[log.txt](https://github.com/espressif/esp-idf/files/13769500/log.txt)

forx157, 2023-12-26:
@ghoshaljayant 
Hi, Because you have both provisioner and node functionalities enabled concurrently. And it is not recommended to do so.

Furthermore, the occurrence of this crash is due to the fact that you have enabled node functionality without providing UUID information. Please note that 'uuid' is used for nodes, while 'prov_uuid' is used for provisioners. You can find specific details in the esp_ble_mesh_prov_t structure.

ghoshaljayant, 2023-12-26:
@forx157 
Hi, As you mentioned both provisioner and node functionalities enabled concurrently, and that is correct. I am trying to combine onoff_node and provisioner code, so that one esp_32 device can do provision as well as act as a onoff_client, as similar to nRF mesh application (provisioner + onoff_client).

Can you recommend any way to do achieve something like that ?
- may be a device cannot be a node as well as provisioner at the same time ?

Other part of your comment, i am trying out now and let you my outcome.

ghoshaljayant, 2023-12-26:
@forx157 it worked, not crashing anymore. Thanks for explaining.

forx157, 2023-12-27:
@ghoshaljayant 
If you want to implement both provisioner and on-off client functionalities on a single node, this is entirely achievable. You can refer to the latest version of our example in the '/examples/bluetooth/esp_ble_mesh/provisioner' directory on GitHub for guidance.