espressif/esp-idf #12794: HTTP Server - Support for HTTP_ANY

hoeken, 2023-12-13:
### Is your feature request related to a problem?

I'm developing [PsychicHttp](https://github.com/hoeken/PsychicHttp), which is a wrapper for the ESP-IDF HTTP server in the style of the popular ESPAsyncWebserver library.  So far it's been a major success, and much more reliable.

One of the major features that I'm unable to implement is support for HTTP_ANY, which should match all HTTP methods.  I've looked into various methods for getting around this, but they all seem to have some downsides

### Describe the solution you'd like.

Implement a HTTP_ANY enum that will match any HTTP method:

```cpp
httpd_uri_t uri_get = {
    .uri      = "/uri",
    .method   = HTTP_ANY,
    .handler  = get_handler,
    .user_ctx = NULL
};
```

### Describe alternatives you've considered.

Currently the users add multiple endpoints for each HTTP method, usually with the same handler.

I've considered a few ways around this, such as:

* implementing our own uri / method checking using httpd_register_err_handler() which is pretty ugly and skips some key features like user_ctx
* internally adding endpoints with url * for every method, but that is also pretty ugly and will be a performance hit.

My conclusion is that this would be better implemented upstream.

hoeken, 2023-12-22:
@hmalpani any feedback on this request?  I'm happy to do it myself and submit a pull request, but I want to know if this feature would be accepted or not.

hmalpani, 2023-12-26:
Hello @hoeken 

I did not see HTTP_ANY method mentioned in the RFC. To support this, we need to update http_parser component which is a third party component. They might also not accept the change as it is not there in the RFC. I would suggest to register same handler for different methods. I don't think it would have a significant performance impact.

hoeken, 2023-12-29:
@hmalpani 

Yes, I understand that HTTP_ANY is not in the RFC, but that is okay because HTTP_ANY would be a wildcard that means the uri callback can called on any request method.  The parser would not need to change, as knowing the method of the request is still useful for handling the request.

I am not setup to do dev work on the esp-idf library, but my proposed changes are this:

in esp_http_server.h add:
```cpp
//use a high value outside the range of the rest of the HTTP_METHOD defines from http_parser
#define HTTP_ANY 100
```

Change line [106 in http_uri.cpp](https://github.com/espressif/esp-idf/blob/5d8fb281e7b4ae3f3827c91aa6dc7ef969580b5a/components/esp_http_server/src/httpd_uri.c#L106) to this:
```cpp
if (hd->hd_calls[i]->method == method || hd->hd_calls[i]->method == HTTP_ANY) {
```

Also, lastly I would like to note that some of my users have 80+ endpoints that use both GET and POST on the same URI and that they have to register 2 handlers for each url.  One of them has reported that this is using an extra 10kb of heap over his previous solution.  I think this is a pretty minor addition that people would really appreciate.

proddy, 2024-01-09:
+1 to have HTTP_ANY implemented

closed 2024-01-30 by espressif-bot

proddy, 2024-02-03:
wonderful, thanks for implementing this.