espressif/esp-idf #12795: Nimble blecent_test.py is missing after 5.1.1

jbart12, 2023-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I'm working through the BLE Nimble Examples and the README directs my to use the python utility `blecent_test.py`, but it has been removed from the codebase in v5.1.1.  Is there a reason it was removed?  Is there an alternative?  Thank you.

rahult-github, 2023-12-14:
The newer IDF version deprecated the TTFW framework. This is why files referring to old framework were removed.

jbart12, 2023-12-14:
I see, thank you.  Is there a new framework that is used now?  Is there a different example project I should be following?  Thanks for your time.

jbart12, 2023-12-14:
I would like to get an example project running where I can communicate between a host and two peripherals.  Thank you.

rahult-github, 2023-12-15:
pytest is the new framework that is to be used.  here are the sample example files you can refer and develop on top of it as per your usecase / requirement. 

[pytest_bleprph.txt](https://github.com/espressif/esp-idf/files/13684787/pytest_bleprph.txt)
[pytest_blecent.txt](https://github.com/espressif/esp-idf/files/13684789/pytest_blecent.txt)

closed 2023-12-18 by jbart12