espressif/esp-idf #12797: BLUFFS: Any official statement?

fpgax, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am writing to inquire about the newly disclosed BLUFFS vulnerability.

References:
https://www.bluetooth.com/learn-about-bluetooth/key-attributes/bluetooth-security/bluffs-vulnerability/
https://francozappa.github.io/publication/2023/bluffs/
https://dl.acm.org/doi/pdf/10.1145/3576915.3623066

Does Espressif have an official statement regarding BLUFFS for its products?
Current mitigations and any future planned additional mitigations?

boblane1, 2023-12-15:
Hi @fpgax,

Thanks for your reporting. BLUFFS attacker employs BIAS to bypass authentication and KNOB to downgrade the entropy of session key, which is called as the KNOB+BIAS attack chain. We have made improvements to mitigate these two security vulnerabilities. 

1. For the KNOB attack, we have increased the minimum encryption key size to 7 octets, which is not anticipated to be possible in real-time during a session. And it is also recommended by Bluetooth SIG.
2. For the BIAS attack, we have enforced secure connection mode between pairing and session establishment when both devices support secure connection. However, ESP32 Classic Bluetooth only supports the legacy secure connection.  And the Bluedriod Host stack requires an additional authentication when the peer device is in Central(Master) role.

So, we can conclude that we reduced the impact of BLUFFS, but were unable to completely resolve it. An official statement will be announced soon.

fpgax, 2023-12-19:
Thank you for your detailed response.

closed 2023-12-19 by fpgax

AxelLin, 2024-01-05:
> So, we can conclude that we reduced the impact of BLUFFS, but were unable to completely resolve it. An official statement will be announced soon.

Where to find the official statement? Seems cannot find it in espressif website.

boblane1, 2024-01-05:
Hi @AxelLin，

The announcement has been drafted and is still in reviewing state. Please wait for a while and continue to pay attention to the advisories in espressif website.

Thanks.

boblane1, 2024-01-25:
Hi @AxelLin, @fpgax,

The announcement has been released on espressif website.
https://www.espressif.com/en/support/documents/advisories