espressif/esp-idf #12798: 5.2 beta1 undocumented change

dannybackx, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2 beta1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

recompile existing code without changes

### What is the actual behavior?

Build failure, due to rename of SOC_GDMA_PAIRS_PER_GROUP to SOC_GDMA_PAIRS_PER_GROUP_MAX.
The macro on the next line as well, apparently, see below.

This change is not documented anywhere. The diff also indicates documentation was deleted, that seems like a strange thing to do.

  Danny

diff  /home/danny/src/github/esp32/esp-idf-v5.{1.2,2-beta1}/components/soc/esp32s3/include/soc/soc_caps.h

146,148c156,159
< #define SOC_GDMA_GROUPS            (1)  // Number of GDMA groups
< #define SOC_GDMA_PAIRS_PER_GROUP   (5)  // Number of GDMA pairs in each group
< #define SOC_GDMA_SUPPORT_PSRAM     (1)  // GDMA can access external PSRAM
---
> #define SOC_AHB_GDMA_VERSION           1U
> #define SOC_GDMA_NUM_GROUPS_MAX        1U
> #define SOC_GDMA_PAIRS_PER_GROUP_MAX   5
> #define SOC_AHB_GDMA_SUPPORT_PSRAM     1

### Steps to reproduce.

see above

suda-morris, 2023-12-15:
sorry, we didn't expect the user will use these macros. May I know how your code is using these macros, instead of calling the `esp_private/gdma.h` driver API? @dannybackx

dannybackx, 2023-12-15:
It's not something I wrote myself.
See the link to the demo application of this device https://www.aliexpress.com/item/1005005235784548.html?spm=a2g0o.order_list.order_list_main.156.6dd71802VHYEvn 

Demo app link : https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848

suda-morris, 2023-12-15:
Thanks @dannybackx  That project is using some unsafe code, for example: https://github.com/wireless-tag-com/ZX3D95CE01S-TR-4848/blob/master/components/bus/8080_lcd_esp32s3.c#L354

Where that component is skipping the GDMA driver to allocate the channel, so in a concurrent scenario, a "free" channel searched by that component may be already occupied by another component, which utilizes the GDMA driver.

I can add those missing macros back again to make the project build again, but again, that project needs refactoring, because it's unsafe and also easy to break when there's a big refactor in the upstream esp-idf.

dannybackx, 2023-12-17:
Note this is also in https://github.com/smartpanle/QMSD-ESP32-BSP .
I pointed them to this thread.

closed 2023-12-23 by espressif-bot