espressif/esp-idf #12799: Connection to WiFi examples

shlomo-edu, 2023-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to use one or more of the examples under WiFi, to connect the ESP32-S3 to a server app that I created on Windows (in the local network).
1. I thought that I will use the: station_example_main.c (under getting_started).
But I see that it only have the intialization phase `wifi_init_sta()` but doesn't have and transmit function and receive events.
2. I see that there is a power_save.c example which may be very relevant to my application.
What is the cons of using it? (the pros are that it reduces the power consumption by half?)
And how do I combile this example with the: station_example_main.c example?

Xiehanxin, 2023-12-15:
hi @shlomo-edu , I think you can refet some exmaple under protocol https://github.com/espressif/esp-idf/tree/master/examples/protocols and power save example is use the power save mode in 802.11 protocol, the chip will close the RF and enter the sleep and wake in a certain interval

shlomo-edu, 2023-12-17:
Thanks, it looks like this is what I was looking for.
I am not closeing this issue yet because I want to try it first and see that I can combine it with the "power save" mode example.

Sherry616, 2024-05-15:
Hi @shlomo-edu, could you please share your latest updates on this issue? Thanks.

shlomo-edu, 2024-05-15:
Hi @Sherry616 
I have moved to using the ESP-AT project instead of building the code in the ESP.
It has a lot of adavatages, and I am good with it.
Thanks.

closed 2024-05-15 by shlomo-edu