espressif/esp-idf #12801: i2c_master_read_from_device times out immidiately

nicklasb, 2023-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I have a simple problem, at least it is simple to describe; i2c_master_read_from_device timeouts (283) regardless of any setting I provide for timeout for my ESP32 WROOM 32D. 
I have looked at https://github.com/espressif/esp-idf/issues/11397, however that doesn't seem to be it. 
Thing is, the peripheral is a Raspberry Pi Pico running the wire library, and it gets the request for data, and tries to send it. 
However I suppose no one will be in time for a <1 ms timeout. 

Am i missing something else? Why would the timeout value basically not matter? I cannot properly discern what actual values are valid as timeout, docs doesn't excplicitly say its ticks or something else?

mythbuster5, 2023-12-15:
It's ticks actually.

nicklasb, 2023-12-15:
> It's ticks actually.
Ok, thanks for the information. 
(For reviewers, note that it does not resolve the original issue)

nicklasb, 2024-04-08:
@mythbuster5 
 
So I am back at getting the RaspberryPy be able to talk to the ESP32 properly again, I am still having the same problem:
I (19053) DEVKIT_V5: I2C Master - << Reading receipt from 4, try 0.
I (19054) DEVKIT_V5: I2C Master - << Read receipt failure, code 263.

This is the code that times out (263 = 107 = ESP_ERR_TIMEOUT) within the millisecond:
read_ret = i2c_master_read_from_device(CONFIG_I2C_CONTROLLER_NUM, peer->i2c_address, dest_data, 2, 10000000);

(it doesn't matter at all what timeout i set)

I am seeing a blip on the logic analyzer around this time, so it seems like it is doing something, however, it seems that it basically immidiately times out, 1 millisecond is just not possibly a reasonable timeout. 
 
Ticks or not, I cannot see that this should ever happen?