espressif/esp-idf #12802: Unable to clone release/v4.4

tanyanfei, 2023-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v4.4

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git

Download successful

### What is the actual behavior?

git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git

Cloning into 'esp-idf'...
fatal: Remote branch release/v4.4 not found in upstream origin

### Steps to reproduce.

git clone -b release/v4.4 --recursive https://github.com/espressif/esp-idf.git

SoucheSouche, 2023-12-15:
Hi @tanyanfei,
I am not really sure what is the cause of this error. It seems to be working just fine on our end.
Have you tried to run
`git clone --recursive https://github.com/espressif/esp-idf.git`
followed by
`git checkout releave/v4.4`
to check if this works for you?

tanyanfei, 2023-12-16:
The same problem.

[image]

tanyanfei, 2023-12-16:
But the release/v4.3 is ok.

[image]

tanyanfei, 2023-12-16:
I need to upgrade to the release/v4.4
Can you tell me how to clone it ?

closed 2023-12-17 by tanyanfei