espressif/esp-idf #12807: Secure Boot V2, flash encryption and merget binary for production ESP32-S3

MacWyznawca, 2023-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

What steps should be taken to generate a merged binary file for production with Secure Boot V2 and flash encryption enabled?

Boot loader signed with 3 keys, flash memory encrypted with the key generated in the device. 

I upload the signed BootLoader and then the other files without any problem. The problem is with the merged file.

I also have no problem uploading the linked binary file to the encrypted module in development mode.
Also, there was no problem with the linked file and flash encryption enabled, but without SecureBoot v2.

For the merge, I use the command: `esptool.py --chip esp32s3 merge_bin --flash_mode dio --flash_size keep --flash_freq 80m 0x0 bootloader-s.bin 0x9000 base/partition_table/partition-table.bin 0x5d000 base/ota_data_initial.bin 0x60000 base/Base.bin -o Base_TotalEncrypted.bin`

The module falls into a loop: 
[image]

mahavirj, 2023-12-18:
@MacWyznawca 

- Bootloader verification looks good here. Can you please confirm if you disabled the logs in bootloader, please share your sdkconfig once?
- How do you flash the merged binary, can you please share all commands that you are using here?
- Please share the device efuse summary

MacWyznawca, 2023-12-18:
Hi @mahavirj 
1. Yes, Bootloader has logs disabled
2. I use `esptool.py  --port /dev/cu.usbserial-11101 write_flash 0x0 Base_TotalEncrypted.bin`
3. :

```BLOCK0          (                ) [0 ] read_regs: 07800f01 000f0c08 a900d1f5 0050004b 00000102 00000000
MAC_SPI_8M_0    (BLOCK1          ) [1 ] read_regs: 187bbda0 00003485 00000000 02040000 5020dcc8 7805b030
BLOCK_SYS_DATA  (BLOCK2          ) [2 ] read_regs: d0dcba6a 829bf6b8 3a0a4bbd c024634c f1abd441 30d6b4b3 c4f2d4b2 3a7ef6e6
BLOCK_USR_DATA  (BLOCK3          ) [3 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY0      (BLOCK4          ) [4 ] read_regs: 9190c301 ec3c0281 e17d4d67 92256964 a613e511 b74e384d 4a23c870 30318952
BLOCK_KEY1      (BLOCK5          ) [5 ] read_regs: 58326c16 8b749cd3 9bda961f c4bf4121 a6b5b538 cbd3f2a1 80621b92 c55d81cd
BLOCK_KEY2      (BLOCK6          ) [6 ] read_regs: db23d31c 890cf942 2af3feca 9309d627 cd331033 34e91ea9 546ef1f3 eea3d0c0
BLOCK_KEY3      (BLOCK7          ) [7 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY4      (BLOCK8          ) [8 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_KEY5      (BLOCK9          ) [9 ] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
BLOCK_SYS_DATA2 (BLOCK10         ) [10] read_regs: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000

BLOCK0          (                ) [0 ] err__regs: 00000000 00000000 00000000 00000000 00000000 00000000
EFUSE_RD_RS_ERR0_REG        0x00000000```

MacWyznawca, 2023-12-18:
And Security features settings @mahavirj 
[image]

mahavirj, 2023-12-18:
@MacWyznawca 

> SPI_BOOT_CRYPT_CNT (BLOCK0)                        Enables flash encryption when 1 or 3 bits are set  = Disable R/W (0b000)
                                                   and disabled otherwise

It appears that flash encryption is not yet enabled on this device, `SPI_BOOT_CRYPT_CNT` is still 0. However, the flash encryption key block is programmed.

May I ask how you programmed the flash encryption key, was it programmed externally using `espefuse.py` tool? 

Additionally, please enable debug logs in the bootloader and see if there is any problem while enabling the flash encryption workflow.

MacWyznawca, 2023-12-18:
@mahavirj 
This is strange, because I did not program the device encryption key externally. It was supposed to be generated inside the device (as in the menuconfig settings).
With the inclusion of the logs I have a problem, because initially in the documentation stated that 0x9000 space is enough, and it turns out that it ran out.

mahavirj, 2023-12-18:
@MacWyznawca 

Okay. Most likely the device could not complete the entire workflow to enable the flash encryption feature, or it was aborted interim of its execution (e.g., by a reset) somehow. 

For the bootloader size issue, please change the partition table offset as highlighted in this doc guide [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/bootloader.html#bootloader-size).

Also, its better to fallback to flash encryption development mode for this device. Just in case, if anything goes wrong we have 1 chance to disable the flash encryption. Please see this guide for development mode [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/flash-encryption.html#development-mode).

MacWyznawca, 2023-12-18:
@mahavirj Thank you, 
I found places to expand the bootloader space. I will switch to developer mode as advised (although I need to do final testing on production anyway).
By the way, I have a question, because I couldn't find specific information in the documentation. At what point and at what stage are signatures added? It seems to me that they are added to the bootloader when signing it (I add all 3). Please confirm or hint 😉. 

Translated with www.DeepL.com/Translator (free version)

MacWyznawca, 2023-12-18:
@mahavirj It seems that something is wrong with the partition map.

Partition map:
```
# Name,   Type, SubType, Offset,  Size, Flags
sec_cert,       0x3F,   ,           0xf000,     0x3000, , # Never mark this as an encrypted partition
luon_0,         data,   66,         0x12000,    300k,       encrypted
otadata,        data,   ota,        ,           8k,
phy_init,       data,   phy,        ,           4k,
ota_0,          app,    ota_0,      0x60000,    3328k,
ota_1,          app,    ota_1,      ,           3328k,
nvs_keys,       data,   nvs_keys,   0x6e1000,   4K,     encrypted
nvs_keys_f,     data,   nvs_keys,   0x6e2000,   4K,     encrypted
factory_nvs,    data,   nvs,        0x6e3000,   16K,
nvs,            data,   nvs,        0x6e7000,   0x119000,
```

Logs:

```
SP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
Valid secure boot key blocks: 0 1 2
secure boot verification succeeded
load:0x3fce3980,len:0x3990
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x535c
entry 0x403c9908
I (69) boot: ESP-IDF v5.1.2-dirty 2nd stage bootloader
I (69) boot: compile time Dec 18 2023 11:49:02
I (69) boot: Multicore bootloader
I (73) boot: chip revision: v0.1
I (77) boot.esp32s3: Boot SPI Speed : 80MHz
I (82) boot.esp32s3: SPI Mode       : DIO
I (86) boot.esp32s3: SPI Flash Size : 8MB
I (91) boot: Enabling RNG early entropy source...
E (96) flash_parts: partition 0 invalid magic number 0x9d24
E (103) boot: Failed to verify partition table
E (108) boot: load partition table error!
```

mahavirj, 2023-12-18:
@MacWyznawca 

Since you have enabled `SECURE_BOOT_BUILD_SIGNED_BINARIES` configuration, the binaries would be signed by the build system as the last stage of the build process. However, please note that the build system only supports supplying path to a single private key and hence for appending more signature blocks (using more than 1 signing key), the signature must be appended manually as highlighted [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/secure-boot-v2.html#multiple-keys).

> E (96) flash_parts: partition 0 invalid magic number 0x9d24

- Please check if the bootloader is rebuilt and flashed after updating the partition table offset, otherwise it may be referring to the partition table at older offset
- Please read the contents of the flash starting from the partition table offset and match them with the build system generated partition table binary

MacWyznawca, 2023-12-18:
I sign the bootloader additionally with the command add signature.

I check partition data with command `esptool.py --no-stub -b 230400 -p /dev/cu.usbserial-11101 read_flash 0xc000 0xC00 ~/Desktop/part.bin` And I see that they do not correspond to the data uploaded to the same address, partition.

Let me remind you of the command I use to merge files: 
`esptool.py  --chip esp32s3 merge_bin --flash_mode dio --flash_freq 80m --flash_size keep 0x0 ../../../LuON/LuONbase/bootloader-S.bin 0x60000 ../../../LuON/LuONbase/LuON_Base.bin 0xc000 ../../../LuON/LuONbase/partition_table/partition-table.bin 0x5d000 ../../../LuON/LuONbase/ota_data_initial.bin  -o  LuON_Base_v4.bin`

If I first upload the signed bootloader and then execute -p /dev/cu.usbserial-11101 monitor flash with the same settings and addresses as I execute merge_bin, the application starts.

The merged file, however, does not want to run.

Is it possible to find somewhere documentation that shows step by step how to prepare merged file for factory upload with encryption and signature enabled?
With encryption alone, I never had a problem.

mahavirj, 2023-12-19:
@MacWyznawca 

There is no separate documentation for the `merge_bin` workflow. But I believe the command itself is only responsible to prepare a combined file based on the offsets and the data supplied. CC @radimkarnis @dobairoland 

Do you see same contents in the merged binary at offset 0xc000 as what you read from the flash? Or they are different?

MacWyznawca, 2023-12-19:
Hi @mahavirj 

Ok. In the flash memory zone for the partition map there are some data that I can not associate with anything.

Today I will check on a new, clean copy of the device, but all operations in the code that I perform using writing to flash memory concern other addresses, and in addition, the application itself is not even launched.

dobairoland, 2023-12-19:
JFYI, here is the documentation for merge_bin: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin

@mahavirj's statement is correct. It just joins pieces together.

MacWyznawca, 2023-12-19:
@dobairoland Thank you, this documentation I know and use. 
However, in the documentation for secureboot v2, the process looks like they recommend uploading the signed bootloader first and then the rest of the files. In production, we prefer to avoid this and upload one file. The encryption process alone slows down the firmware upload process a bit in production.

MacWyznawca, 2023-12-19:
Hi @mahavirj, @dobairoland 
Is it possible that the problem is caused by the absence of the digit 6 in the subtype of the `sec_cert` partition?

After completion, it was able to upload and run with developer encryption mode and signature secyreboot v2. Signed OTA updates also work.

I'm still waiting for a test in release mode, but I'd rather hold off until a response from you.

```
# Name,   Type, SubType, Offset,  Size, Flags
sec_cert,       0x3F,   6,          0x10000,    0x3000, 
```

mahavirj, 2023-12-20:
@MacWyznawca 

If the `subtype` is not defined for a partition, then it would be set to 6 (`undefined`). This was fixed in https://github.com/espressif/esp-idf/commit/9b654db032e2489cfd218171201a514829bb6d41. Even the `esp_secure_cert` partition format [guide](https://github.com/espressif/esp_secure_cert_mgr/blob/main/docs/format.md) do not specify explicit `subtype`. 

So, I think this should be okay, as long as the build system is generating the partition table binary.

mahavirj, 2024-01-08:
@MacWyznawca Any further update on this or we can close the issue?

closed 2024-01-08 by MacWyznawca