espressif/esp-idf #12809: task_wdt: esp_task_wdt_reset(783): task not found

Lagunaxx, 2023-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

Chip is ESP32-D0WDQ6 (revision v1.0)

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO-T-Display

### Power Supply used.

USB

### What is the expected behavior?

moving project from esp-idf 3.*.* to 5.*.*
on esp-idf 3.*.*watchdog didn't trigger any error

### What is the actual behavior?

after flashing chip error occured:
_task_wdt: esp_task_wdt_reset(783): task not found_

it triggers on every _esp_task_wdt_reset();_ 

### Steps to reproduce.

project here:
[https://gitflic.ru/project/lagunax/esp-32-ah-base-station]https://gitflic.ru/project/lagunax/esp-32-ah-base-station(url)

components for building project here (but  i commented most code and components do not makes any sens):
[https://gitflic.ru/project/lagunax/esp32-spidisplay]https://gitflic.ru/project/lagunax/esp32-spidisplay(url)
[https://gitflic.ru/project/lagunax/esp32system]https://gitflic.ru/project/lagunax/esp32system(url)

### Debug Logs.

```plain
Building in: /home/esp32/eclipse-workspace/BaseScreen/build
...
Build complete (0 errors, 155 warnings): /home/esp32/eclipse-workspace/BaseScreen/build
/home/esp32/.espressif/python_env/idf5.1_py3.11_env/bin/python /home/esp32/esp-idf/tools/idf_size.py /home/esp32/eclipse-workspace/BaseScreen/build/BaseScreen.map
Total sizes:
Used static DRAM:   12096 bytes ( 168640 remain, 6.7% used)
      .data size:    8640 bytes
      .bss  size:    3456 bytes
Used static IRAM:   55226 bytes (  75846 remain, 42.1% used)
      .text size:   54199 bytes
   .vectors size:    1027 bytes
Used Flash size :  225803 bytes
           .text:  163819 bytes
         .rodata:   61728 bytes
Total image size:  289669 bytes (.bin may be padded larger)
/home/esp32/.espressif/python_env/idf5.1_py3.11_env/bin/python /home/esp32/esp-idf/components/partition_table/gen_esp32part.py /home/esp32/eclipse-workspace/BaseScreen/build/partition_table/partition-table.bin
Total time taken to build the project: 44,392 ms

esp32@esp32:~/eclipse-workspace/BaseScreen$ ~/esp-idf/tools/idf.py -p $PORT monitor
Executing action: monitor
Running idf_monitor in directory /home/esp32/eclipse-workspace/BaseScreen
Executing "/home/esp32/.espressif/python_env/idf5.1_py3.11_env/bin/python /home/esp32/esp-idf/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --target esp32 --revision 0 /home/esp32/eclipse-workspace/BaseScreen/build/BaseScreen.elf -m '/home/esp32/.espressif/python_env/idf5.1_py3.11_env/bin/python' '/home/esp32/esp-idf/tools/idf.py' '-p' '/dev/ttyUSB0'"...
--- esp-idf-monitor 1.3.4 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
E (5320) task_wdt: esp_task_wdt_reset(783): task not foundI (13) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (13) boot: compile time Dec 15 2023 16:36:03
I (13) boot: Multicore bootloader
I (16) boot: chip revision: v1.0
I (20) boot.esp32: SPI Speed      : 40MHz
I (25) boot.esp32: SPI Mode       : DIO
I (29) boot.esp32: SPI Flash Size : 4MB
I (34) boot: Enabling RNG early entropy source...
I (39) boot: Partition Table:
I (43) boot: ## Label            Usage          Type ST Offset   Length
I (50) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (58) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (65) boot:  2 app0             OTA app          00 10 00010000 00200000
I (72) boot:  3 spiffs           Unknown data     01 82 00210000 001f0000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=0f220h ( 61984) map
I (115) esp_image: segment 1: paddr=0001f248 vaddr=3ffb0000 size=00dd0h (  3536) load
I (117) esp_image: segment 2: paddr=00020020 vaddr=400d0020 size=27fech (163820) map
I (180) esp_image: segment 3: paddr=00048014 vaddr=3ffb0dd0 size=013f0h (  5104) load
I (182) esp_image: segment 4: paddr=0004940c vaddr=40080000 size=0d7bch ( 55228) load
I (215) boot: Loaded app from partition at offset 0x10000
I (216) boot: Disabling RNG early entropy source...
E (345) task_wdt: esp_task_wdt_reset(783): task not found
E (370) task_wdt: esp_task_wdt_reset(783): task not found
E (395) task_wdt: esp_task_wdt_reset(783): task not found
E (420) task_wdt: esp_task_wdt_reset(783): task not found
E (445) task_wdt: esp_task_wdt_reset(783): task not found
E (470) task_wdt: esp_task_wdt_reset(783): task not found
E (495) task_wdt: esp_task_wdt_reset(783): task not found
E (520) task_wdt: esp_task_wdt_reset(783): task not found
E (545) task_wdt: esp_task_wdt_reset(783): task not found
E (570) task_wdt: esp_task_wdt_reset(783): task not found
E (595) task_wdt: esp_task_wdt_reset(783): task not found
...
...
...
```

### More Information.

commenting/removing __esp_task_wdt_reset_() removes those errors

SoucheSouche, 2023-12-19:
Hi @Lagunaxx, do you encounter the same issue if you execute the loop in `app_main` instead of in a separate task?

Dazza0, 2024-01-12:
@Lagunaxx I've taken a quick look at the code you've provided. In the `loopTask`, the task resets the TWDT every loop, but I don't see where the `loopTask` is subscribed to the TWDT.

Please ensure that all tasks that call `esp_task_wdt_reset()` have subscribed themselves to the TWDT using `esp_task_wdt_add()`. For further details, please refer to the [TWDT API Reference](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/api-reference/system/wdts.html#task-watchdog-timer-twdt).

closed 2024-01-12 by Dazza0