espressif/esp-idf #12812: ret undeclared in esp_ota_ops.c when CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE

jschott780, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-892-g692c1fcc52

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successful build when CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE enabled.

### What is the actual behavior?

Compile error:
components/app_update/esp_ota_ops.c:970:5: error: 'ret' undeclared (first use in this function)                      
  970 |     ret = esp_ota_get_state_partition(running_app_part, &running_app_state); 

### Steps to reproduce.

#define CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE 1

### More Information.

Move the declaration of ret above the #ifdef CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE
esp_err_t ret = ESP_OK;
#ifdef CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE

hmalpani, 2023-12-18:
Fix in https://github.com/espressif/esp-idf/pull/12813

hmalpani, 2024-01-10:
[PR](https://github.com/espressif/esp-idf/pull/12813) for the fix has been merged. Closing this issue.

closed 2024-01-10 by hmalpani