espressif/esp-idf #12814: uxTaskGetSystemState example in latest docs with IDF 5.1.2 does not compile

avrono, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#_CPPv420uxTaskGetSystemStatePC12TaskStatus_tK11UBaseType_tPC27configRUN_TIME_COUNTER_TYPE](url)

Code given in the above link results in a compile error even when `configUSE_TRACE_FACILITY` is enabled 

 `error: unknown type name 'configRUN_TIME_COUNTER_TYPE'`

Adding `configGENERATE_RUN_TIME_STATS ` does not change the compile error either

According to [https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/freertos_idf.html#_CPPv427ulTaskGetIdleRunTimeCounterv](url)

`configGENERATE_RUN_TIME_STATS` + `configUSE_STATS_FORMATTING_FUNCTIONS` + `INCLUDE_xTaskGetIdleTaskHandle` must all be enabled for these functions to be available. However there is no obvious way to enable `INCLUDE_xTaskGetIdleTaskHandle`

igrr, 2023-12-16:
configRUN_TIME_COUNTER_TYPE was only introduced recently, so the new example won't work as is with an older version of IDF. In general, try the examples bundled with the version of IDF you are using.

avrono, 2023-12-18:
@igrr the documentation refers to the latest 5.1.2 which is the version I am running. Please advise ...

igrr, 2023-12-18:
That's not the case, `latest` docs (i.e. https://docs.espressif.com/projects/esp-idf/en/latest/) are for the `master` branch. Please take a look here for instructions on finding the docs for your version: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#releases.

(Github also labels the most recent stable release as "Latest" on the release page, the Readthedocs docs URL for the latest stable release is https://docs.espressif.com/projects/esp-idf/en/stable/  — which at the time of writing points to 5.1.2 docs.)

avrono, 2023-12-18:
My Mistake, thanks for the help.

closed 2023-12-18 by avrono