espressif/esp-idf #12815: Windows install of IDF (tried versions 4.0 - 5.1.2) fails at Python Environment install.

JoeB2, 2023-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-tools-setup-offline-5.1.2.exe

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

The installer should install the idf.  FYI: To date no IDF installer for Windows has EVER worked for me!  I have attempted over the past 5 years on 5 different W7, W8, W8.1, or W10 machines.  Sometimes I can get it installed by manually installing each prerequisite followed by git clone followed by ..\install.bat and executing export.bat prior to command line use.

This occurrence - As administrator: cmd.exe
d:\esp32>esp-idf-tools-setup-offline-5.1.2.exe  /DIR="d:\esp32\" /FORCECLOSEAPPLICATIONS /VERYSILENT /SUPPRESSMSGBOXES /SP- /NOCANCEL
Also tried without parameters same error in same result FAILED with same logged error.

### What is the actual behavior?

The Log File from the last success action to its end:

2023-12-16 10:14:17.002   --------
2023-12-16 10:14:17.002   ESP-IDF Python Virtual environment does not exist, creating the environment: D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2\python.exe -m venv "D:\esp32\xtensa-esp32-elf\bin\python_env\idf5.1_py3.11_env"
2023-12-16 10:14:24.078   Done, exit code=0
2023-12-16 10:14:24.078   --------
2023-12-16 10:14:24.078   Running command: D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2\python.exe -m venv "D:\esp32\xtensa-esp32-elf\bin\python_env\idf5.1_py3.11_env"
                          
2023-12-16 10:14:24.078   --------
2023-12-16 10:14:24.078   Installing Python environment:D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2\python.exe "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py" "--idf-path" "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2"   install-python-env
2023-12-16 10:14:24.390   Done, exit code=1
2023-12-16 10:14:24.390   --------
2023-12-16 10:14:24.390   Running command: D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2\python.exe "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py" "--idf-path" "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2"   install-python-env
                          Creating a new Python environment in c:\windows
                          Error: [WinError 5] Access is denied: 'c:\\windows\\addins\\FXSEXT.ecf'
                          Traceback (most recent call last):
                            File "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 2687, in <module>
                              main(sys.argv[1:])
                            File "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 2679, in main
                              action_func(args)
                            File "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py", line 2098, in action_install_python_env
                              subprocess.check_call([sys.executable, '-m', 'venv',
                            File "subprocess.py", line 413, in check_call
                          subprocess.CalledProcessError: Command '['D:\\esp32\\xtensa-esp32-elf\\bin\\tools\\idf-python\\3.11.2\\python.exe', '-m', 'venv', '--clear', '--upgrade-deps', 'c:\\windows']' returned non-zero exit status 1.
                          
2023-12-16 10:14:24.390   --------
2023-12-16 10:14:24.390   Message box (OK):
                          Installation has failed with exit code 1
2023-12-16 10:17:38.555   User chose OK.
2023-12-16 10:17:38.570   Message box (Yes/No):
                          Installation log has been created, it may contain more information about the problem.
                          Display the installation log now?
2023-12-16 10:17:46.707   User chose Yes.
2023-12-16 10:17:46.738   CurStepChanged raised an exception.

### Steps to reproduce.

1. Step: start cmd as administrator
[... 46 lines cut ...]
Attempted to create a test.txt file in directory as user c:\windows\addins using explorer as non-admin - FAILED access denied
Attempted to create a test.txt file in directory as admin c:\windows\addins using explorer as non-admin - success

Attempts to change access permission on directory c:\windows\addins fails with all options for all users/groups grayed out
Attempts to change access permission on file c:\windows\addins\FXSEXT.ecf fails with all options for all users/groups grayed out

cmd.exe as admin tried:
D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2>python.exe "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py" "--idf-path" "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2"   install-python-env

Result:
"Creating a new Python environment in c:\windows
Error: [WinError 5] Access is denied: 'c:\\windows\\addins\\FXSEXT.ecf'"

Renamed directory: c:\windows\addins to c:\windows\addinsxxx
cmd.exe as admin:
D:\esp32\xtensa-esp32-elf\bin\tools\idf-python\3.11.2>python.exe "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2\tools\idf_tools.py" "--idf-path" "D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v5.1.2"   install-python-env

Result:
"Creating a new Python environment in c:\windows
Error: [WinError 5] Access is denied: 'c:\\windows\\addinsxxx\\FXSEXT.ecf'"

WTF: How did the python installer find the new folder name?
File c:\windows\addins\FXSEXT.ecf can't be rename, deleted, moved, or have its permissions changed - at least not by me, admin, or "trusted installer".

I turned off windows Defender and retried: Success!  I looked in the docs to see whether I missed instructions to disable defender during install.  Its probably there, but I never found it.  Had I found it or the install specified turning odd defender hours upon hours of frustration could have been avoided.

closed 2023-12-18 by dobairoland

jakub-kocka, 2023-12-18:
Also the paths looks not correct, if you don't have it set differently in your user/system environmental variables maybe it causes the desired install location. The default path should solve this as well.

Again, we are sorry for the troubles and also glad that you have finally solved it.

JoeB2, 2023-12-26:
Hi Roland,

Sorry for not responding to you sooner.

Yes, at the moment I am able to compile and have even gotten debug to work twice in a row!  My goal is to consistently use the IDF for my projects, although I do have admit to having given up more than once and resorting to PlatformIO.
I prefer the IDF approach to development primarily because of the superb documentation and my expectation that the all of the added "torture" is actually resulting in better chip architecture alignment.

I install my idf(s) at: D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v#.#.#...  example:  D:\esp32\xtensa-esp32-elf\bin\frameworks\esp-idf-v4.4.6\...
Right now, I have ...\esp-idf-v4.4.6 and ...\esp-idf-v5.1.2 copied from GitHub using git clone.  Sadly, I have yet to get the tools installed such that VSCode works correctly, unless I allow the install process to put the tools into my Windows10 profile (yuk).

In an ideal world the tools installation would install and reliably work with for any custom root directory installation for VSCode.  I prefer to have the option to delete the entire tree (IDF and tools) and simply git clone replace whenever needed or desired.

Roland, please feel free to comment, you can't hurt my feelings, what is your preferred approach (VSCode, PlatformIO, IDF, 4.4.4, 4.4.6, 5.1.2, etc).
Do you have a recommended IDF version for best performance, reliability (build and/or code execution)?  I am primarily leveraging the awesome ESP32S3 - and commonly use the following features: MCPWM, NETIF wifi, ESPNOW, I2C, PCNT.

I have included a couple snap-shots of directories from my multi-IDF versions' installs.  Other than my non-standard stuff, I am hoping that you might consider commenting on some of the official espressif sub-frameworks directories' intended function(s).

[cid:7561c06a-8ba6-43c1-9e7d-a1b2df42c7be][cid:c9af8ef0-f080-4698-b450-c5bf200f03fd][cid:e750c588-b507-45b0-b21b-ab117e95f0af]

Regardless, I greatly appreciate your getting back to me on my original installation problem - thank you.

Kind regards,
Joe Belson