espressif/esp-idf #12817: Connecting to ESP32 from STM32

shlomo-edu, 2023-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I working now with ESP32-S3 DevKit.
Now I want to migrate the code to STM32 and connect to the ESP32 via the UART.
How do I do it?

suda-morris, 2023-12-18:
https://github.com/espressif/esp-at

shlomo-edu, 2023-12-18:
I though about this question some more, and now I think that maybe all I need is just to take the code that I develop in the DevKit, and install it (with its dependencies) on the the STM32.
Am I right or wrong?

mantriyogesh, 2025-02-28:
Hello @shlomo-edu , Are you still looking for answer or can we close this thread?

Alvin1Zhang, 2025-03-14:
Thanks for reporting, will close due to inactivity, feel free to reopen.