espressif/esp-idf #12818: Difference between esp_event_handler_instance_[un]register*() and esp_event_handler_[un]register*() not very clear in documentation

hl037, 2023-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Maybe it should be explicitly stated that `esp_event_handler_instance_[un]register*()` does the same thing as `esp_event_handler_[un]register*()`, but a same same handler to the same event return yet another instance instead of replacing the previous one.

Also, I'm wondering why would one register twice the same handler to the same event ? Except if it is to pass a different argument. If that is so (That is what I think, but I had to check the sources to get to it), then that should also be documented.

hl037, 2023-12-17:
There is also this "legacy" things, which also not very meaningful : is it deprecated ? Is it because it's in process of refactoring, and later this instance pointer will become the handle ?

SoucheSouche, 2023-12-19:
Hi @hl037,

I think that the meaning of the `instance` parameter is fairly well defined in [esp_event_handler_instance_register](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/esp_event.html#_CPPv435esp_event_handler_instance_register16esp_event_base_t7int32_t19esp_event_handler_tPvP28esp_event_handler_instance_t) and any function in which it appears in general.

I agree that there could be a note added to inform that calling `esp_event_handler_instance_register` with `instance` set to NULL is the same as calling `esp_event_handler_register`.

closed 2024-01-30 by espressif-bot