espressif/esp-idf #12819: WiFi STA with AP reconnection starnge behavior

Lagunaxx, 2023-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-D0WDQ6 (revision 1)

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

TTGO-T-Display + WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

normal wi-fi disconnection and next normal connection

### What is the actual behavior?

after STA lost power AP do not try to check if connection lost, next connection try makes strange things on AP side

### Steps to reproduce.

1)AP on TTGO-T-Display
2)STA on WROOM-32
3)AP and STA power off
4)AP power on
5)wait AP boot
6)STA power on
 here STA normaly connected to AP (log1) 
7)STA power switch off
 here AP do not do anything (in AP log) 
8)STA power on

 here STA tries to connect to AP but with strange error:
_E (4048) wifi:Association refused temporarily, comeback time 1536 mSec_ 
and finaly shows that it connected:
```
I (5585) wifi:state: assoc -> assoc (0)
I (5594) wifi:state: assoc -> run (10)
I (5613) wifi:connected with wf_base, aid = 1, channel 1, 40U, bssid = 3c:71:bf:f0:5c:75
I (5614) wifi:security: WPA2-PSK, phy: bgn, rssi: -18
I (5616) wifi:pm start, type: 2
I (5622) [WiFi]: WiFi_handler event: STA connected
I (5659) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6624) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
```
AP doing strange things:
```
I (39844) system: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
AP_STADISCONNECTED. Clients=2
I (39848) wifi:starting SA query procedure with STA(a4:cf:12:5e:94:7c)
I (39858) wifi:Send SA Query req with transaction id 8138
I (39864) [WiFi]: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
I (40058) wifi:Send SA Query req with transaction id 8139
I (40258) wifi:Send SA Query req with transaction id 813a
I (40458) wifi:Send SA Query req with transaction id 813b
I (40658) wifi:Send SA Query req with transaction id 813c
I (40858) wifi:Send SA Query req with transaction id 813d
I (40882) wifi:STA not responded to 6 SA Query attempts, Reset connection sending disassoc
I (40886) wifi:station: a4:cf:12:5e:94:7c leave, AID = 1, bss_flags is 658547, bss:0x3fff558c
I (40887) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (40895) system: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
AP_STADISCONNECTED. Clients=1
I (40920) [WiFi]: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
```
on AP side do not trigger event AP_STACONNECTED

9) next STA poweroff + poweron  triggers event AP_STACONNECTED,,,

### Debug Logs.

```plain
AP log:

I (39844) system: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
AP_STADISCONNECTED. Clients=2
I (39848) wifi:starting SA query procedure with STA(a4:cf:12:5e:94:7c)
I (39858) wifi:Send SA Query req with transaction id 8138
I (39864) [WiFi]: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
I (40058) wifi:Send SA Query req with transaction id 8139
I (40258) wifi:Send SA Query req with transaction id 813a
I (40458) wifi:Send SA Query req with transaction id 813b
I (40658) wifi:Send SA Query req with transaction id 813c
I (40858) wifi:Send SA Query req with transaction id 813d
I (40882) wifi:STA not responded to 6 SA Query attempts, Reset connection sending disassoc
I (40886) wifi:station: a4:cf:12:5e:94:7c leave, AID = 1, bss_flags is 658547, bss:0x3fff558c
I (40887) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (40895) system: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
AP_STADISCONNECTED. Clients=1
I (40920) [WiFi]: WiFi_handler event: station a4:cf:12:5e:94:7c disconnected, AID=1
I (41408) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (41410) wifi:station: a4:cf:12:5e:94:7c join, AID=1, bgn, 40U
I (41429) system: WiFi_handler event: station a4:cf:12:5e:94:7c connected, AID=1
AP_STACONNECTED. Clients=1
I (41432) [WiFi]: WiFi_handler event: station a4:cf:12:5e:94:7c connected, AID=1
I (41665) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (42450) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2

=====================================================================================
STA log1 (after first power on):

I (701) wifi_init: WiFi IRAM OP enabled
I (705) wifi_init: WiFi RX IRAM OP enabled
I (711) [WiFi]: Setting WiFi configuration SSID wf_base...
I (718) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (823) wifi:mode : sta (a4:cf:12:5e:94:7c)
I (824) wifi:enable tsf
I (826) [WiFi]: WiFi_handler event: STA starting
I (833) wifi:Set ps type: 2

I (855) MB_TCP_SLAVE_PORT: Protocol stack initialized.
I (864) MB_TCP_SLAVE_PORT: Socket (#54), listener  on port: 502, errno=0
I (3279) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4030) wifi:state: init -> auth (b0)
I (4034) wifi:state: auth -> assoc (0)
I (4045) wifi:state: assoc -> run (10)
I (4057) wifi:connected with wf_base, aid = 1, channel 1, 40U, bssid = 3c:71:bf:f0:5c:75
I (4058) wifi:security: WPA2-PSK, phy: bgn, rssi: -19
I (4060) wifi:pm start, type: 2

I (4066) [WiFi]: WiFi_handler event: STA connected
I (4092) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (5068) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1

=====================================================================================
STA log2 (after second power on):

I (711) [WiFi]: Setting WiFi configuration SSID wf_base...
I (718) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (822) wifi:mode : sta (a4:cf:12:5e:94:7c)
I (823) wifi:enable tsf
I (825) [WiFi]: WiFi_handler event: STA starting
I (832) wifi:Set ps type: 2

I (854) MB_TCP_SLAVE_PORT: Protocol stack initialized.
I (864)G MB_TCP_SLAVE_PORT: Socket (#54), listener  on port: 502, errno=0
PIO33 = 0
I (3278) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4028) wifi:state: init -> auth (b0)
I (4032) wifi:state: auth -> assoc (0)
E (4048) wifi:Association refused temporarily, comeback time 1536 mSec
I (5585) wifi:state: assoc -> assoc (0)
I (5594) wifi:I (5595) gpio: GPIO[33]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
state: assoc -> run (10)
I (5613) wifi:connected with wf_base, aid = 1, channel 1, 40U, bssid = 3c:71:bf:f0:5c:75
I (5614) wifi:security: WPA2-PSK, phy: bgn, rssi: -18
I (5616) wifi:pm start, type: 2

I (5622) [WiFi]: WiFi_handler event: STA connected
I (5659) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6624) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1

=====================================================================================
STA log3 (after thirs power on):

I (711) [WiFi]: Setting WiFi configuration SSID wf_base...
I (718) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (822) wifi:mode : sta (a4:cf:12:5e:94:7c)
I (823) wifi:enable tsf
I (825) [WiFi]: WiFi_handler event: STA starting
I (832) wifi:Set ps type: 2

I (854) MB_TCP_SLAVE_PORT: Protocol stack initialized.
I (8GPIO3364) MB_TCP_SLAVE_PORT: Socket (#54), listener  on port: 502, errno=0
 = 0
I (3279) wifi:new:<1,1>, old:<1,0>, ap:<255,255>, sta:<1,1>, prof:1
I (4031) wifi:state: init -> auth (b0)
I (4034) wifi:state: auth -> assoc (0)
E (4052) wifi:Association refused temporarily, comeback time 1536 mSec
I (5589) wifi:state: assoc -> assoc (0)
I (5599) wifi:state: assoc -> run (10)
I (5612) wifi:connected with wf_base, aid = 1, channel 1, 40U, bssid = 3c:71:bf:f0:5c:75
I (5613) wifi:security: WPA2-PSK, phy: bgn, rssi: -19
I (5615) wifi:pm start, type: 2

I (5621) [WiFi]: WiFi_handler event: STA connected
I (5686) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (6622) esp_netif_handlers: sta ip: 192.168.4.2, mask: 255.255.255.0, gw: 192.168.4.1
```

### More Information.

registered 2 handler functions for wifi messages on AP side on different places.

nishanth-radja, 2023-12-18:
@Lagunaxx This is a expected behavior, PMF is enabled by default so when the STA leaves the AP without sending deauth and tries to connect back to the AP, SA query process will start .
This is a security feature,to make sure Rouge STA does not send spoofed packets to disrupt the network.

Lagunaxx, 2023-12-20:
> @Lagunaxx This is a expected behavior, PMF is enabled by default so when the STA leaves the AP without sending deauth and tries to connect back to the AP, SA query process will start . This is a security feature,to make sure Rouge STA does not send spoofed packets to disrupt the network.

but when STA leaves AP do not monitor it and think that STA still connected

nishanth-radja, 2024-03-04:
@Lagunaxx  Yes, when the STA leaves AP without sending the deauth, AP will  think that STA is still connected. AP's sta list will not be cleared. There is a optional feature,called keep alive timer. If the AP does not receive anything packet from the STA for a period of "keep_alive timer" it will deauth the sta. But this feature is not present on most of the AP's.

vik-gokhale, 2024-03-20:
This GitHub ticket can be closed since the behaviour is as mentioned in the WIFI specification.