espressif/esp-idf #12820: [Heap Trace Standalone] perf bug causes double traversal

chipweinberger, 2023-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

relevant issue: https://github.com/espressif/esp-idf/issues/11173

In response to that issue, we added this code:

```
static HEAP_IRAM_ATTR heap_trace_record_t* map_find_and_remove(void *p)
{
    size_t idx = hash_idx(p);
    heap_trace_record_t *r_cur = NULL;
    SLIST_FOREACH(r_cur, &hash_map[idx], slist_hashmap) {
        if (r_cur->address == p) {
            total_hashmap_hits++;
            SLIST_REMOVE(&hash_map[idx], r_cur, heap_trace_record_t, slist_hashmap);
            return r_cur;
        }
    }
    total_hashmap_miss++;
    return NULL;
}
```

however, it uses `SLIST_REMOVE` which causes another traversal. We should have used `SLIST_REMOVE_AFTER`

something like this:

```
static HEAP_IRAM_ATTR heap_trace_record_t* map_find_and_remove(void *p)
{
    size_t idx = hash_idx(p);
    heap_trace_record_t *r_cur = NULL;
    heap_trace_record_t *r_prev = NULL;
    SLIST_FOREACH(r_cur, &hash_map[idx], slist_hashmap) {
        if (r_cur->address == p) {
            total_hashmap_hits++;
            if (r_prev) {
               SLIST_REMOVE_AFTER(r_prev, slist_hashmap);
            } else {
                SLIST_REMOVE_HEAD(r_cur, slist_hashmap);
            }
            return r_cur;
        }
        r_prev = r_cur;
    }
    total_hashmap_miss++;
    return NULL;
}
```

chipweinberger, 2023-12-17:
@SoucheSouche

SoucheSouche, 2023-12-18:
@chipweinberger, I will have a look at it during the week.

SoucheSouche, 2023-12-18:
@chipweinberger, I will update `map_find_and_remove` as suggested. I forgot that `SLIST_REMOVE` does go through the list to remove element unlike `TAILQ_REMOVE`  used in the former implementation. Thanks for noticing :)

closed 2023-12-23 by espressif-bot