espressif/esp-idf #12821: Online installer fails to complete installation of 5.1 branch on Windows10

idea--list, 2023-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I was expecting being able to install the 5.1 release branch on a clean windows 10 installation.

### What is the actual behavior?

The online installer cloned the 5.1 release branch, resolved deltas and even updated all the files. But then it just left there hanging without ever completing the installation.

### Steps to reproduce.

1. Clean install a windows 10 on your computer
2. Try to install the 5.1 branch using the online installer

jakub-kocka, 2023-12-18:
Hello @idea--list,
can you please provide installation log file?

idea--list, 2023-12-18:
@jakub-kocka 
Sorry, i can not do that as i almost immediately deleted the directory after the failed installation.
Right after that i installed v 5.1.2 using the offline installer... and now i re-run the online installer and chose the 5.1 release branch.
This time it successfully installed that version in paralell to 5.1.2.

jakub-kocka, 2023-12-19:
@idea--list,
I have made a few clean installs of Windows and try to use online installer to get 5.1 release branch as you described.

In one case I got the exact same issue as you (others run successfully) so it is not repeating behaviour and what fixed it for me in that case is to **re-run the installation process**.

We will need to investigate this more, however as I understand correctly you have been able to overcome this issue so I am glad for that.

jakub-kocka, 2023-12-21:
It turns out that installer have been using older git version and it should cause this issue.
So the git version was updated.
Also the "release branch" is cloned from GitHub, the "release version" is downloaded as a zip file. This has been also updated to be more descriptive when choosing.

The most reliable process is to use the "release version" and than switch to "release branch",
basically as you did but for both the online installer could have been used.

New idf-installer release (2.24) has above mentioned updates included and it is already available.

closed 2024-01-08 by dobairoland