espressif/esp-idf #12822: Adding esp_lcd component to build only serves the purpose of adding an include...

kdschlosser, 2023-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0.2

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

when a component is added to `IDF_COMPONENTS ` what takes place when that is done? It appears that eith the esp_lcd component the only thing that gets done is the include path to the head files gets added to the build. The source files are already added to the list of files to be compiled.

I would have thought that the source files only get added when a component gets added to `IDF_COMPONENTS`. why compile something that doesn't get used at all?

If a component has not been added to `IDF_COMPONENTS` and I add the include path to that component I would expect the build to fail. There should be linking errors because the source files have not been compiled.

### What is the actual behavior?

I am able to successfully build having a dependency on the esp_lcd component without having to add the component to `IDF_COMPONENTS`. I am able to do this by adding the include path to the esp_lcd headers to the compiler options. I do not have to add any of the source files so they get compiled.

### Steps to reproduce.

have code that uses some part of the esp_lcd component. don't add esp_lcd to `IDF_COMPONENTS`. In the cmake file add `${IDF_PATH}/components/esp_lcd/include/` to the includes. 

You should get linker errors but that doesn't happen

### More Information.

This has been tested on 2 different computers both with clean clones of the IDF version 5.0.2

suda-morris, 2023-12-18:
> How did you build your project?
Command line with Make

IDF has dropped the Make build system support since 5.0: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/build-system.html#migrating-from-gnu-make-build-system

If your project doesn't use esp_lcd component, and you don't want to see that component get included to the build, one quick solution is: https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/spi_lcd/CMakeLists.txt#L5 and the specify the dependency for your main component manually: https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/spi_lcd/main/CMakeLists.txt#L7

kdschlosser, 2023-12-18:
I'm compiling for MicroPython and that is the build system they use for the main entry point. The only MCU that MicroPython supports that uses CMAKE is the ESP32. The make build system sets everything up and then it runs cmake to do the build. I kind of goofed when I stated make it should have been cmake because that is what is really being used.

My question then becomes what is the purpose of `IDF_COMPONENTS`. If all of the source files are added by default than there is no reason to even bother with `IDF_COMPONENTS` may as well just add the include paths since the source files are already being compiled anyhow.

I do really believe this is a bug because the purpose of `IDF_COMPONENTS` is to add components to be built. If all components are already added then they shouldn't be because that is the purpose of `IDF_COMPONENTS`, hence that would be a bug.

kdschlosser, 2023-12-18:
adding `set(COMPONENTS main)`  doesn't work. The esp_lcd component still gets added to the build. I know this is happening because the only thing I have to do is add an include to the header files and my code compiles without any errors. That means the esp_lcd source files have been added to the build when I have not instructed the IDF to add them.

closed 2023-12-18 by kdschlosser

igrr, 2023-12-18:
> My question then becomes what is the purpose of IDF_COMPONENTS

This variable looks like something defined in Micropython build system, over here: https://github.com/micropython/micropython/blob/3270d856fda58585d44dd05aefa7a95551fa76cc/ports/esp32/esp32_common.cmake#L103. It is not something that is intrinsic to the IDF build system, so it's a bit hard to comment about the purpose from ESP-IDF project perspective.

> adding set(COMPONENTS main) doesn't work. 

@kdschlosser Could you please attach the log (preferably as a text file) of the build process, when doing a clean build?

To help us you troubleshoot this, you can also enable dependency graph generation in the project CMakeLists.txt file, like it is done here:
 https://github.com/espressif/esp-idf/blob/692c1fcc52b9b936c73dead4ef0c2ea1fbdfb602/tools/test_apps/system/g1_components/CMakeLists.txt#L13

This should result in a component dependency graph being generated during the build, you should find it in `build/component_deps.dot`. Please attach this file to the issue as well. It should help find, which of the components brings the dependency on esp_lcd.