espressif/esp-idf #12824: An error occurs when firmware upgrade using http after mqtt has been connected

LHYYSY, 2023-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.1

### Espressif SoC revision.

ESP32 C3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C3-13

### Power Supply used.

USB

### What is the expected behavior?

After you have connected to mqtt, use http for firmware upgrade

### What is the actual behavior?

连接到mqtt后,进行http连接后,读取数据报错:“E (34450) esp-tls-mbedtls:读取错误:0 x7100:

E (34451) transport_base: esp_tls_conn_read error, errno=Success

E (34451) HTTP_CLIENT: transport_read: error -1 | ESP_FAIL . E

data_read: 1

E (34459) native_esp_ota:错误:SSL数据读取错误-1"

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Debug Logs.

```plain
I (34239) native_esp_ota: New firmware version: 94dc6cc6-dirty
I (34243) native_esp_ota: Running firmware version: c9559fae-dirty
I (34249) native_esp_ota: esp_ota_begin succeeded
I (34337) native_esp_ota: current_dl_len:1024,progress_tmp:0
data_read :1024
I (34377) native_esp_ota: current_dl_len:2048,progress_tmp:0
E (34414) esp-tls-mbedtls: read error :-0x7100:
E (34415) transport_base: esp_tls_conn_read error, errno=Success
E (34416) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
data_read :223
I (34424) native_esp_ota: current_dl_len:2271,progress_tmp:0
E (34450) esp-tls-mbedtls: read error :-0x7100:
E (34451) transport_base: esp_tls_conn_read error, errno=Success
E (34451) HTTP_CLIENT: transport_read: error - -1 | ESP_FAIL
data_read :-1
E (34459) native_esp_ota: Error: SSL data read error -1
```

### More Information.

I have Settings:Using dynamic TX/RX buffer and Using dynamic TX/RX buffer

ESP-YJM, 2023-12-18:
你能把 mbedtls debug 打开看下日志吗？

LHYYSY, 2023-12-18:
> 你能把 mbedtls debug 打开看下日志吗？
I (26618) mbedtls: ssl_msg.c:2124 => flush output

I (26618) mbedtls: ssl_msg.c:2133 <= flush output

I (26623) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_HELLO_REQUEST

I (26631) mbedtls: ssl_msg.c:2124 => flush output

I (26655) mbedtls: ssl_msg.c:2133 <= flush output

I (26656) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_CLIENT_HELLO

I (26658) mbedtls: ssl_client.c:906 => write client hello

W (26669) mbedtls: ssl_client.c:258 got supported group(001d)

W (26671) mbedtls: ssl_client.c:258 got supported group(0017)

W (26692) mbedtls: ssl_client.c:258 got supported group(0018)

W (26693) mbedtls: ssl_client.c:258 got supported group(0019)

W (26699) mbedtls: ssl_client.c:258 got supported group(001a)

W (26705) mbedtls: ssl_client.c:258 got supported group(001b)

W (26730) mbedtls: ssl_client.c:258 got supported group(001c)

I (26733) mbedtls: ssl_msg.c:2554 => write handshake message

I (26733) mbedtls: ssl_msg.c:2714 => write record

I (26740) mbedtls: ssl_msg.c:2851 <= write record

I (26744) mbedtls: ssl_msg.c:2675 <= write handshake message

I (26769) mbedtls: ssl_client.c:994 <= write client hello

I (26770) mbedtls: ssl_msg.c:2124 => flush output

I (26770) mbedtls: ssl_msg.c:2138 message length: 224, out_left: 224

I (26779) mbedtls: ssl_msg.c:2145 ssl->f_send() returned 224 (-0xffffff20)

I (26785) mbedtls: ssl_msg.c:2172 <= flush output

I (26811) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_HELLO

I (26811) mbedtls: ssl_msg.c:1926 => fetch input

I (26815) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (26821) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (26827) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (26847) mbedtls: ssl_msg.c:2111 <= fetch input

I (26848) mbedtls: ssl_tls12_client.c:1195 => parse server hello

I (26855) mbedtls: ssl_msg.c:3887 => read record

I (26861) mbedtls: ssl_msg.c:1926 => fetch input

I (26866) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 5

I (26886) mbedtls: ssl_msg.c:2111 <= fetch input

I (26887) mbedtls: ssl_msg.c:1926 => fetch input

I (26891) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 70

I (26898) mbedtls: ssl_msg.c:2086 in_left: 5, nb_want: 70

I (26904) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 65 (-0xffffffbf)

I (26925) mbedtls: ssl_msg.c:2111 <= fetch input

I (26927) mbedtls: ssl_msg.c:3959 <= read record

I (26932) mbedtls: ssl_tls12_client.c:1446 server hello, total extension length: 21

I (26940) mbedtls: ssl_tls12_client.c:1660 <= parse server hello

I (26962) mbedtls: ssl_msg.c:2124 => flush output

I (26963) mbedtls: ssl_msg.c:2133 <= flush output

I (26965) mbedtls: ssl_tls.c:3859 client state: MBEDTLS_SSL_SERVER_CERTIFICATE

I (26973) mbedtls: ssl_msg.c:1926 => fetch input

I (26978) mbedtls: ssl_msg.c:2066 in_left: 0, nb_want: 5

I (27001) mbedtls: ssl_msg.c:2086 in_left: 0, nb_want: 5

I (27002) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 5 (-0xfffffffb)

I (27007) mbedtls: ssl_msg.c:2111 <= fetch input

I (27012) mbedtls: ssl_tls.c:7559 => parse certificate

I (27018) mbedtls: ssl_msg.c:3887 => read record

I (27041) mbedtls: ssl_msg.c:1926 => fetch input

I (27042) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 5

I (27045) mbedtls: ssl_msg.c:2111 <= fetch input

I (27051) mbedtls: ssl_msg.c:1926 => fetch input

I (27058) mbedtls: ssl_msg.c:2066 in_left: 5, nb_want: 3090

I (27080) mbedtls: ssl_msg.c:2086 in_left: 5, nb_want: 3090

I (27081) mbedtls: ssl_msg.c:2089 ssl->f_recv(_timeout)() returned 3085 (-0xfffff3f3)

I (27085) mbedtls: ssl_msg.c:2111 <= fetch input

I (27132) mbedtls: ssl_msg.c:3959 <= read record

***ERROR*** A stack overflow in task esp_http_ota_ta has been detected.

ESP-YJM, 2023-12-18:
你如果单独跑 https 会有问题吗？如果把 dynamic TX/RX buffer 关了会有问题吗？看这日志是客户端解析出的 Mbedtls 数据长度与对端发过来的长度不一样。

LHYYSY, 2023-12-18:
> 你如果单独跑 https 会有问题吗？如果把 dynamic TX/RX buffer 关了会有问题吗？看这日志是客户端解析出的 Mbedtls 数据长度与对端发过来的长度不一样。

单独跑https是没问题的，我开了dynamic TX/RX buffer 是为了节省一些内存，Mbedtls 数据长度是跟dynamic TX/RX buffer有影响？
[image]
我可以认为mqtt和https是可以同时使用的吧，只是我在menuconfig的配置有问题？

LHYYSY, 2023-12-18:
我把ynamic TX/RX buffer 关了也出现了这个报错

> 你如果单独跑 https 会有问题吗？如果把 dynamic TX/RX buffer 关了会有问题吗？看这日志是客户端解析出的 Mbedtls 数据长度与对端发过来的长度不一样。

ESP-YJM, 2023-12-18:
TLS maximum incoming fragment length 先改成默认的 16K，这个值一般情况下最好不要改小。

ESP-YJM, 2023-12-18:
正常的 https 和 mqtts 是可以一起跑的，不影响。dynamic TX/RX buffer 建议关了是排查看是不是由于该配置导致的。

LHYYSY, 2023-12-18:
> TLS maximum incoming fragment length 先改成默认的 16K，这个值一般情况下最好不要改小。

使用mqtt 连接亚马逊，使用到的证书验证花费了大量内存，因此我修改了参照文档修改了TLS maximum incoming fragment length，
[image]
请问如果我不修改TLS maximum incoming fragment length，我在mqtt 连接握手时我该如何的减少内存的开销？

LHYYSY, 2023-12-18:
> 改 TLS maximum incoming fragment length 这个值的确是可以省内存，但是有些服务器不支持协商 TLS maximum incoming fragment length 字段，导致服务器发的数据还是按照 16K 最大的发，导致我们收包解包就会有问题，所以除非确定服务器支持按照你设置的 5K 发包，否则不建议修改。另外握手减小开销的话，目前只能开启动态 mbedtls buffer 相关的配置，其他的好像也没有可以减少开销的。

非常感谢您的解答，开启动态 mbedtls buffer 相关的配置可以减少大概多少内存的使用

LHYYSY, 2023-12-18:
非常感谢您的解答

closed 2023-12-18 by LHYYSY