espressif/esp-idf #12828: How to migrate the code from ESP32-DevKit, to STM32 that uses the ESP32 via its UART?

shlomo-edu, 2023-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I developed the code for WiFi and for Bluetooth on the DevKit.
Now I want to add the ESP module to my STM32 based logger, and use the same code.
Can I just add the code (as is) to the STM32?
Is there a document that explain how to do the migraton?

mantriyogesh, 2024-12-17:
Hello @shlomo-edu , does [ESP-Hosted-MCU](https://github.com/espressif/esp-hosted-mcu) code base suit your needs?

As part of ESP-Hosted-MCU, also supports uart as transport.

mantriyogesh, 2024-12-17:
We also have stm32 ready port for wifi, if in case interested.

mantriyogesh, 2024-12-17:
Related https://github.com/espressif/esp-idf/issues/12817

mantriyogesh, 2025-02-28:
Hello @shlomo-edu , Are you still looking for answer or can we close this thread?

Alvin1Zhang, 2025-03-14:
Thanks for reporting, will close due to inactivity, feel free to reopen.

closed 2025-03-14 by Alvin1Zhang