espressif/esp-idf #12830: [ESP32S3] v4.4 SPI driver allocating interrupt even if polling is used

KonssnoK, 2023-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

More than a bug, this is a request for optimization.
Our system currently finished the available interrupts on CPU0. 
By analyzing the code of the IDF,
we saw that our SPI3 driver is currently using an interrupt, which is strange since we use such device in polling.

by looking at the code we can see that

```
static esp_err_t spi_master_init_driver(spi_host_device_t host_id)
{
...

    if (host_id != SPI1_HOST) {
        // interrupts are not allowed on SPI1 bus
        err = esp_intr_alloc(spicommon_irqsource_for_host(host_id),
                            bus_attr->bus_cfg.intr_flags | ESP_INTR_FLAG_INTRDISABLED,
                            spi_intr, host, &host->intr);
        if (err != ESP_OK) {
            goto cleanup;
        }
    }
```
independently on how the peripheral will be used.

It would be better to allocate such interrupt only if needed.
Thanks

Icarus113, 2023-12-19:
Thanks for the request, we'll consider this and give you a feedback later. @KonssnoK

KonssnoK, 2023-12-19:
Thanks,
this request is strictly related to https://github.com/espressif/esp-idf/issues/12832 .
If in the end the ESP32S3 has many more interrupts available this really becomes a minor optimization. :)