espressif/esp-idf #12831: BLE Connection panic abort after update idf 5.0.5

ffamar, 2023-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

 ESP-IDF v5.0.5

### Espressif SoC revision.

ESP32-D0WD chip revision: v1.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

The app connects to esp32 via BLE

### What is the actual behavior?

Panic abort

### Steps to reproduce.

Perform a ble connection

### Debug Logs.

```plain
I (27) boot: ESP-IDF v5.0.5 2nd stage bootloader
I (27) boot: compile time 16:21:17
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v1.0
I (34) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (43) boot.esp32: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 otadata          OTA data         01 00 00009000 00002000
I (72) boot:  1 nvs              WiFi data        01 02 0000b000 00505000
I (79) boot:  2 ota_0            OTA app          00 10 00510000 00500000
I (87) boot:  3 ota_1            OTA app          00 11 00a10000 00500000
I (94) boot: End of partition table
I (98) esp_image: segment 0: paddr=00510020 vaddr=3f400020 size=45a60h (285280) map
I (183) esp_image: segment 1: paddr=00555a88 vaddr=3ffbdb60 size=05138h ( 20792) load
I (189) esp_image: segment 2: paddr=0055abc8 vaddr=40080000 size=05450h ( 21584) load
I (196) esp_image: segment 3: paddr=00560020 vaddr=400d0020 size=e7f38h (950072) map
I (451) esp_image: segment 4: paddr=00647f60 vaddr=40085450 size=196a0h (104096) load
I (498) boot: Loaded app from partition at offset 0x510000
[... 111 lines cut ...]

0x4009adc1: __assert_func at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/newlib/assert.c:78

0x401b0ea1: xQueueSemaphoreTake at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:1556 (discriminator 1)

0x40102659: osi_mutex_lock at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/common/osi/mutex.c:49

0x401113f8: btc_config_lock at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/host/bluedroid/btc/core/btc_config.c:333

0x40104db5: btc_storage_update_active_device at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/host/bluedroid/btc/core/btc_storage.c:330

0x400e82e9: btc_gatts_cb_handler at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gatts.c:906

0x401010f1: btc_thread_handler at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/common/btc/core/btc_task.c:207

0x4010274b: osi_thread_run at C:/Works/Toolchain/esp/Espressif/frameworks/esp-idf-v5.0.5/esp-idf/components/bt/common/osi/thread.c:165

ELF file SHA256: 195623bfb1a147dc
```

### More Information.

After updating idf from 5.0.4 to 5.0.5 I get the panic interrupt when my app tries to connect to the device via BLE. The same firmware, has been working without problems for months with idf 5.0.4.

Thanks in advance for the help.

zhp0406, 2023-12-19:
How should this issue be reproduced?
Can you use the “git bisect” command to find the first problematic commit?

ffamar, 2023-12-19:
> How should this issue be reproduced? Can you use the “git bisect” command to find the first problematic commit?

Hi, @zhp0406 this is the result of git bisect:

```
git bisect good
dc85f670aa96e2c7d00b3ce916252e41cc41141d is the first bad commit
commit dc85f670aa96e2c7d00b3ce916252e41cc41141d
Author: gongyantao <gongyantao@espressif.com>
Date:   Fri Sep 8 18:56:47 2023 +0800

    fix(bt/bluedroid): update the newest active device in bluetooth

    1: update the newest active device when an ACL link is established and
       the remote device info is not in bond list. This updates will not stored
       into NVRAM until a new device is paired.

 components/bt/common/osi/config.c                  | 26 +++++++++
 components/bt/common/osi/include/osi/config.h      |  5 ++
 .../bt/host/bluedroid/btc/core/btc_ble_storage.c   | 48 -----------------
 components/bt/host/bluedroid/btc/core/btc_config.c | 10 +++-
 components/bt/host/bluedroid/btc/core/btc_dm.c     | 14 ++++-
 .../bt/host/bluedroid/btc/core/btc_storage.c       | 61 ++++++++++++++++------
 .../bluedroid/btc/include/btc/btc_ble_storage.h    |  2 -
 .../bt/host/bluedroid/btc/include/btc/btc_config.h |  1 +
 .../host/bluedroid/btc/include/btc/btc_storage.h   | 38 +++++++++-----
 .../bluedroid/btc/profile/std/gatt/btc_gattc.c     | 13 ++++-
 .../bluedroid/btc/profile/std/gatt/btc_gatts.c     | 16 +++++-
 11 files changed, 150 insertions(+), 84 deletions(-)
```

Using our firmware and application, the problem is systematic. In this firmware we use both WiFi st and BLE connection, 2 mqtt connections and one uart.
Thanks.

zhp0406, 2023-12-19:
@ffamar 
thanks for report the problem,
This is the patch to fix the issue. In the future, we will incorporate this fix into the latest version update.
[0001-fix-bt-bluedroid-fixed-btc_config_lock-error.patch](https://github.com/espressif/esp-idf/files/13715071/0001-fix-bt-bluedroid-fixed-btc_config_lock-error.patch)
(url)

ffamar, 2023-12-19:
@zhp0406 
This patch applied to idf version 5.0.5 solves the problem.
Thank you for your support.

closed 2023-12-19 by ffamar