espressif/esp-idf #12835: esp_event_handler_register doesn't always register two handlers for WIFI_EVENT_STA_DISCONNECTED

dannybackx, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am responding to this piece of the official documentation : 
**Registering multiple handlers to events is possible. Registering a single handler to multiple events is also possible. However, registering the same handler to the same event multiple times would cause the previous registrations to be overwritten.**

I have a component to handle WiFi connectivity. It should reconnect so it registers the STA_DISCONNECT. The application itself also wants to know that to stop some functionality when disconnected.
When I enable both calls, only one of them works.
Easy to reproduce by trying to register for an nonexisting access point.

```
ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &disconnect, &web_server));

ESP_ERROR_CHECK(esp_event_handler_register(WIFI_EVENT, WIFI_EVENT_STA_DISCONNECTED, &connect_handler_on_wifi_disconnect, NULL));
```

See 
svn+ssh://dannybackx@svn.code.sf.net/p/esp32-s3-86-box-project-base/code/trunk/alarm/main/App.cpp
and
svn+ssh://dannybackx@svn.code.sf.net/p/esp32-s3-86-box-project-base/code/trunk/alarm/components/wifi_connect/wifi_connect.c

dannybackx, 2023-12-19:
Note : currently using 5.2 beta 1

igrr, 2023-12-20:
I think the situation described by the documentation doesn't apply in your case, since you are registering different handler functions. The situation you are describing should work and we do have a unit test for it in components/esp_event/test_apps/main/test_event_common.cpp, so there must be a bug somewhere...

To troubleshoot this, could you try enabling CONFIG_ESP_EVENT_LOOP_PROFILING and then use `esp_event_dump(stdout)` to dump the information about the event loop, after the event has been dispatched? Please also enable CONFIG_LOG_DEFAULT_LEVEL_DEBUG and capture the logs generated while the event is dispatched.

dannybackx, 2023-12-20:
[script.ivan.txt](https://github.com/espressif/esp-idf/files/13729110/script.ivan.txt)

You can reference the exact sources for this on svn+ssh://dannybackx@svn.code.sf.net/p/esp32-s3-86-box-project-base/code/trunk/alarm/main/App.cpp and Network.cpp, see conditional compiles on IVAN_BUG_REPORT .