espressif/esp-idf #12836: [Mesh] v4.4

KonssnoK, 2023-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hello @zhangyanjiaoesp .
I have a question on when the device is running in fixed root.
Sometimes the nodes of the network cannot connect to the parent, for various reasons 
( WIFI_REASON_ASSOC_TOOMANY, MESH_REASON_PARENT_IDLE, MESH_REASON_CYCLIC etc).

Now, how can i tell the device that on such errors it need to stop trying to connect to such devices?

To start the connection i use
```
ESP_ERROR_CHECK(esp_mesh_set_parent(
                &best_result.parent,
                (mesh_addr_t*)&best_result.parent_assoc.mesh_id,
                best_result.type,
                best_result.layer
            ));
```
but i don t know how to do something like `esp_mesh_reset_parent`.

Maybe something like a "disconnect" ?

for reference, esp_mesh_disconnect does not work

```
I (156025) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156027) wifi:state: init -> auth (b0)
I (156031) wifi:state: auth -> init (65c0)
I (156031) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.283) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.287) mesh_hand: Stop mesh auto-reconnect
I (156130) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156131) wifi:state: init -> auth (b0)
I (156142) wifi:state: auth -> init (65c0)
I (156143) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.393) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.395) mesh_hand: Stop mesh auto-reconnect
I (156230) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156231) wifi:state: init -> auth (b0)
I (156235) wifi:state: auth -> init (65c0)
I (156236) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.487) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.492) mesh_hand: Stop mesh auto-reconnect
I (01:02:35.556) processor: send status... (1000ms)
W (156314) wifi:Haven't to connect to a suitable AP now!
I (156332) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156334) wifi:state: init -> auth (b0)
I (156338) wifi:state: auth -> init (65c0)
I (156339) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.590) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.595) mesh_hand: Stop mesh auto-reconnect
I (156435) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156437) wifi:state: init -> auth (b0)
I (156442) wifi:state: auth -> init (65c0)
I (156443) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.693) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.698) mesh_hand: Stop mesh auto-reconnect
I (156537) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156538) wifi:state: init -> auth (b0)
I (156542) wifi:state: auth -> init (65c0)
I (156543) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.795) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.799) mesh_hand: Stop mesh auto-reconnect
I (156639) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156641) wifi:state: init -> auth (b0)
I (156645) wifi:state: auth -> init (65c0)
I (156646) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.897) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:35.902) mesh_hand: Stop mesh auto-reconnect
I (156738) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156739) wifi:state: init -> auth (b0)
I (156744) wifi:state: auth -> init (65c0)
I (156745) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:35.996) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.000) mesh_hand: Stop mesh auto-reconnect
I (156844) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156845) wifi:state: init -> auth (b0)
I (156849) wifi:state: auth -> init (65c0)
I (156850) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.102) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.106) mesh_hand: Stop mesh auto-reconnect
I (156945) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (156947) wifi:state: init -> auth (b0)
I (156951) wifi:state: auth -> init (65c0)
I (156952) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.203) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.208) mesh_hand: Stop mesh auto-reconnect
I (157049) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157050) wifi:state: init -> auth (b0)
I (157054) wifi:state: auth -> init (65c0)
I (157055) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.307) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.311) mesh_hand: Stop mesh auto-reconnect
I (157151) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157153) wifi:state: init -> auth (b0)
I (157173) wifi:state: auth -> init (65c0)
I (157174) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.424) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.426) mesh_hand: Stop mesh auto-reconnect
I (157256) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157257) wifi:state: init -> auth (b0)
I (157261) wifi:state: auth -> init (65c0)
I (157262) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.514) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.518) mesh_hand: Stop mesh auto-reconnect
I (01:02:36.556) processor: send status... (1000ms)
W (157314) wifi:Haven't to connect to a suitable AP now!
I (157352) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157353) wifi:state: init -> auth (b0)
I (157358) wifi:state: auth -> init (65c0)
I (157359) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.610) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
W (01:02:36.614) mesh_hand: Stop mesh auto-reconnect
I (157459) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (157460) wifi:state: init -> auth (b0)
I (157465) wifi:state: auth -> init (65c0)
I (157466) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (01:02:36.717) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 101 MESH_REASON_PARENT_IDLE
```

KonssnoK, 2023-12-20:
for reference, this condition leads to the wifi scanning 0 stations

```
W (14:48:21.805) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAIL
I (14:48:21.806) mesh_main: <MESH_EVENT_SCAN_DONE>number:0
I (1525062) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1525064) wifi:state: init -> auth (b0)
I (1525082) wifi:state: auth -> init (5c0)
I (1525083) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:21.892) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:21.894) mesh_hand: Stop mesh auto-reconnect
I (14:48:22.143) processor: send status... (1000ms)
W (1525342) wifi:Haven't to connect to a suitable AP now!
I (14:48:23.143) processor: send status... (1000ms)
W (1526342) wifi:Haven't to connect to a suitable AP now!
W (14:48:24.143) monitor: 3W3P:: rule deactivated
W (14:48:30.143) monitor: 3W3P:: rule activated
I (14:48:30.144) processor: send status... (7000ms)
W (1533342) wifi:Haven't to connect to a suitable AP now!
W (14:48:31.143) monitor: 3W3P:: rule deactivated
I (1539398) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1539399) wifi:state: init -> auth (b0)
I (1539409) wifi:state: auth -> init (5c0)
I (1539410) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:36.218) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:36.220) mesh_hand: Stop mesh auto-reconnect
W (14:48:41.143) monitor: 3W3P:: rule activated
I (14:48:41.144) processor: send status... (11000ms)
W (1544342) wifi:Haven't to connect to a suitable AP now!
I (14:48:42.143) processor: send status... (1000ms)
W (1545342) wifi:Haven't to connect to a suitable AP now!
I (1545686) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1545687) wifi:state: init -> auth (b0)
I (1545696) wifi:state: auth -> init (5c0)
I (1545696) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:42.505) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:42.507) mesh_hand: Stop mesh auto-reconnect
I (14:48:43.143) processor: send status... (1000ms)
W (1546342) wifi:Haven't to connect to a suitable AP now!
I (14:48:44.143) processor: send status... (1000ms)
W (1547342) wifi:Haven't to connect to a suitable AP now!
W (14:48:45.143) monitor: 3W3P:: rule deactivated
I (1548612) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1548614) wifi:state: init -> auth (b0)
I (1548623) wifi:state: auth -> init (5c0)
I (1548624) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:45.432) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:45.434) mesh_hand: Stop mesh auto-reconnect
W (14:48:51.143) monitor: 3W3P:: rule activated
I (14:48:51.144) processor: send status... (7000ms)
W (1554342) wifi:Haven't to connect to a suitable AP now!
I (1554900) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1554902) wifi:state: init -> auth (b0)
I (1554911) wifi:state: auth -> init (5c0)
I (1554912) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:51.720) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:51.722) mesh_hand: Stop mesh auto-reconnect
I (14:48:51.795) mesh_hand: Checking wifi connectivity (count 0)[LOCAL]
W (14:48:51.803) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAIL
I (14:48:51.804) mesh_main: <MESH_EVENT_SCAN_DONE>number:0
W (14:48:52.143) monitor: 3W3P:: rule deactivated
I (1557843) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1557844) wifi:state: init -> auth (b0)
I (1557850) wifi:state: auth -> init (5c0)
I (1557851) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:48:54.660) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:48:54.666) mesh_hand: Stop mesh auto-reconnect
I (14:49:00.146) offline: object saved to PSRAM, free 516413
W (14:49:02.143) monitor: 3W3P:: rule activated
I (14:49:02.144) processor: send status... (11000ms)
[... 16 lines cut ...]
I (1574361) wifi:state: auth -> init (5c0)
I (1574362) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:49:11.172) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:49:11.176) mesh_hand: Stop mesh auto-reconnect
W (14:49:12.143) monitor: 3W3P:: rule activated
I (14:49:12.144) processor: send status... (7000ms)
W (1575342) wifi:Haven't to connect to a suitable AP now!
W (14:49:13.143) monitor: 3W3P:: rule deactivated
I (1577289) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1577290) wifi:state: init -> auth (b0)
I (1577297) wifi:state: auth -> init (5c0)
I (1577298) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:49:14.107) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:49:14.110) mesh_hand: Stop mesh auto-reconnect
I (1583575) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (1583576) wifi:state: init -> auth (b0)
I (1583583) wifi:state: auth -> init (5c0)
I (1583584) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (14:49:20.392) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (14:49:20.396) mesh_hand: Stop mesh auto-reconnect
I (14:49:21.795) mesh_hand: Checking wifi connectivity (count 0)[LOCAL]
W (14:49:21.804) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAIL
I (14:49:21.805) mesh_main: <MESH_EVENT_SCAN_DONE>number:0
E (14:49:21.809) mesh_main: Wifi cannot scan. Reset.
```

KonssnoK, 2023-12-20:
i confirm that also with the latest mesh version the issue is happening

```

W (15:45:57.133) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (15:45:57.141) mesh_hand: Stop mesh auto-reconnect
W (15:45:57.456) monitor: 3W3P:: rule activated
I (15:45:57.457) processor: send status... (7000ms)
W (2752031) wifi:Haven't to connect to a suitable AP now!
I (15:45:58.055) mesh_hand: Checking wifi connectivity (count 0)
W (15:45:58.060) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAIL
W (15:45:58.063) mesh_hand: Stop mesh auto-reconnect
I (15:45:58.067) mesh_main: <MESH_EVENT_SCAN_DONE>number:0
I (2752720) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (2752722) wifi:state: init -> auth (b0)
I (2752729) wifi:state: auth -> init (5c0)
I (2752730) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (15:45:58.159) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (15:45:58.166) mesh_hand: Stop mesh auto-reconnect
W (15:45:58.456) monitor: 3W3P:: rule deactivated
I (15:46:00.460) offline: object saved to PSRAM, free 520130
I (2755793) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (2755794) wifi:state: init -> auth (b0)
I (2755800) wifi:state: auth -> init (5c0)
I (2755801) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (15:46:01.231) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (15:46:01.239) mesh_hand: Stop mesh auto-reconnect
I (2758864) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (2758866) wifi:state: init -> auth (b0)
I (2758870) wifi:state: auth -> init (5c0)
I (2758871) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (15:46:04.302) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (15:46:04.310) mesh_hand: Stop mesh auto-reconnect
I (2759887) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
I (2759889) wifi:state: init -> auth (b0)
I (2759895) wifi:state: auth -> init (5c0)
I (2759896) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,1>, prof:1
W (15:46:05.325) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 5 WIFI_REASON_ASSOC_TOOMANY
W (15:46:05.333) mesh_hand: Stop mesh auto-reconnect
I (2760911) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,1>, prof:1
```

zhangyanjiaoesp, 2024-01-04:
@KonssnoK 
Since you have called `esp_mesh_set_parent()` for the node, when the parent disconnected, the node will always try to connect to the same parent, it won't scan select a new one. During the reconnection, the user scan will be cancelled, so the scan result always 0.
You can call `esp_mesh_set_self_organized(false, false)` to see whether it can stop the reconnection.

KonssnoK, 2024-01-08:
> @KonssnoK Since you have called `esp_mesh_set_parent()` for the node, when the parent disconnected, the node will always try to connect to the same parent, it won't scan select a new one. During the [...]

ok i ll try that

KonssnoK, 2024-01-11:
i confirm that the calling` esp_mesh_set_self_organized(false, false)` stops the reconnection attempts

closed 2024-01-11 by KonssnoK